Koenkk/zigbee2mqtt #19269: Sonoff TRVZB is missing attributes can be exposed, yet configurable using a SEND, SET command

aussiebum71, 2023-10-12:
### What happened?

The new Sonoff TRV https://www.zigbee2mqtt.io/devices/TRVZB.html is missing some attributes that can be exposed and changed. For instance **local_temperature_calibration** can be adjusted sending a Zigbee SET command. But it is not showing up in the Exposes page in Zigbee2MQTT. The TRV also has a child lock feature, that can possibly be exposed. 

Occupied heating setpoint cannot be set in Auto mode, it defaults to 16 degrees celsius. Not sure what Auto mode does? Should it be ECO mode?

TRVZB firmware verision 1.1.1

[image]

[image]

[image]

### What did you expect to happen?

Be able to see local temperature calibration, child lock and other missing, features and attributes in the Expose tab of the device

### How to reproduce it (minimal and precise)

Click on the Exposes tab of the TRVZB to see missing attributes

### Zigbee2MQTT version

1.33.1-1

### Adapter firmware version

20221226

### Adapter

Sonoff Zigbee USB  zStack3x0

infinitedestiny, 2023-11-01:
I think it is necessary to expose at least the setpoint relating to AUTO mode. Now it seems to be fixed at 16 degrees.

photomoose, 2023-11-01:
I've done some work to partially expose the child lock functionality which will appear in the next release... I've also received some documentation from Sonof regarding the proprietary cluster information. I'm expecting to receive my Zigbee packet sniffer in the next couple of days, so I will analyse what the app sends via the bridge and try to add more functionality for this device in the upcoming days.

hubi-IT, 2023-11-01:
I also asked Sonoff for documentation, I will try to help improve that device. I bought 6 pcs.

ferrarid61, 2023-11-04:
The calibration attribute is exposed by zigbee
[image]
I tryed and it works. Is it possible to send this  command via HA automation?

photomoose, 2023-11-04:
I am now actively working on this. 

**Child Lock**

Support for child lock functionality was added in release 1.33.2 - note that this version only "reports" the child lock status; it is not possible to set it remotely in this version. I have since completed the work to set it remotely and I will create a PR later tonight so that it can be made available in the next release.

**Open Window Detection**

I've implemented the reporting and setting of open window detection - I will also push these changes tonight.

Next quick wins are likely to be Frost Protection and Temperature Calibration - I will look into these tonight.

@aussiebum71 With respect to auto mode, the TRV has two modes: manual and auto. Manual is exactly what you'd expect - you manually dial in a temperature by twisting the TRV (or by setting it remotely via Home Assistant / Z2M) and it will stay at that temperature until you manually change it to something else (or a Home Assistant routine changes it etc). Auto mode (indicated with a clock icon on the TRV when you press the button) allows the TRV to automatically change the desired temperature according to a programmed schedule - non Z2M users that use the Sonoff Bridge and eWeLink app can define temperature schedules and program their TRVs to operate automatically. I will look into how this information can be exposed once I've completed the above two items. I imagine such functionality probably isn't essential given we're using Z2M to "manually" set the temperature from the TRV perspective. However, I can such functionality being useful as a fail safe (i.e. in case Home Assistant or Z2M becomes unavailable, at least the TRV will still operate according the programmed schedule).

@Koenkk could you assign this issue to me, please?

infinitedestiny, 2023-11-04:
@photomoose I think that AUTO Mode uses a sort of PID control to modulate the opening value of the valve.
In HEAT Mode the valve is just on/off controlled. No way to control partial open.

photomoose, 2023-11-04:
@infinitedestiny that could be a possibility, although I imagine that sort of control is internal rather than being operated over Zigbee.

Attached is photo of the schedule that can be set in the app. When in auto mode, the TRV automatically changed its temperature according to the time (the TRV polls the coordinator for the current time every so often).

[image]

infinitedestiny, 2023-11-04:
@photomoose yes I think that the PID control is internal, but it is useful to partially close the valve when reaching the set point. This helps to reduce the hysteresis. So it could be very useful to have the possibility to change setpoints in AUTO Mode. 
Another useful thing (but probably this value is not exposed) is to manual control the opening value of the valve, in this way you can use an external PID. 
I'd like to help, but I'm sorry... I don't have both packet sniffer and sonoff gateway :(...

kolkoo, 2023-11-05:
Could you expose the local_temperature_calibration if it is supported by the device as mentioned above? Doing this should allow (hopefully) to use this with "Better Thermostat" and integrate it with a remote temperature sensor as far as I understand. Cheers.

photomoose, 2023-11-05:
@kolkoo I'm working on temperature calibration next.

photomoose, 2023-11-06:
Next up is Frost Protection, coming later today.

photomoose, 2023-11-07:
Now taking a look at weekly schedule functionality. It's definitely supported in the traces I've taken - just testing whether it will work with Z2M natively.

Koenkk, 2023-11-07:
> does the existing schedule functionality work in the Z2M frontend? When I enter values, they disappear when moving to the next field.

This should be fixed in z2m 1.33.2

dafunkydan, 2023-11-07:
Great Development going on here! Looks like the best is yet to come ;-)

I am missing the `running_state:` as Entity in Homeassistant. It is already exposed in Z2M 1.33.2. I hope i'm just neither too dumb nor to blind - any chance to get a Hand on that Value?

aussiebum71, 2023-11-07:
This is how it's looking in the current DEV branch. *running_state* is
exposed. Battery low could be merged with Battery. Hopefully Sonoff will
provide a further firmware update to provide an ECO mode in addition to
Manual and Auto. And also an external temperature sync. Then this TRV will
rival pretty much most of the Zigbee TRVs on the market, I'll be replacing
my Moes TRVs with these, as you're able to query to get most settings and
not have to wait for an message update.

Photomoose is doing a fantastic job adding these extra attributes, they are
also working on exposing the schedules for Auto mode.

[image: image.png]

aussiebum71, 2023-11-07:
@dafunkydan 

This is how it's looking in the current DEV branch.`running_state` is exposed. Battery low could be merged with Battery. Hopefully Sonoff will provide a further firmware update to provide an ECO mode in addition to Manual and Auto. And also an external temperature sync. Then this TRV will rival pretty much most of the Zigbee TRVs on the market, I'll be replacing my Moes TRVs with these, as you're able to query to get most settings and not have to wait for an message update.
@photomoose is doing a fantastic job adding these extra attributes, they are also working on exposing the schedules for Auto mode.

[image]

photomoose, 2023-11-07:
@aussiebum71 I think @dafunkydan is talking about an explicit sensor entity for the running state in *Home Assistant*, not Z2M.

The TRV is exposed as a "climate" entity in Home Assistant - it's main state is the HVAC mode (off/auto/heat). If you want to get access to the `running_state` then you will have to query the climate entity's attributes or create a template sensor to expose this. You can see all the available attributes by looking at the climate entity in Developer Tools -> States in Home Assistant:

[image]

Here's an example of a binary sensor I created which looks at the `hvac_action` attribute of each TRV to signal whether any TRV is demanding heat. I then use this in an automation to turn on/off my boiler:

```
  - binary_sensor:
      - name: "Heating Demand"
        device_class: heat
        state: >
          {% set radiators = [
            states.climate.snug_radiator,
            states.climate.living_room_radiator_left,
            states.climate.living_room_radiator_right,
            states.climate.master_bedroom_radiator,
            states.climate.matilda_radiator,
            states.climate.landing_radiator,
            states.climate.jasper_radiator,
            states.climate.rupert_radiator

          ] %}
          {{ radiators | selectattr('attributes.hvac_action','eq','heating') | list | count > 0 }}
```

aussiebum71, 2023-11-07:
@photomoose, correct, @dafunkydan can create a value template sensor to extract the running_state. Example below

```
template:
  - sensor:
      - name: "TRV Heating State"
        state: "{{ state_attr('climate.living_room_trv_door', 'running_state') }}"
```

dafunkydan, 2023-11-08:
Guys you are super kind! I should have written more and/or precise, sorry.

Yes, my Problem is, that not all Values from Z2M are available in HomeAssistant:
[image]

Entity or Attribute doesn't matter, as long as it's available somewhere ;-) 
Comparing with @photomoose s Screenshot, there are some other Attributes missing (probably bc of Dev-Branch?), and some Attributes are individual Entities on my HA (e.g. Battery, Child Lock,...)
I am on Z2M 1.33.2, HA 2023.10.5, TRVZB is on FW 1.1.1. 
It still feels like i am just stupidly overseeing a basic Setting or st...

photomoose, 2023-11-08:
**Weekly Schedule Functionality**

I've been down a bit of a rabbit hole with this one...

Firstly, I've been unable to get Z2M frontend to form the correct payload when configuring the schedule. @Koenkk I have been using Z2M with v0.6.142 of the zigbee2mqtt-frontend package and I've also been using Z2M v1.33.2-1 as a HA Add-on and experience the same issue. When focus changes on the time textboxes, the partially complete payload is sent down the websocket and gets rejected (e.g. the hour is sent without the minute or vice-versa). Not sure if I'm doing something wrong here...

[image]

I did, however, manage to successfully send a schedule to the TRV by publishing the correct payload to the MQTT topic (e.g. zigbee2mqtt/[TRV-device]/set) instead of using the UI.

The second problem I encountered was that setting a schedule for today (Wednesday) had no effect on the TRV, but setting a schedule for Monday took effect today, so the schedule from Z2M was incorrect by 2 days. I sniffed the Zigbee packets for the `Set Weekly Schedule` command that was sent from both Z2M and the Sonoff Bridge and they both looked correct and comparable with each other. A bit more digging, and I discovered a discrepancy with the `Read Attributes Response` sent by Z2M and the Sonoff Bridge whenever the TRV asks the coordinator to read the `Time` cluster.

From the[ ZCL spec:](https://zigbeealliance.org/wp-content/uploads/2019/12/07-5123-06-zigbee-cluster-library-specification.pdf)

>[The Time] cluster provides a basic interface to a real-time clock. The clock time MAY be read and also written, in
order to synchronize the clock (as close as practical) to a time standard. This time standard is the number of
seconds since 0 hrs 0 mins 0 sec on 1st January 2000 UTC (Universal Coordinated Time). 

The response from Z2M correctly shows the current time and is[ indeed the number of seconds since 1st Jan 2000](https://github.com/Koenkk/zigbee-herdsman/blob/81ec998d2b5493564c5633f2f9f1e872a0784fed/src/controller/model/device.ts#L230):
```
ZigBee Cluster Library Frame, Command: Read Attributes Response, Seq: 2
    Frame Control Field: Profile-wide (0x18)
    Sequence Number: 2
    Command: Read Attributes Response (0x01)
    Status Record, UTC
        Attribute: Time (0x0000)
        Status: Success (0x00)
        Data Type: UTC Time (0xe2)
        UTC: Nov  8, 2023 14:09:17.000000000 GMT (0x2CDE530D)
    Status Record, Uint32: 752767757
        Attribute: Local Time (0x0007)
        Status: Success (0x00)
        Data Type: 32-Bit Unsigned Integer (0x23)
        Uint32: 752767757 (0x2cde530d)
```

The response from the Sonoff Bridge, however is 30 years in the future, suggesting that Sonoff has incorrectly implemented the ZCL spec by using a date 30 years prior to 1st Jan 2000 - i.e. the Unix Epoch, 1st Jan 1970:

```
ZigBee Cluster Library Frame, Command: Read Attributes Response, Seq: 0
    Frame Control Field: Profile-wide (0x18)
    Sequence Number: 0
    Command: Read Attributes Response (0x01)
    Status Record, UTC
        Attribute: Time (0x0000)
        Status: Success (0x00)
        Data Type: UTC Time (0xe2)
        UTC: Nov  7, 2053 14:20:04.000000000 GMT (0x654B9914)
    Status Record, Uint32: 1699456768
        Attribute: Local Time (0x0007)
        Status: Success (0x00)
        Data Type: 32-Bit Unsigned Integer (0x23)
        Uint32: 1699456768 (0x654ba700)
```

In order for the Sonoff TRVs to work correctly with the Sonoff Bridge, the TRVs must, therefore, also be implemented to incorrectly expect the time response to be number of seconds since the Unix Epoch. If I take the response from Z2M above _(UTC: Nov  8, 2023 14:09:17.000000000 GMT (0x2CDE530D))_ and assume that to be a Unix timestamp, I get a date exactly 30 years prior to today: 8th November 1993 - this date happened to fall on a *Monday*, which would explain why setting a schedule for Monday in Z2M worked today whereas setting a schedule for Wednesday did not.

@Koenkk have you seen this time issue before? The issue must apply to all Sonoff devices... I doubt Sonoff would fix the bug on their side (they would need to issue firmware updates for every device that depends on time from the Sonoff Bridge, and then existing user configuration for schedules would be messed up). Alternatively, we could add functionality for zigbee-herdsman to emulate the Sonoff Bridge behaviour for the Time cluster, however I'm not sure whether the device type is known at that level in the code. I guess another question is whether anyone specifically wants the ability to program schedules directly from Z2M....if not, then I guess we can ignore it and choose not to expose it.

photomoose, 2023-11-09:
@Koenkk I've thought of another possible workaround for the weekly schedule issue... We know that the Sonoff time implementation is 30 years out - i.e. it thinks the Z2M current time is 946684800 seconds earlier - or 10957 days earlier. Subtracting 10957 days from the current day results in a day of a week two days earlier. This means we could do a somewhat _dirty_ hack in the Z2M frontend and create a custom expose for the Sonoff device that maps the days as follows, without having to touch the Z2M implementation for the time cluster:

| Z2M Day | ZCL Payload Day |
|-|-|
| Sunday | Friday |
| Monday | Saturday |
| Tuesday | Sunday |
| Wednesday | Monday |
| Wednesday | Tuesday |
| Wednesday | Wednesday |
| Wednesday | Thursday |

It'd technically be wrong if you look at the packets being sent to the device, but in this case two wrongs would make a right.

What are your thoughts?

photomoose, 2023-11-09:
@dafunkydan not sure what's causing your missing attributes, these were available to me before I started this current work. I'm currently on HA 2023.11 and see them all fine, although I am sure they were present in earlier versions. It might be worth trying to delete the device from Z2M and re-pair it again, to see if the "pairing handshake" makes it come back.

dafunkydan, 2023-11-09:
After digging a bit deeper, i found that all states arrive in Mosquitto. 
I found that Issue https://github.com/home-assistant/core/issues/103414#issuecomment-1793848466 with a similar Behavior, mentioning that its a Z2M Problem, beeing fixed in 1.33.3. So that is supposedly the Culprit. Think im gonna wait for the next Update then...

On your elaborations @photomoose i can't contribute anything. This is like magic for me. Really astonshing what you are doing there, thank you so much!

Koenkk, 2023-11-09:
- You can use [this example](https://github.com/Koenkk/zigbee-herdsman-converters/blob/5b5532ea2734ee607ba7ba3b2da892a7ca5bc9e5/src/devices/xiaomi.ts#L2163) to send a custom time response
- I fixed the frontend schedule issue. Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

photomoose, 2023-11-09:
Perfect! I'll get that in tonight/tomorrow then! 👍

photomoose, 2023-11-13:
Sonoff have helped me identify further attributes available in their manufacturer-specific cluster. Most of these are diagnostic, but I have seen some of them in my traces and will expose them in the next day or two:

- Fault Code
- Rapid Heating Temperature (not used, will ignore)
- Number of Idle Steps - _number of steps used for calibration (number of no-load steps)_
- Number of Closing Steps - _number of steps it takes for the valve to close_
- Valve Opening Limit Voltage - _valve opening limit position voltage value_
- Valve Closing Limit Voltage - _valve closing limit position voltage value_
- Valve Motor Running Voltage - _valve motor operating voltage value_
- Manual Mode Temperature - _set temperature in manual mode_ (I assume this is the "saved" temperature of manual mode, for when you switch between auto and back to manual, it knows what value to go back to)
- Auto Mode Temperature - _set temperature in automatic mode_ (as above, I presume)

TomaszDom, 2023-11-15:
in the current stable version of z2m, does anyone else experience this:

1) system mode heat, set heating setpoint > local temperature, running state enters heat until heating setpoint < local temperature, goes to idle
2) next time heating setpoint > local temperature, running state is always idle, never opens the valve until I mess with the temperature again, effectively making the trv useless

photomoose, 2023-11-15:
@TomaszDom I think the valves have some leniency in them to prevent them continuously switching on/off as soon as the setpoint is unmet/met. I don't know what the figure is, but could be 1C either side or something.

My TRVs work perfectly - and sure enough the local temperature is just under the setpoint right now. If it drops more, the TRV will eventually come on.

[image]

TomaszDom, 2023-11-15:
I suppose I've gotten used to the behaviour of TS0601_thermostat trvs, which I am in the process of replacing with this sonoff, it looks like the valve is either fully on or fully off, maybe the new version of z2m will expose a PID-controlled system mode.

photomoose, 2023-11-15:
The Sonoff documentation is quite vague on how this TRV functions. I think you're right in saying the valve is either fully on or fully off - I'm not aware otherwise. I have exposed some further [rather useless] attributes about the valve [here.](https://github.com/Koenkk/zigbee-herdsman-converters/pull/6469)

I assume there must be some sort of PID internally - I have exposed a temperature calibration setting allowing you to offset the setpoint by a defined value, but there is no way to actually control the physical position of the valve.

I have exposed the majority of the attributes available (as witnessed in my traces and from what Sonoff has told me).

Schedule functionality probably needs some work, I'm still looking into the complications here.

TomaszDom, 2023-11-16:
@photomoose earlier you said that sonoff helped you figure some things out, which is rather unprecedented, maybe you could also ask them about on/off or pid control?

pawol, 2023-11-16:
I am almost sure this is next rebrand of GTZ06 from id3.pl aka TRV601 aka _TZE200_z1tyspqw. Look there.

photomoose, 2023-11-20:
@pawol it looks very similar; I will take a look at the existing code and see whether I can send the same payloads to my Sonoff TRVs.

sobrarbe, 2023-11-21:
Hi, I just received my sonoff tubes and I just integrated them with zigbee2mqtt 1.33.2-1 perfectly.  But I see that the parameters that you mentioned about child lock, open window and valve calibration are missing.  Do you know if this data will be added in the next zigbee2mqtt updates?  Thank you.

photomoose, 2023-11-21:
> Hi, I just received my sonoff tubes and I just integrated them with zigbee2mqtt 1.33.2-1 perfectly. But I see that the parameters that you mentioned about child lock, open window and valve calibration are missing. Do you know if this data will be added in the next zigbee2mqtt updates? Thank you.

@sobrarbe Child Lock, Open Window and Temperature Calibration are available in v15.109.0 of zigbee-herdsman-converters; the current release of Z2M uses v15.106.0. The next official release of Z2M should incorporate this functionality.

The features are available now if you're prepared to switch to the dev branch. If you're using Z2M as an add-on in Home Assitant, then you can install the Edge version to get the latest features (see https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html).

sobrarbe, 2023-11-22:
@photomoose Thank you very much for your prompt response.  As you say and I understand.  In the next update of zigbee2mqtt will the improvements that you mention about being able to calibrate temperature and the child and open window locking options be included?  If so, I will wait since the option you mentioned is in the zigbee2mqtt development branch. I don't dare to, being a newbie.

photomoose, 2023-11-22:
@sobrarbe yes, they will be in the next release - and it looks like a new release of zigbee2mqtt is made available at the beginning of each month, so you've probably only got about a week left to wait.

sobrarbe, 2023-11-22:
@photomoose  Thank you very much for your great work and for helping with this project.  For newbies like me it is still a complicated topic.  That's why your great work helps us.  Furthermore, the Sonoff valves are very beautiful and it would be a shame if it were not for your help that they would not be able to get the most out of them.  I thank you again.

mklooss, 2023-11-23:
@photomoose thank for your work :) does the TRVZB does not support partial opening valve or is this a lack of informations about the trv?
Sometimes my Gas boiler runs in an security stop, bcz the return temperatur. with normal trv/manual i did not run into this state :/

photomoose, 2023-11-23:
@mklooss I'm not aware of it supporting partial opening; I haven't seen any sort of feature in the eWeLink app. The closest thing I have found are valve opening/closing voltages and the number of steps to close the valve, which will be exposed when I complete #6469 (based on information sent to me from Sonoff). These attributes appear to be read-only. I have no idea how useful they are.

I will have another play to see if I can send it some commands similar to other TRVs.

jayme-github, 2023-11-30:
Did you had any success setting local_temperature_calibration? For me it does not seem to make a difference switching from -4 to 4 for example. The reported local_temperature does not change (e.g. the offset is not applied there) and the TRV does not switch to heating even though it should (according to external thermometers).

jarrodstrachan96, 2023-12-01:
@photomoose Hi, thanks for your great work exposing these settings in Z2QMTT. Are you aware of any way on this TRV to prevent "OFF" system mode? The problem arises when combining with Better Thermostat, that if the TRV's manual control dial is used to turn the TRV to OFF, that the preset temperatures within Better Thermostat will no longer work, as the device state is OFF.

If you or anybody has any suggestions I'd appreciate it, thanks for the great work.

jchidley, 2023-12-02:
> Did you had any success setting local_temperature_calibration? For me it does not seem to make a difference switching from -4 to 4 for example. The reported local_temperature does not change (e.g. [...]

I just tried this on a couple of my TRVZBs by changing the offset to 10 and then changing it back to 0. One TRVZB responded correctly but the other added 10 but when I set the offset back to 0, it used a -10 offset (i.e. a 20 swing).

SjoerdSterk, 2023-12-02:
> Did you had any success setting local_temperature_calibration? For me it does not seem to make a difference switching from -4 to 4 for example. The reported local_temperature does not change (e.g. [...]

I decided to skip the calibration altogether, and just feed it (on change) the delta between the external temperature and the desired temperature:

```
{% set temp_current = states('sensor.climate_indoors_bme280_temperature') | float %}
{% set temp_target = states('input_number.livingroom_target_temperature') | float %}
{% set temp_delta = (temp_target - temp_current) | round(1) %}
{% set temp_trv = states('sensor.sonoff_trv_current_temperature2') | float %}
{% set temp_trv_target = (temp_trv + temp_delta) | round(1, 'half', 0) %}

Current temperature: {{ temp_current }}
Target temperature:  {{ temp_target }}
Temperature delta:   {{ temp_delta }}
TRV temperature:     {{ temp_trv }}
TRV target:          {{ temp_trv_target }}
```

photomoose, 2023-12-02:
@jayme-github the calibration _should_ adjust the local temperature, as shown in the screen recording below that I captured a few minutes ago... However, you will notice that when I return the calibration to 0C, the device wrongly reports a local temperature of 9.1C - refreshing the value does not seem to take affect. It's only when I make a few more adjustments to the calibration temperature that the local temperature reverts to being reported correctly.

This might explain why my radiators occasionally seem to be always on when I use Better Thermostat. I will get my sniffer out and investigate what Zigbee messages are being sent between the TRV and Z2M, and will report back here shortly.

https://github.com/Koenkk/zigbee2mqtt/assets/1983638/7e7427e3-2bf8-4e2c-9b93-60cbdc4a43f8

TomaszDom, 2023-12-02:
Damn it, I bought them specifically to use with better thermostat and
because they were marketed specifically to work great with z2m. Hopefully
there will be over the air updates that fix this. I'm tired of creating
automations with workarounds that never seem to work consistently. Will OTA
via z2m be even possible or do we have to buy their bridge?

dktrekkie, 2023-12-02:
And BTW: The only way I can get it to behave again, is to factory reset the TRV by taking out the battery and holding the button. But afterwards, setting the calibration will eventually fail again.

SjoerdSterk, 2023-12-02:
> And BTW: The only way I can get it to behave again, is to factory reset the TRV by taking out the battery and holding the button. But afterwards, setting the calibration will eventually fail again.

Same here, with 1 out of 1 TRV's. Which is why I decided to circumvent calibrating and be done with it.

jarrodstrachan96, 2023-12-02:
I've grown tired of my current approach of syncing Zigbee TRVs to an external temp sensor. I've tried a few ways to do this, from various automation blueprints to using Better Thermostat integration. The problem is it's not entirely accurate or reliable, whether using TRV local temperature calibration or TRV Target temperature based calibration. 

SO, I've thought about a different approach:

1 Template Switch, we will call "Virtual TRV switch"
Switch_On will set the target temperature of the ZIgbee TRV to 35c (the max), to fully open the valve.
Switch_Off will set the target temperature of the Zigbee TRV to 5c (to close the valve)
Effectively this switch will serve as a binary OPEN or CLOSED for my TRV

1 Generic Thermostat entity
The thermostat heater attribute is set to the Virtual TRV Switch
The temperature attribute is set to my external temperature sensor.

If I've overlooked anything or if anybody has a better idea please let me know. But in practice for the last 12 hours it seems to be working flawlessly.

photomoose, 2023-12-02:
@dktrekkie yes, I have seen that happen on one of my TRVs before - I assumed it was either something I had done as a result of sending unsupported Zigbee messages to the device when I was testing or the fact that the TRV cover sometimes holds down the black button under the display, which also seems to cause weird behaviour/freezing.

I'll see if I can reproduce the issue and watch the Zigbee messages going over the radio network...

jchidley, 2023-12-02:
> And BTW: The only way I can get it to behave again, is to factory reset the TRV by taking out the battery and holding the button. But afterwards, setting the calibration will eventually fail again.

I got mine to take a 0 offset correctly by narrowing the offset range (10 to 0 then 2 to 0 then 1 to 0 then 0.5 to 0).

dktrekkie, 2023-12-02:
> I've grown tired of my current approach of syncing Zigbee TRVs to an external temp sensor. I've tried a few ways to do this, from various automation blueprints to using Better Thermostat integration. The problem is it's not entirely accurate or reliable, whether using TRV local temperature calibration or TRV Target temperature based calibration.
> 
> SO, I've thought about a different approach:
> 
> 1 Template Switch, we will call "Virtual TRV switch" Switch_On will set the target temperature of the ZIgbee TRV to 35c (the max), to fully open the valve. Switch_Off will set the target temperature of the Zigbee TRV to 5c (to close the valve) Effectively this switch will serve as a binary OPEN (35c target) or CLOSED (5c target) for my TRV
> 
> 1 Generic Thermostat entity The thermostat heater attribute is set to the Virtual TRV Switch The temperature attribute is set to my external temperature sensor.
> 
> If I've overlooked anything or if anybody has a better idea please let me know. But in practice for the last 12 hours it seems to be working flawlessly.

Cool. I guess that means that you only use the Home Assistant interface or whatever, to set the temperature or turn on the TRV? What about family members who just want to turn on the TRV - on device? Do you have some kind of automation to handle that then, or?

jarrodstrachan96, 2023-12-02:
> Cool. I guess that means that you only use the Home Assistant interface or whatever, to set the temperature or turn on the TRV? What about family members who just want to turn on the TRV - on [...]

Yep that's the trade off, the TRVs themself are childlocked. There's a wall mounted tablet to control them, or Google speakers in each room. For my use case it makes sense, BUT if I could think of a sensible way to reintroduce the TRV's controls I would

jarrodstrachan96, 2023-12-03:
> Cool. I guess that means that you only use the Home Assistant interface or whatever, to set the temperature or turn on the TRV? What about family members who just want to turn on the TRV - on device? Do you have some kind of automation to handle that then, or?

Your comment actually made me wonder whether there was a solution to introduce the TRV controls. What I've done is working, a bit clunky behind the scenes but works great so far.

1 Automation
The automation simply serves to syncronise the target temperature between the TRV and the Home Assistant Climate entity.

1 Template Switch, we will call "Virtual TRV switch"
Switch_On will publish a MQTT topic to set the the local temperature calibration of the TRV to -12
Switch_Off will publish a MQTT topic to set the the local temperature calibration of the TRV to +12
Effectively this switch (in conunction with the automation) will serve as a binary OPEN (35c target) or CLOSED (5c target) for my TRV

1 Generic Thermostat entity
The thermostat heater attribute is set to the Virtual TRV Switch
The temperature attribute is set to my external temperature sensor.

This retains the TRV's dial control, but also means 100% accurate and reliable remote control of the TRV using external temp sensors. Thought I'd share incase anybody has the same annoyances of trying to utilise Zigbee TRV's with external sensors.

EgerUS, 2023-12-04:
Hm, so the temperature calibration is behaving a bit weird.
I have 3 TRVs and it's similar in all of them.
Sometimes the temperature jumps even in the larger differences.

https://github.com/Koenkk/zigbee2mqtt/assets/3864110/623be7f0-d6f4-4b2e-a072-94b189862bd0

photomoose, 2023-12-04:
@EgerUS I'll look into this tonight and confirm whether it's an issue with the TRV itself or Z2M.

photomoose, 2023-12-04:
I've been looking at the traces and the issue with calibration does appear to be an issue with the TRV itself. Occasionally the TRV reports back the wrong local temperature after making a change to the calibration. I have captured an example of the traces in the video below - it happens at the 01:10 mark, when the local temperature jumps from 16.3C to 36.3C after changing the calibration from -5C to +5C. I'll see if I can reproduce it on the Sonoff Bridge tomorrow instead of Z2M.

https://github.com/Koenkk/zigbee2mqtt/assets/1983638/26b509a5-c2df-431f-af3c-8ef741acf0bd

photomoose, 2023-12-04:
> @photomoose Great job, but I have one question - why are not all values published as climate entity attributes, but as separate sensors, see e.g. local temperature calibration ? ![image](https://private-user-images.githubusercontent.com/3864110/287668449-9cd98021-2954-4f14-b824-7f156e21f66d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDE3MzI1MDgsIm5iZiI6MTcwMTczMjIwOCwicGF0aCI6Ii8zODY0MTEwLzI4NzY2ODQ0OS05Y2Q5ODAyMS0yOTU0LTRmMTQtYjgyNC03ZjE1NmUyMWY2NmQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTIwNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMDRUMjMyMzI4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZTdiZGRiMGUyZjBkODQ0ODkwMzg5ODY4NGY3NjQ3Yjc0ODNhMjRmNGQwYjcyYTVlOWNiYzFmZGI5NGNkYjM1MSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.ZD2K6lupajf7Uhx0IFApzsk2O85d9FkAmzz0dce70Zk)

@EgerUS I'm using existing Z2M functionality to expose the temperature calibration as a separate sensor. I believe the reason why it is a separate sensor and not a Climate Entity attribute is because the Climate Entity attribute does not support this attribute. I cannot see it listed in the official HA docs: https://developers.home-assistant.io/docs/core/entity/climate/

photomoose, 2023-12-05:
I managed to reproduce the temperature calibration issue using the official eWeLink app and the Sonoff Bridge...it looks like this is definitely an issue with the TRV firmware and not a Z2M problem. I will raise this directly with Sonoff.

The correct local temperature in this case was about 20.1C - but after changing the temperature calibration in the eWeLink app several times (from negative to positive to negative etc.), I was able to get the TRV to report the local temperature as a toasty 41.1C after setting the calibration to be +7C. The local temperature should have been reported as 27.1C, so it was out by 14C.

My suspicion is that the TRV might be receiving/processing the LocalTemperatureCalibration message multiple times in quick succession (perhaps as a result of repeaters/routers in my Zigbee network) and each message is being compounded to the local temperature.

[image]

[image]

sobrarbe, 2023-12-07:
Hello, do you know if the problem you mention regarding the regulation of temperature calibration can be solved?  I am planning to buy these valves because they are very well priced but I am afraid that they will not work well.  Thanks above all to the colleagues who are working on this project for its smooth functioning.

[186 more replies]

closed 2024-03-06 by Koenkk