Koenkk/zigbee2mqtt #19271: Zigbee2MQTT Docker Won't Start with External Converters

bobobado, 2023-10-12:
### What happened?

I have Zigbee2MQTT running on a Raspberry Pi 3 B+ with a Sonoff USB Dongle (type E) built with docker compose. Most of my Hue devices are working great. I recently purchased a Tuya Presence Sensor https://www.aliexpress.us/item/3256804518651916.html?spm=a2g0o.order_list.order_list_main.5.6e8b1802onnWl4&gatewayAdapt=glo2usa and have been attempting to add it to my instance via the external converters method. I can pair the device but if I add "external_converters" manually to my configuration.yaml file or via the UI, Zigbee2MQTT will no longer start. I've also tried adding just "external_converters" with no .js file specified and it will not start up either. Once that component is removed from my config, all works as expected again. The logs show this has something to do with the zigbee-herdsman-converters. Logs are below. Any help is greatly appreciated. 

/app/node_modules/zigbee-herdsman-converters/index.js:79
        if (converter.options) {
                      ^
TypeError: Cannot read properties of undefined (reading 'options')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:79:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)
    at Timeout.restart [as _onTimeout] (/app/index.js:17:5)
Using '/app/data' as data directory

/app/node_modules/zigbee-herdsman-converters/index.js:79
        if (converter.options) {
                      ^
TypeError: Cannot read properties of undefined (reading 'options')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:79:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)
Using '/app/data' as data directory

/app/node_modules/zigbee-herdsman-converters/index.js:79
        if (converter.options) {
                      ^
TypeError: Cannot read properties of undefined (reading 'options')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:79:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)
Using '/app/data' as data directory

### What did you expect to happen?

The ability to add external_converters

### How to reproduce it (minimal and precise)

Add "external_converters" to my config

### Zigbee2MQTT version

1.33.1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

ITead Sonoff Zigbee 3.0 USB Dongle Plus V2 model "ZBDongle-E"

ericfri, 2023-12-16:
I'm having a similar issue, although I don't see anything in the logs. As soon as I add `external_converters` to my configuration file, zigbee2mqtt refuses to start. As soon as it's removed it starts again without issue.

OlegPitsenko, 2023-12-27:
Та же проблема

dvbit, 2024-04-01:
same issue. z2m running in docker (homeasssitant add-on): could it be related?

closed 2024-10-30 by github-actions[bot]