Koenkk/zigbee2mqtt #19275: [Feature request]: External converter for temperature and humidity sensor KOJIMA-THS-ZG-LCD / TS0601 - Manufacturer Zigbee_TZE200_dikkika5

do6pbln9l, 2023-10-12:
### Is your feature request related to a problem? Please describe

The temperature and humidity sensor (Model: KOJIMA-THS-ZG-LCD) works according to the protocol: Zigbee 3.0 IEEE 802.15, connects to zigbee2mqtt, but then writes that it is not supported.

### Describe the solution you'd like

A converter is required for the device: Model Zigbee
TS0601 - Manufacturer Zigbee_TZE200_dikkika5. 
The device is written on the box - Model: KOJIMA-THS-ZG-LCD.
The temperature and humidity sensor operates according to the protocol: Zigbee 3.0 IEEE 802.15.4

### Describe alternatives you've considered

I tried to follow the instructions [(zigbee2mqtt instruction),](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html) but I don't understand where to get the necessary converter. I do not have enough experience to write myself yet (((( Please help with the solution.

I tried to find the device in the [Zigbee2MQTT](https://github.com/Koenkk/zigbee-herdsman-converters/tree/master/src/devices) developer branch.

### Additional context

Warning 2023-10-12 19:33:34Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_dikkika5'
Warning 2023-10-12 19:33:34Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

do6pbln9l, 2023-10-12:
My working code for the external converter for temperature and humidity sensor **KOJIMA-THS-ZG-LCD / TS0601 - Manufacturer Zigbee_TZE200_dikkika5**.

[TuYa ZG-227ZL](https://www.zigbee2mqtt.io/devices/ZG-227ZL.html)  is taken as an example.

```python
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition =     {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_dikkika5']),
    model: 'TS0601',
    vendor: 'TuYa',
    description: 'Temperature & humidity LCD sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [e.temperature(), e.humidity(), tuya.exposes.temperatureUnit(), tuya.exposes.temperatureCalibration(),
        tuya.exposes.humidityCalibration(), e.battery()],
    whiteLabel: [
        tuya.whitelabel('TuYa', 'TS0601', 'Temperature and humidity sensor', ['_TZE200_dikkika5']),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'temperature', tuya.valueConverter.divideBy10],
            [2, 'humidity', tuya.valueConverter.raw],
            [4, 'battery', tuya.valueConverter.raw],
            [9, 'temperature_unit', tuya.valueConverter.temperatureUnit],
            [23, 'temperature_calibration', tuya.valueConverter.divideBy10],
            [24, 'humidity_calibration', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;
```

closed 2023-10-12 by do6pbln9l

Koenkk, 2023-10-13:
@do6pbln9l added, thanks!

do6pbln9l, 2023-10-13:
**@Koenkk** Hi, thanks. I am glad that my work came in handy.