Koenkk/zigbee2mqtt #19280: With 1.33.0 much higher CPU usage

maumi, 2023-10-12:
### What happened?

After the update to 1.33 the cpu usage of z2m increased noticable. 
I run z2m on a lxc node in alpine installed with tteck's installer. But in the end it is just the "official" apk package. 
Before 1.33 my cpu usage was nearly zero, so maye 0.3% cpu usage and now I have a cpu usage about 6%. 
i know it isn't much but I would like to know where this higher cpu load could come from. 

### Zigbee2MQTT version

1.33.0

### Adapter firmware version

20221226

### Adapter

Sonoff Dongle Plus

lux73, 2023-10-14:
can't second this

with 1.33.1 i have CPU Usage ~1.9% - same as before with my APU plain installed z2m @ Debian

maumi, 2023-10-14:
Good to know. I didn't expected it to be a general problem. But maybe someone of you can help me to get on the right track where the higher cpu usage come from

lux73, 2023-10-15:
sorry, i'm not familiar with your

"**I run z2m on a lxc node in alpine installed with tteck's installer. But in the end it is just the "official" apk package.**"

Setup

btw. AFAIK there is no official Package as you mentioned - the official Ways for Install & Update are described below

if you are using the [recommended Installation](https://www.zigbee2mqtt.io/guide/installation/01_linux.html#installing) & [recommended Updating](https://www.zigbee2mqtt.io/guide/installation/01_linux.html#for-later-update-zigbee2mqtt-to-the-latest-version) nothing has changed for me

all official Setups are descibed [here](https://www.zigbee2mqtt.io/guide/installation/)

maumi, 2023-10-17:
I switched back to debian and now problem seems to be solved.

closed 2023-10-17 by maumi