Koenkk/zigbee2mqtt #19281: Aqara / Xiaomi SSM-U01 unable to bind to battery switch

txxdk, 2023-10-12:
### What happened?

I have 4 of the [SSM-U01](https://www.zigbee2mqtt.io/devices/SSM-U01.html#xiaomi-ssm-u01) devices. The first one I have have successfully bin able to bind to an IKEA [E1743](https://www.zigbee2mqtt.io/devices/E1743.html#ikea-e1743) :
[image]
And I can see that first SSM-U01 device that works has and input cluster called genOnOff.:
[image]

The problem is that the last 3 SSM-U01 devices do no display the input cluster genOnOff. So I am not able to even try and bind it to a switch.
[image]
[image]
Here is how it looks when I try to bind the SSM-U01 device, the OnOff checkmark is not there.
[image]

The devices otherwise works fine. I can turn them on and off from the Exposes tab, and se power consumption. 

### What did you expect to happen?

I would hope that all devices could have the input cluster "genOnOff" , so that can bind it to a switch.

### How to reproduce it (minimal and precise)

I have tried to force delete the devices and read them to zigbee2mqtt., but it does not help.

### Zigbee2MQTT version

1.33.1

### Adapter firmware version

20221226

### Adapter

sonoff ZBDongle-E

### Debug log

[zigbee2mqtt.log](https://github.com/Koenkk/zigbee2mqtt/files/12887181/zigbee2mqtt.log)

acheles-pub, 2023-11-13:
Similar problem.
Cluster genOnOff disappeared after rejoining.

txxdk, 2024-01-19:
I have now updated Zigbee2MQTT version to:
1.35.1
Koordinator type
zStack3x0
Adapter firmware version to
20230507

The problem persist. for device: lumi.switch.n0agl1 /  Aqara single switch module T1 (with neutral) /  xiaomi SSM-U01 

I have now also tried the same condinator and SSM-U01 device in ZHA and there the bind function works.

txxdk, 2024-01-19:
In a last attempt, i enabled new device pairing in z2m and then but the zigbee device in to pain mode. without deleting it.
and now it works. 
I dont now if it was the z2m version change or the coordiantor version change, that did the trick. but it works now.

txxdk, 2024-01-20:
I retested with 5 devices. now two work and 3 still dos not work. Can there be some hardware differences between the zigbee devices?

closed 2024-08-19 by github-actions[bot]

sstyle, 2026-04-09:
Same issue. Any solution?