Koenkk/zigbee2mqtt #19282: Z2M fRONTEND playing up

cloudbr34k84, 2023-10-13:
### What happened?

A few times today, when i load the Z2m FRONTEND its rendered like below

[image]

### Zigbee2MQTT version

1.33.1-dev commit: afb83fc

### Adapter firmware version

20230922

### Adapter

Zigbee ESP32GW (Zigstar)

MerekW, 2023-10-13:
I also experienced this (large, non-scrollable front-end behavior) as well as a bad gateway message while accessing from the sidebar:
[image]

cloudbr34k84, 2023-10-13:
yep i got that screen as well. Pressing crtl f5 helps

briangunderson, 2023-10-13:
I was hoping someone besides me would have this problem too! Of course it happened to me while I was in the middle of migrating a bunch of devices from one zigbee environment to another so I assumed it was something I did! Guess not! Let me know if screenshots/logs would be helpful to anyone.

alexmil72, 2023-10-13:
It happened to me too, also the Zigbee devices are unresponsive. Tried to access the Z2M configuration page on HA and I got "The add-on seems to not be ready, it might still be starting. Do you want to try again?", tried to restart Z2M and HA still doesn't work. There are no errors in Z2M logs and HA.

Tntdruid, 2023-10-13:
The last HA update broke it.

msaario, 2023-10-13:
I confirm same issue after Home Assistant Core 2023.10.2 update.

alexmil72, 2023-10-13:
Fixed reverting back HA to 2023.10.1

kted, 2023-10-13:
Same thing happens with ESPhome, so the update changed something that affects more than one integration.

cloudbr34k84, 2023-10-13:
Awesome

CattDamon, 2023-10-13:
> Fixed reverting back HA to 2023.10.1

ha core update --version=2023.10.1       in case anyone didn't know the commands.

jvuotila, 2023-10-13:
Home Assistant Core 2023.10.2 update and same problems here.

MerekW, 2023-10-13:
**Appears to be resolved in HA Core 2023.10.3**

Tntdruid, 2023-10-13:
Works again in Home Assistant 2023.10.3

closed 2023-10-13 by Koenkk