Koenkk/zigbee2mqtt #19283: OTA update of Ubisys Power Switch S2 fails with maxContentLength error

adriaanvanrossum, 2023-10-13:
### What happened?

When I'm doing a OTA update on the [Ubisys Power Switch S2](https://www.ubisys.de/en/products/light-and-consumers/power-switch-s2/) I get this error:

[image]

> failed (maxContentLength size of -1 exceeded)

It might be related to this issue in Axios: https://github.com/axios/axios/issues/4806

### What did you expect to happen?

That the update would start.

### Zigbee2MQTT version

1.30.4

### Adapter firmware version

26580700

### Adapter

ConBee II

Koenkk, 2023-10-13:
Can you try with the latest z2m dev? It contains fixes related to Ubisys updates. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

m0wlheld, 2023-10-16:
Same for H1 and S1 devices with 1.33.1-dev commit: [81a8df9](https://github.com/Koenkk/zigbee2mqtt/commit/81a8df9)

adriaanvanrossum, 2023-10-16:
Thanks for checking, it's unfortunately not fixed for me as well with the latest-dev version: 1.33.1-dev commit: [81a8df9](https://github.com/Koenkk/zigbee2mqtt/commit/81a8df9).

Koenkk, 2023-10-16:
Can you provide the debug log of this?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

m0wlheld, 2023-10-17:
From the moment of hitting "OK" on update confirm dialog until the failure message appears:

```
Zigbee2MQTT:info  2023-10-17 14:45:58: Updating '0x001fee0000009838' to latest firmware
2023-10-17T12:45:58.534Z zigbee-herdsman:controller:endpoint Read 0x001fee0000009838/1 genBasic(["dateCode","swBuildId"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2023-10-17T12:45:58.536Z zigbee-herdsman:controller:endpoint Request Queue (0x001fee0000009838/1): send read request immediately (sendWhen=immediate)
2023-10-17T12:45:58.536Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x001fee0000009838:46657/1 (0,0,1)
2023-10-17T12:45:58.538Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":46657,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":34,"options":0,"radius":30,"len":7,"data":{"type":"Buffer","data":[16,24,0,6,0,0,64]}}
2023-10-17T12:45:58.539Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,17,36,1,65,182,1,1,0,0,34,0,30,7,16,24,0,6,0,0,64,182]
2023-10-17T12:45:58.551Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2023-10-17T12:45:58.551Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2023-10-17T12:45:58.551Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2023-10-17T12:45:58.552Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2023-10-17T12:45:58.553Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-10-17T12:45:58.556Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,34,228]
2023-10-17T12:45:58.556Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,34,228]
2023-10-17T12:45:58.556Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,34] - 228
2023-10-17T12:45:58.557Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":34}
2023-10-17T12:45:58.558Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-10-17T12:46:00.696Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,7,69,196,65,182,2,220,239,115,94,109]
2023-10-17T12:46:00.697Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,7,69,196,65,182,2,220,239,115,94,109]
2023-10-17T12:46:00.697Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 7 - 2 - 5 - 196 - [65,182,2,220,239,115,94] - 109
2023-10-17T12:46:00.698Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":46657,"relaycount":2,"relaylist":[61404,24179]}
2023-10-17T12:46:00.698Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-10-17T12:46:00.702Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,53,68,129,0,0,0,0,65,182,1,1,0,43,0,194,68,153,0,0,33,24,24,1,6,0,0,66,15,50,48,50,51,48,50,49,53,45,68,69,45,70,66,49,0,64,0,66,5,49,46,51,46,49,115,94,27,107]
2023-10-17T12:46:00.703Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,53,68,129,0,0,0,0,65,182,1,1,0,43,0,194,68,153,0,0,33,24,24,1,6,0,0,66,15,50,48,50,51,48,50,49,53,45,68,69,45,70,66,49,0,64,0,66,5,49,46,51,46,49,115,94,27,107]
2023-10-17T12:46:00.703Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 53 - 2 - 4 - 129 - [0,0,0,0,65,182,1,1,0,43,0,194,68,153,0,0,33,24,24,1,6,0,0,66,15,50,48,50,51,48,50,49,53,45,68,69,45,70,66,49,0,64,0,66,5,49,46,51,46,49,115,94,27] - 107
2023-10-17T12:46:00.705Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":0,"srcaddr":46657,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":43,"securityuse":0,"timestamp":10044610,"transseqnumber":0,"len":33,"data":{"type":"Buffer","data":[24,24,1,6,0,0,66,15,50,48,50,51,48,50,49,53,45,68,69,45,70,66,49,0,64,0,66,5,49,46,51,46,49]}}
2023-10-17T12:46:00.709Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":24,"manufacturerCode":null,"commandIdentifier":1},"Payload":[{"attrId":6,"status":0,"dataType":66,"attrData":"20230215-DE-FB1"},{"attrId":16384,"status":0,"dataType":66,"attrData":"1.3.1"}],"Command":{"ID":1,"name":"readRsp","parameters":[{"name":"attrId","type":33},{"name":"status","type":32},{"name":"dataType","type":32,"conditions":[{"type":"statusEquals","value":0}]},{"name":"attrData","type":1000,"conditions":[{"type":"statusEquals","value":0}]}]}},"address":46657,"endpoint":1,"linkquality":43,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
2023-10-17T12:46:00.729Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-10-17T12:46:00.738Z zigbee-herdsman:controller:endpoint CommandResponse 0x001fee0000009838/1 genOta.imageNotify({"payloadType":0,"queryJitter":100}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2023-10-17T12:46:00.739Z zigbee-herdsman:controller:endpoint Request Queue (0x001fee0000009838/1): send imageNotify request immediately (sendWhen=immediate)
[... cut ...]
2023-10-17T12:46:07.533Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,36,226]
2023-10-17T12:46:07.533Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,36,226]
2023-10-17T12:46:07.534Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,36] - 226
2023-10-17T12:46:07.534Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":36}
2023-10-17T12:46:07.534Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
```

m0wlheld, 2023-10-17:
Btw: Are the images downloaded from an external source? I am running PiHole DNS with Ad-filtering, maybe this could be a cause.

adriaanvanrossum, 2023-10-17:
Here you go:

```
Zigbee2MQTT:debug 2023-10-17 16:13:40: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/update' with data '{"id":"ubisys_s2_xxx","transaction":"rx5eh-1"}'
Zigbee2MQTT:info  2023-10-17 16:13:40: Updating 'ubisys_s2_xxx' to latest firmware
Zigbee2MQTT:debug 2023-10-17 16:13:40: Received Zigbee message from 'ubisys_s2_xxx', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20230925-DE-FB1","swBuildId":"2.2.1"}' from endpoint 1 with groupID null
Zigbee2MQTT:debug 2023-10-17 16:13:40: Updating to latest '0x001fee000000771b' (S2 (5502))
Zigbee2MQTT:debug 2023-10-17 16:13:40: Using endpoint '232'
Zigbee2MQTT:debug 2023-10-17 16:13:41: Received Zigbee message from 'ubisys_s2_xxx', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":35718162,"imageType":31539,"manufacturerCode":4338}' from endpoint 232 with groupID null
Zigbee2MQTT:debug 2023-10-17 16:13:41: Got OTA request '{"fieldControl":1,"manufacturerCode":4338,"imageType":31539,"fileVersion":35718162}'
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: got firmware page, status: 200, data.length: 14318
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: image found: 10F2-7B01-0000-0006-01110206-spo-fmd.ota.zigbee
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: image found: 10F2-7B01-0000-0006-0192020D-spo-fmd.ota1.zigbee
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: image found: 10F2-7B02-0000-0001-010F0206-spo-fms.ota.zigbee
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: image found: 10F2-7B02-0000-0001-0192020D-spo-fms.ota1.zigbee
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: image found: 10F2-7B02-0002-0007-010F0206-spo-fms.ota.zigbee
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: image found: 10F2-7B02-0002-0007-0192020D-spo-fms.ota1.zigbee
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: image found: 10F2-7B03-0000-0006-010E0206-spo-fms2.ota.zigbee
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: image found: 10F2-7B03-0000-0006-0191020D-spo-fms2.ota1.zigbee
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: image found: 10F2-7B04-0000-0007-01160206-spo-fmsh.ota.zigbee
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: image found: 10F2-7B04-0000-0007-0191020D-spo-fmsh.ota1.zigbee
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: image found: 10F2-7B05-0000-0004-010F0206-spo-rms.ota.zigbee
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: image found: 10F2-7B05-0000-0004-0191020D-spo-rms.ota1.zigbee
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: image found: 10F2-7B06-0000-0004-010E0206-spo-rms2.ota.zigbee
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: image found: 10F2-7B06-0000-0004-0191020D-spo-rms2.ota1.zigbee
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: image found: 10F2-7B07-0000-0004-01160206-spo-rmsh.ota.zigbee
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: image found: 10F2-7B07-0000-0004-0191020D-spo-rmsh.ota1.zigbee
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: image found: 10F2-7B08-0000-0004-01110206-spo-rmd.ota.zigbee
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: image found: 10F2-7B08-0000-0004-0192020D-spo-rmd.ota1.zigbee
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: image found: 10F2-7B09-0000-0004-01090206-spo-fmi4.ota.zigbee
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: image found: 10F2-7B09-0000-0004-0192020D-spo-fmi4.ota1.zigbee
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: image found: 10F2-7B0A-0000-0005-010A0206-m7b-r0.ota.zigbee
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: image found: 10F2-7B0A-0000-0005-0193020D-m7b-r0.ota1.zigbee
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: image found: 10F2-7B0B-0000-0001-01000206-m7b-h10.ota.zigbee
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: image found: 10F2-7B0B-0000-0001-01900210-m7b-h10.ota1.zigbee
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: image found: 10F2-7B0C-0000-0000-01000206-m7b-wd1.ota.zigbee
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: image found: 10F2-7B0D-0000-0001-01140240-m7b-h1.ota.zigbee
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: image found: 10F2-7B0D-0000-0001-011C0300-m7b-h1.ota.zigbee
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: image found: 10F2-7B0D-0000-0001-01200403-m7b-h1.ota.zigbee
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: image found: 10F2-7B0D-0000-0001-01400422-m7b-h1.ota.zigbee
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: image found: 10F2-7B11-0000-0001-00920210-m7b-ld6.ota.zigbee
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: image found: 10F2-7B11-0000-0001-00940240-m7b-q95.ota.zigbee
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: image found: 10F2-7B21-0000-0006-0193020E-spo-fmd.ota.zigbee
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: image found: 10F2-7B21-0000-0006-0194020E-spo-fmd.ota.zigbee
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: image found: 10F2-7B22-0000-0001-0193020D-spo-fms-rev0-1.ota.zigbee
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: image found: 10F2-7B22-0002-0007-0193020D-spo-fms.ota.zigbee
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: image found: 10F2-7B23-0000-0006-0192020D-spo-fms2.ota.zigbee
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: image found: 10F2-7B24-0000-0007-0192020D-spo-fmsh.ota.zigbee
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: image found: 10F2-7B25-0000-0004-0192020D-spo-rms.ota.zigbee
[... cut ...]
Zigbee2MQTT:debug 2023-10-17 16:13:41: AxiosError: maxContentLength size of -1 exceeded
    at IncomingMessage.handlerStreamAborted (/app/node_modules/axios/lib/adapters/http.js:544:23)
    at IncomingMessage.emit (node:events:517:28)
    at IncomingMessage._destroy (node:_http_incoming:224:10)
    at _destroy (node:internal/streams/destroy:109:10)
    at IncomingMessage.destroy (node:internal/streams/destroy:71:5)
    at Socket.socketCloseListener (node:_http_client:461:11)
    at Socket.emit (node:events:529:35)
    at TCP.<anonymous> (node:net:350:12)
```

adriaanvanrossum, 2023-10-17:
Maybe the URL just changed from http://fwu.ubisys.de/smarthome/OTA/release/10F2-7B33-0000-0006-02300422-spo-fms2.ota.zigbee to https://www.ubisys.de/wp-content/uploads/10F2-7B33-0000-0006-02300422-spo-fms2.ota.zigbee?

m0wlheld, 2023-10-17:
Just checked, the URL is not blocked by PiHole and the file length (145kB) is properly reported by HTTP header:

```bash
$ curl -I http://fwu.ubisys.de/smarthome/OTA/release/10F2-7B33-0000-0006-02300422-spo-fms2.ota.zigbee
HTTP/1.1 200 OK
Server: nginx/1.10.3 (Ubuntu)
Date: Tue, 17 Oct 2023 14:32:00 GMT
Content-Length: 148990
Connection: keep-alive
Last-Modified: Wed, 27 Sep 2023 12:09:24 GMT
ETag: "245fe-606560fe6c500"
Accept-Ranges: bytes
```

Koenkk, 2023-10-17:
In the z2m folder, can you try to edit `node_modules/zigbee-herdsman-converters/lib/ota/ubisys.js` and change `const firmwarePage = await axios.get(firmwareHtmlPageUrl);` to `const firmwarePage = await axios.get(firmwareHtmlPageUrl, {maxContentLength: 99999});`

If you are running in Docker/HA addon you can follow [these instructions](https://github.com/zigbee2mqtt/hassio-zigbee2mqtt#testing-changes-locally) to edit files locally.

m0wlheld, 2023-10-18:
No luck. The line now reads:

```javascript
const firmwarePage = await axios.get(firmwareHtmlPageUrl, {maxContentLength: 99999});
```

but still the same error.  1.33.1-dev commit: [3461f11](https://github.com/Koenkk/zigbee2mqtt/commit/3461f11)

m0wlheld, 2023-10-18:
Interesting side node:

* launching an OTA request via WebUI for Ubisys **H1** takes 6-10 seconds until error message appears
* launching an OTA request via WebUI for Ubisys **S1** immediately returns the error message

adriaanvanrossum, 2023-10-18:
This might fix it? https://github.com/louislam/uptime-kuma/commit/3e68cf2a1cfab8cb8cc7969c915fba5535c7d827

```diff
- const firmwarePage = await axios.get(firmwareHtmlPageUrl);
+ const firmwarePage = await axios.get(firmwareHtmlPageUrl, { headers: { "Accept-Encoding": "gzip, deflate" }, decompress: true });
```

Can test later.

adriaanvanrossum, 2023-10-18:
Looks like the firmwarePage is working fine, right? Because the logs show:

```
Zigbee2MQTT:debug 2023-10-17 16:13:41: OTA ubisys: image found ...
```

m0wlheld, 2023-10-18:
> This might fix it? [louislam/uptime-kuma@3e68cf2](https://github.com/louislam/uptime-kuma/commit/3e68cf2a1cfab8cb8cc7969c915fba5535c7d827)
> 
> ```diff
> - const firmwarePage = await axios.get(firmwareHtmlPageUrl);
> + const firmwarePage = await axios.get(firmwareHtmlPageUrl, { headers: { "Accept-Encoding": "gzip, deflate" }, decompress: true });
> ```
> 
> Can test later.

Nope, doesn't. Same error.

```
Zigbee2MQTT:info  2023-10-18 11:22:17: Updating '0x001fee0000009838' to latest firmware
Zigbee2MQTT:info  2023-10-18 11:22:20: MQTT publish: topic 'zigbee2mqtt/0x001fee0000009838', payload '{"battery":100,"linkquality":25,"local_temperature":26.5,"occupancy":true,"occupied_heating_setpoint":21,"pi_heating_demand":0,"running_mode":"off","system_mode":"heat","unoccupied_heating_setpoint":16,"update":{"installed_version":19989522,"latest_version":20972578,"state":"available"},"update_available":null,"vacation_mode":false,"voltage":3500}'
Zigbee2MQTT:info  2023-10-18 11:22:20: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"0x001fee0000009838"},"error":"Update of '0x001fee0000009838' failed (maxContentLength size of -1 exceeded)","status":"error","transaction":"uc8tv-2"}'
Zigbee2MQTT:error 2023-10-18 11:22:20: Update of '0x001fee0000009838' failed (maxContentLength size of -1 exceeded)
```

m0wlheld, 2023-10-18:
> In the z2m folder, can you try to edit `node_modules/zigbee-herdsman-converters/lib/ota/ubisys.js` and change `const firmwarePage = await axios.get(firmwareHtmlPageUrl);` to `const firmwarePage = [...]

Had a look at [Axios](https://github.com/axios/axios/blob/a48a63ad823fc20e5a6a705f05f09842ca49f48c/lib/adapters/http.js#L529) and setting maxContentLength to -1 disables the check. This obviously avoids the error, though I don't know if disabling maxContentLength checks is a good thing.

Firmware update for Unisys S1 is running now, will report once and if it finished.

m0wlheld, 2023-10-18:
Update suceeded, although there was a "timeout warning" during completing the update, I'll guess that was related to the device restarting.

Koenkk, 2023-10-18:
@m0wlheld thanks for the fix! Assuming this can be closed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-10-18 by Koenkk