Koenkk/zigbee2mqtt #19284: [New device support]: WiFi Human Presence Sensor (WZ-M100-W)

Ansuel, 2023-10-13:
### Link

https://it.aliexpress.com/item/1005005986490584.html?spm=a2g0o.order_list.order_list_main.5.2a6336968xBZqL&gatewayAdapt=glo2ita

### Database entry

{"id":6,"type":"Router","ieeeAddr":"0xa4c13876758d9e1a","nwkAddr":34039,"manufId":4417,"manufName":"_TZE204_e5m9c5hl","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":".��,i/��,i1��,i3��,i4��,i6��,i7��,iI��,i","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":74}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1697196752398,"defaultSendRequestWhen":"immediate"}

### Comments

Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE204_e5m9c5hl'

In theory it should be an alternative version of the ZY-M100 (as always scammy aliexpress with not so precise description at taking a product for another.)

I open the device and it does have the same receiver and the same arm microcontroller (hoping it's not all fake)

Burned695, 2023-10-16:
Hi,

Just got the same sensors. In one out of six boxes was a little slip to here that has some instructions: https://github.com/wzwenzhi/Wenzhi-ZigBee2mqtt/tree/main

There are two js files, the one you need is wenzhi_tuya_M100-230908.js

I modified my version slightly to make the presence senor true and false like my other sensors are.

Ansuel, 2023-10-16:
@Burned695 do you have problem with setting the value too?

Burned695, 2023-10-16:
> @Burned695 do you have problem with setting the value too?

No just following the instructions in the link I sent in my last comment and it worked perfectly

Ansuel, 2023-10-16:
No I mean with the zigbee2mqtt and setting values like sensitivity and max distance... it seems sometime value are not set and gets reset to default

Burned695, 2023-10-16:
It doesn't seem to like some values, like max distance seemed to go up in intervals of 0.1 but I couldn't set 2.2 but would take 2.15 instead

Ansuel, 2023-10-18:
@Burned695 took care of adding support for this so now we have to wait for a new release to come. Yor finding are correct and with the upstreamed version it goes in step of .15... It seems not even the OEM was aware of this limitation (as it's 100% a clone of the TuYa one rebranded) (TuYa one have .15 steps of settings ahahah)

closed 2023-10-18 by Ansuel

Adaaks, 2023-10-29:
> Hi,
>
> Just got the same sensors. In one out of six boxes was a little slip to here that has some instructions: https://github.com/wzwenzhi/Wenzhi-ZigBee2mqtt/tree/main
>
> There are two js files, the [...]

can you release this modified version down below as my presence sensor is not appearing in automations... thanks!