Koenkk/zigbee2mqtt #19286: 1.33 to 1.33.1 fails - Please commit your changes....

Busta999, 2023-10-13:
### What happened?

From 1.33 to 1.33.1 - update fails

git pull
Updating 7ee207fc..7e630391
error: Your local changes to the following files would be overwritten by merge:
	package-lock.json
Please commit your changes or stash them before you merge.
Aborting

### What did you expect to happen?

Update to occur

### How to reproduce it (minimal and precise)

Performed 

 git checkout HEAD -- package.json
 zigbee2mqtt % git pull                         
Updating 7ee207fc..7e630391
error: Your local changes to the following files would be overwritten by merge:
	package-lock.json
Please commit your changes or stash them before you merge.
Aborting

### Zigbee2MQTT version

1.33

### Adapter firmware version

20230507

### Adapter

zStack3x0

### Debug log

Debug logging did not capture anything after zigbee2MQTT was shut down

Koenkk, 2023-10-13:
Do a `git checkout HEAD -- package-lock.json`

Busta999, 2023-10-14:
Fixed it many thanks

closed 2023-10-14 by Busta999