Koenkk/zigbee2mqtt #19287: Zigbee adapter stop to communicate with devices

rdl987, 2023-10-13:
### What happened?

Hello,
I have the same problem but in my case it's regarding all devices, it became unresponsive.

I'm using zigbee2mqtt on docker running in a qnap nas and a SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P.

After a while I got the following error in the log:

`Publish 'set' 'state' to '0x.......' failed: 'Error: Command 0x........../1 genOnOff.on({},`

and every device stop to responde at every zigbee comand.

The only solution is to unplug e plug in again the zigbee adapter or reboot the nas.

Could anyone help me?
Thanks

### Zigbee2MQTT version

1.33.1

### Adapter firmware version

20221226

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

StefanGruell, 2023-10-13:
I have the same problem with my Nous A1Z, i use also the SONOFF Dongle. The Nous A1Z reports its data but i cant turn it off via the Z2M UI.

TEF2one, 2023-10-14:
Same problem with by SLZB-06 coordinateur a many devices connected no longer respondoing.

biokys, 2023-10-15:
Same problem with Deconz

rdl987, 2023-10-15:
But anyone has a solution?

StefanGruell, 2023-10-15:
> But anyone has a solution?

For me unplugging also didnt work, i had to remove all devices which were connected to my NOUS A1Z plug. Then remove the plug from the network and repair it in close proximity (for a good LQI). Then i was able to send commands to the plug again and reconnect all other sensors.

Shchepotin, 2023-10-16:
Same problem with SONOFF (Z2M 1.33.1)

rdl987, 2023-10-17:
I didn't understand if the problem is related only to docker or some problem with configuration between the container and the nas

Shchepotin, 2023-10-17:
@rdl987 also problem happen with Operating System installed to Raspberry PI

rdl987, 2023-10-19:
So, just to recap, the only installation that works is as addon to Home Assistant Operating System. Right?

Could please, anyone, explain the differences between the docker version and addon version?

Thank you

closed 2024-05-17 by github-actions[bot]