Koenkk/zigbee2mqtt #19292: [New device support]:

bratmat, 2023-10-13:
### Link

https://pl.aliexpress.com/i/1005005946786561.html

### Database entry

{"id":32,"type":"Router","ieeeAddr":"0xa4c13845235ed012","nwkAddr":24896,"manufId":4417,"manufName":"_TZE204_kyhbrfyl","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u001eV�,i\u001fV�,i","65506":56,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZE204_kyhbrfyl","zclVersion":3,"appVersion":74,"modelId":"TS0601","powerSource":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1440156621},"lastSeen":1697223398826,"defaultSendRequestWhen":"immediate"}

### Comments

I tried but it's  not working correctly - I'm a noob.
What is working:
Presence, battery level, radar sensitivity , fading time
not working target distance, approach distance and not know if settings works (detection range, motion sensivity)

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE204_kyhbrfyl']),
    model: 'RTX PRO',
    vendor: 'TuYa',
    description: 'Human presence sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, 
    configure: tuya.configureMagicPacket,
   
    exposes: [
        e.presence(), 
           
        exposes.enum('approach_distance', ea.STATE, ['none', 'small_movement', 'large_movement'])
            .withDescription('traffic of movement from the point of view of the radar'),
        exposes.numeric('fading_time', ea.STATE_SET).withValueMin(0).withValueMax(600).withValueStep(1)
            .withDescription('Sensitivity of the radar'),
        exposes.numeric('detection_range', ea.STATE_SET).withValueMin(150).withValueMax(600).withValueStep(10)
            .withDescription('far sensitivity of the radar'),
        exposes.numeric('radar_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(7).withValueStep(1)
            .withDescription('Sensitivity of the radar'),
        exposes.numeric('motion_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(7).withValueStep(1)
            .withDescription('Motionless sensitivity of the radar'),
        exposes.numeric('target_distance', ea.STATE).withDescription('Distance to target').withUnit('cm'),
        exposes.numeric('battery_level', ea.STATE).withDescription('battery level').withUnit('%'),
           
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'presence', tuya.valueConverter.trueFalse1],
            [11, 'approach_distance', tuya.valueConverterBasic.lookup({'none': 0, 'small_movement': 1, 'large_movement': 2})],
            [12, 'fading_time', tuya.valueConverter.raw],
            [13, 'detection_range', tuya.valueConverter.raw],
            [15, 'radar_sensitivity', tuya.valueConverter.raw],
            [16, 'motion_sensitivity', tuya.valueConverter.raw],
            [19, 'target_distance', tuya.valueConverter.raw],
            [101, 'battery_level', tuya.valueConverter.raw],
            
         
        ],
    },
};

module.exports = definition;
```

bratmat, 2023-10-28:
only presence sensor work good, no current distance , no aproach distance reporting

hlawati14, 2023-10-31:
Hi, 

I used the same convertor as yours, but got the following errors:

```
Oct 31 10:35:33 DUSUN npm[21785]: Zigbee2MQTT:debug 2023-10-31 10:35:33: Received Zigbee message from '0xa4c138fae2234ed7', type 'attributeReport', cluster 'genBasic', data '{"65503":"���,i���,i"}' from endpoint 1 with groupID 0
Oct 31 10:35:33 DUSUN npm[21785]: Zigbee2MQTT:warn  2023-10-31 10:35:33: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE204_kyhbrfyl'
Oct 31 10:35:33 DUSUN npm[21785]: Zigbee2MQTT:warn  2023-10-31 10:35:33: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
```

Any thoughts on how to make it work?

bratmat, 2023-11-02:
> Hi,
>
> I used the same convertor as yours, but got the following errors:
>
> ```
> Oct 31 10:35:33 DUSUN npm[21785]: Zigbee2MQTT:debug 2023-10-31 10:35:33: Received Zigbee message from [...]
Try this one - It works but no reporting and I don't know if settings works

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE204_kyhbrfyl']),
    model: 'RTX PRO',
    vendor: 'TuYa',
    description: 'Human presence sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, 
    configure: tuya.configureMagicPacket,
   
    exposes: [
        e.presence(), 
           
        exposes.enum('approach_distance', ea.STATE, ['none', 'small_movement', 'large_movement'])
            .withDescription('traffic of movement from the point of view of the radar'),
        exposes.numeric('detection_delay', ea.STATE_SET).withValueMin(0).withValueMax(600).withValueStep(1)
            .withDescription('detection delay'),
        exposes.numeric('detection_range', ea.STATE_SET).withValueMin(150).withValueMax(600).withValueStep(10)
            .withDescription('far sensitivity of the radar'),
        exposes.numeric('radar_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(7).withValueStep(1)
            .withDescription('Sensitivity of the radar'),
        exposes.numeric('motion_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(7).withValueStep(1)
            .withDescription('Motionless sensitivity of the radar'),
        exposes.numeric('current_distance', ea.STATE).withDescription('Distance to target').withUnit('cm'),
        exposes.numeric('battery_level', ea.STATE).withDescription('battery level').withUnit('%'),
           
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'presence', tuya.valueConverter.trueFalse1],
            [11, 'approach_distance', tuya.valueConverterBasic.lookup({'none': 0, 'small_movement': 1, 'large_movement': 2})],
            [12, 'detection_delay', tuya.valueConverter.raw],
            [13, 'detection_range', tuya.valueConverter.raw],
            [15, 'radar_sensitivity', tuya.valueConverter.raw],
            [16, 'motion_sensitivity', tuya.valueConverter.raw],
            [19, 'current_distance', tuya.valueConverter.raw],
            [101, 'battery_level', tuya.valueConverter.raw],
            
         
        ],
    },
};

module.exports = definition;`

bratmat, 2023-11-11:
Someone should check that converter:

[TZE204_kyhbrfyl.txt](https://github.com/Koenkk/zigbee2mqtt/files/13325954/TZE204_kyhbrfyl.txt)

seblu, 2023-11-12:
With the proposed converter, device is recognized, but there is no `Presence` or `Approach distance`.  `Target distance` is updating itself.

[image]

koriwi, 2023-11-16:
try this @seblu @hlawati14 
```js
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE204_kyhbrfyl']),
    model: 'RTX PRO',
    vendor: 'TuYa',
    description: 'Human presence sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: console.log, 
    // configure: tuya.configureMagicPacket,
   
    exposes: [
        e.presence(), 
           
        exposes.enum('movement_type', ea.STATE, ['none', 'small', 'large'])
            .withDescription('traffic of movement from the point of view of the radar'),
        exposes.numeric('detection_delay', ea.STATE_SET).withValueMin(0).withValueMax(600).withValueStep(1).withUnit('s')
            .withDescription('detection delay'),
        exposes.numeric('detection_range', ea.STATE_SET).withValueMin(150).withValueMax(600).withValueStep(10)
            .withDescription('far sensitivity of the radar'),
        exposes.numeric('radar_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(7).withValueStep(1)
            .withDescription('Sensitivity of the radar'),
        exposes.numeric('motion_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(7).withValueStep(1)
            .withDescription('Motionless sensitivity of the radar'),
        exposes.numeric('current_distance', ea.STATE).withDescription('Distance to target').withUnit('cm'),
        exposes.numeric('battery_level', ea.STATE).withDescription('battery level').withUnit('%'),
           
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'presence',tuya.valueConverter.trueFalse1],
            [11, 'movement_type', tuya.valueConverterBasic.lookup({'none': 0, 'small': 1, 'large': 2})],
            [12, 'detection_delay', tuya.valueConverter.raw],
            [13, 'detection_range', tuya.valueConverter.raw],
            [15, 'radar_sensitivity', tuya.valueConverter.raw],
            [16, 'motion_sensitivity', tuya.valueConverter.raw],
            [19, 'current_distance', tuya.valueConverter.raw],
            [101, 'battery_level', tuya.valueConverter.raw],
            
         
        ],
    },
};

module.exports = definition;
```

bratmat, 2023-11-16:
> With the proposed converter, device is recognized, but there is no `Presence` or `Approach distance`. `Target distance` is updating itself.
> 
You should completly remove that motion sensor from the list and find it again with a new converter. Mine is working fine but not excellent :-(.

koriwi, 2023-12-20:
@Koenkk ive been using this converter every day now and it works flawless for me.
@bratmat how is it going for you?

Koenkk, 2023-12-20:
Could you create a pull request to add this for out-of-the-box support?

seblu, 2023-12-20:
Just to let you know I have returned the device, so I can't help anymore on this one.

koriwi, 2023-12-23:
> Could you create a pull request to add this for out-of-the-box support?

will try my best in the next days, hope the kid lets me :P

VerodinDev, 2024-01-04:
Received this one today, eagerly awaiting build in support. 

Btw, after pairing I got completly spammed by not suppoprted device popups and a please see support new devices link. They just kept on coming and only stopped when I removed the device. Not so fun behaviour.

VerodinDev, 2024-01-04:
@koriwi Tryin out your converter now, but 1 thing I noticed right away is that presense and movement type are null. Current distance seems to work and its value make sense when moving in front of it. Same result as seblu got. I can do some limited testing if needed (limited as in time limited due to a 5 yo running around stealing pappas toys)

What I did was force remove the device, stop zigbee2mqtt docker container, add converter and start container again.

jdmc, 2024-02-23:
I recently experimented with this device by utilizing and modifying the approach mentioned above, where I employed a TuYa Hub to access the official Datapoints from the Tuya IoT Platform. I attempted to align the official Tuya keys with the keys accessible under the "zigbee-herdsman-converters/src/lib" directory. However, I am uncertain if these mappings are accurate or if there are more optimal alternatives available. For testing purposes, I currently have one device connected with the Smart Life App and another with Z2M, aiming to compare the outcomes of both setups.

Further:
Did some check on manufacture/model have updated the file as well.
Official Neo product website: https://www.szneo.com/en/products/show.php?id=277 
Product model: NAS-PS09B2

Product model matches with : https://www.zigbee2mqtt.io/supported-devices/#v=Neo

 If someone wants to have a look and try things out feel free so we can have this final as a new device support

Table with data from Tuya Iot and matching:

<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:w="urn:schemas-microsoft-com:office:word"
xmlns:m="http://schemas.microsoft.com/office/2004/12/omml"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Word.Document>
<meta name=Generator content="Microsoft Word 15">
<meta name=Originator content="Microsoft Word 15">
<link rel=File-List
href="file:///C:/Users/josed/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">

<link rel=themeData
href="file:///C:/Users/josed/AppData/Local/Temp/msohtmlclip1/01/clip_themedata.thmx">
<link rel=colorSchemeMapping
href="file:///C:/Users/josed/AppData/Local/Temp/msohtmlclip1/01/clip_colorschememapping.xml">

<style>

</style>

</head>

<body lang=EN-US style='tab-interval:36.0pt;word-wrap:break-word'>

DataPoints ID | Key TuYa | Status |   | Key Match Z2M   ? |  
-- | -- | -- | -- | -- | --
1 | presence_state | "none",   "presence" |   | **presence_state** Working correct   with original key |  
11 | human_motion_state | "none",   "small_move",   "large_move" |   | **human_motion_state**      Working correct   with original key |  
12 | presence_time | "unit":   "s",   "min":   3,   "max":   600,   "scale":   0,   "step":   1 |   | **departure_delay**       (detection_delay) |  
13 | motion_far_detection | "unit":   "cm",   "min":   150,   "max":   600,   "scale":   0,   "step":   75 | Motion distance   detection | **radar_range**       (detection_range) |  
15 | motion_sensitivity_value | "unit":   "",   "min":   0,   "max":   7,   "scale":   0,   "step":   1 |   | **radar_sensitivity** |  
16 | motionless_sensitivity | "unit":   "",   "min":   0,   "max":   7,   "scale":   0,   "step":   1 |   | **presence_sensitivity**       (motion_sensitivity) |  
19 | dis_current | "unit":   "cm",   "min":   0,   "max":   1000,   "scale":   0,   "step":   1 |  Distance to target | **dis_current**       Working correct   with original key |  
101 | battery_level |   |   | Using USB |  

</body>

</html>

Code:

````javascript
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const {} = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');

const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE204_kyhbrfyl']),
    model: 'NAS-PS09B2 ',
    vendor: 'Neo',
    description: 'Human presence sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: console.log,
   
    exposes: [   
        e.enum('presence_state', ea.STATE, ['none', 'presence']).withDescription('Presence State'),        
        exposes.enum('human_motion_state', ea.STATE, ['none', 'small', 'large']).withDescription('Human Motion State'),        
        exposes.numeric('departure_delay', ea.STATE_SET).withUnit('s').withValueMin(3).withValueMax(600).withValueStep(1).withDescription('Presence Time'),                
        exposes.numeric('radar_range', ea.STATE_SET).withUnit('cm').withValueMin(150).withValueMax(600).withValueStep(75).withDescription('Motion Range Detection'),        
        exposes.numeric('radar_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(7).withValueStep(1).withDescription('Motion Detection Sensitivity'),        
        exposes.numeric('presence_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(7).withValueStep(1).withDescription('Motionless Detection Sensitivity'),        
        exposes.numeric('dis_current', ea.STATE).withUnit('cm').withValueMin(0).withValueMax(1000).withValueStep(1).withDescription('Current Distance'),
        exposes.numeric('battery_level', ea.STATE).withUnit('%').withDescription('battery level'),
        
    ],
    meta: {
        tuyaDatapoints: [
            [1,  'presence_state',tuya.valueConverter.trueFalse1],
            [11, 'human_motion_state', tuya.valueConverterBasic.lookup({'none': 0, 'small': 1, 'large': 2})],
            [12, 'departure_delay', tuya.valueConverter.raw],
            [13, 'radar_range', tuya.valueConverter.raw],
            [15, 'radar_sensitivity', tuya.valueConverter.raw],
            [16, 'presence_sensitivity', tuya.valueConverter.raw],
            [19, 'dis_current', tuya.valueConverter.raw],
            [101,'battery_level', tuya.valueConverter.raw],         
        ],
    },
};

module.exports = definition;

```

nublemeduble, 2024-02-27:
Yes please support them

tomlut, 2024-03-04:
> If someone wants to have a look and try things out feel free so we can have this final as a new device support

Working perfectly for me. Thanks!

[image]

trinode, 2024-03-22:
Hey,

I made a few tweaks so it would show up correctly in HA, and it all appears to work fine:- 

* Swapped presence_state with the preset for occupancy (using the presence preset shows the sensor as "home" like it's an individual, occupancy shows as "Motion: Detected")
* Added a label for dis_current and made the description more description like

[image]

[image]

```js
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const {} = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');

const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE204_kyhbrfyl']),
    model: 'NAS-PS09B2 ',
    vendor: 'Neo',
    description: 'Human presence sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: console.log,
   
    exposes: [   
        e.occupancy(),
        exposes.enum('human_motion_state', ea.STATE, ['none', 'small', 'large']).withDescription('Human Motion State'),        
        exposes.numeric('departure_delay', ea.STATE_SET).withUnit('s').withValueMin(3).withValueMax(600).withValueStep(1).withDescription('Presence Time'),                
        exposes.numeric('radar_range', ea.STATE_SET).withUnit('cm').withValueMin(150).withValueMax(600).withValueStep(75).withDescription('Motion Range Detection'),        
        exposes.numeric('radar_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(7).withValueStep(1).withDescription('Motion Detection Sensitivity'),        
        exposes.numeric('presence_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(7).withValueStep(1).withDescription('Motionless Detection Sensitivity'),        
        exposes.numeric('current_distance', ea.STATE).withUnit('cm').withValueMin(0).withValueMax(1000).withValueStep(1).withDescription('Current Distance'),
    ],
    meta: {
        tuyaDatapoints: [
            [1,  'occupancy',tuya.valueConverter.trueFalse1],
            [11, 'human_motion_state', tuya.valueConverterBasic.lookup({'none': 0, 'small': 1, 'large': 2})],
            [12, 'departure_delay', tuya.valueConverter.raw],
            [13, 'radar_range', tuya.valueConverter.raw],
            [15, 'radar_sensitivity', tuya.valueConverter.raw],
            [16, 'presence_sensitivity', tuya.valueConverter.raw],
            [19, 'current_distance', tuya.valueConverter.raw],
        ],
    },
};
```

tomlut, 2024-03-23:
Nice one. Thank you.

shalak, 2024-05-18:
I'm having the `null` issue:
[image]

I was trying to add the converter mentioned by @trinode above, but it fails to load:

```
[2024-05-18 14:30:30] error: 	z2m: Failed to load external converter file 'Neo_NAS-PS09B2.js' (Cannot find module 'zigbee-herdsman-converters/lib/extend'
Require stack:
- /app/dist/util/utils.js
- /app/dist/util/settings.js
- /app/index.js)
```

I'm using Z2M add-on version 1.37.1-1

How can I solve this? I found [this thread](https://github.com/Koenkk/zigbee2mqtt/discussions/11554#discussioncomment-2229495), but I'm already on more recent version that mentioned there...

shalak, 2024-05-18:
uhh, please disregard the `null` issue above, I just found [this release notes](https://github.com/Koenkk/zigbee-herdsman-converters/pull/7256#issuecomment-2043196307):

> A quick fix is to just unplug the sensor and plug it back in and it'll send the correct states

It works OK now.

But the failure to import the converter is still valid :thinking:

closed 2025-01-04 by github-actions[bot]