Koenkk/zigbee2mqtt #19294: HA addon  shut down every some days with : MQTT error: getaddrinfo ENOTFOUND core-mosquitto

claustro, 2023-10-13:
### What happened?

Running the addon in last home operating system 10.5 it shut down by itself every 1 or 2 days 

### Zigbee2MQTT version

1.33.1-1 

### Adapter firmware version

20230507

### Adapter

SONOFF ZigBee 3.0 USB Dongle Plus, TI CC2652P   zStack3x0

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/12899747/log.txt)
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/12899749/log.txt)
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/12899753/log.txt)

ToncaCZ, 2024-01-03:
I have the same problem ....

Zigbee2MQTT

[09:57:19] INFO: Preparing to start...
[09:57:23] INFO: Socat not enabled
[09:57:35] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:error 2024-01-03 09:59:43: MQTT error: getaddrinfo EAI_AGAIN core-mosquitto
Zigbee2MQTT:error 2024-01-03 09:59:43: MQTT failed to connect, exiting...

RPi3 + Gateway SONOFF Dongle Plus-E ZigBee

lilp69, 2024-04-17:
Same problem here with all plugin and HA up to date and SonOff Dongle Plus

julianxv, 2024-04-23:
I've the same errors like @ToncaCZ since a week. Zigbee2mqtt crashes every half hour. I'm using up-to-date HA, Zigbee2Mqtt, Sonoff Dongle Plus and SiliconLabs Multiprotocol.

lilp69, 2024-05-03:
I've solved my problem by upgrading my hardware from RPI3 to RPI5. No more problem right now.

closed 2024-11-30 by github-actions[bot]