Koenkk/zigbee2mqtt #19296: Cannot set schedule on Moes TRV BRT100 (TS0601 _TZE200_kds0pmmv)

osmeest, 2023-10-14:
### What happened?

Going to the Zigbee2MQTT Web GUI, in the Exposes section, the schedule for every day is displayed. Above those displays, a day selector and a schedule string should allow to edit the schedule for each day. Unfortunately, selecting a day or modifying the schedule does not work. A red error popup is displayed with "Publish 'set' 'schedule' to 'Moes TV Jardin' failed: 'TypeError: Cannot read properties of undefined (reading 'toString')".
Trying to set the same corresponding state in HomeAssistant Developer Tools is not functional.

Device identification:
[image]

Schedule controls in Exposes page:
[image]

### What did you expect to happen?

Selecting a day should update the editable schedule string with corresponding daily schedule. Editing the schedule string should update the TRV and the displayed schedules underneath.

### How to reproduce it (minimal and precise)

- go to Zigbee2MQTT
- select TRV in Devices page
- select Exposes tab
- select a day in Schedule/day ==> error popup
- edit the schedule string and move focus away ==> same error popup

### Zigbee2MQTT version

1.33.1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus V2 model "ZBDongle-E"

### Debug log

info  2023-10-14 09:54:30: Checking if update available for 'Moes TRV Jardin'
info  2023-10-14 09:54:31: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Moes TRV Jardin"},"error":"Failed to check if update available for 'Moes TRV Jardin' (No image available for imageType '5634')","status":"error","transaction":"9xp3c-1"}'
error 2023-10-14 09:54:31: Failed to check if update available for 'Moes TRV Jardin' (No image available for imageType '5634')
info  2023-10-14 09:54:34: MQTT publish: topic 'zigbee2mqtt/Moes TRV Jardin', payload '{"battery_low":false,"boost_timeset_countdown":0,"child_lock":"UNLOCK","comfort_temperature":20,"current_heating_setpoint":20,"eco_temperature":17,"error_status":0,"frost_protection":"OFF","heating_stop":"OFF","holiday_start_stop":"2021/01/01 01:01 | 2021/01/01 01:01","holiday_temperature":10,"linkquality":160,"local_temperature":20.7,"local_temperature_calibration":1,"online":"ON","open_window":"OFF","open_window_temperature":15,"preset":"auto","schedule":{"schedule":"00:00/17 07:00/21 09:00/19 19:00/20 23:00/17 24:00/17","week_day":"sunday"},"schedule_friday":"00:00/17 06:00/21 08:30/19 19:00/20 23:00/17 24:00/17","schedule_monday":"00:00/17 06:00/21 08:30/19 19:00/20 23:00/17 24:00/17","schedule_saturday":"00:00/17 07:00/21 09:00/19 19:00/20 23:00/17 24:00/17","schedule_sunday":"00:00/17 07:00/21 09:00/19 19:00/20 23:00/17 24:00/17","schedule_thursday":"00:00/17 06:00/21 08:30/19 19:00/20 23:00/17 24:00/17","schedule_tuesday":"00:00/17 06:00/21 08:30/19 19:00/20 23:00/17 24:00/17","schedule_wednesday":"00:00/17 06:00/21 08:30/19 19:00/20 23:00/17 24:00/17","system_mode":"heat","update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":null,"working_day":"mon_fri+sat+sun"}'
error 2023-10-14 09:55:11: Publish 'set' 'schedule' to 'Moes TRV Jardin' failed: 'TypeError: Cannot read properties of undefined (reading 'toString')'
info  2023-10-14 09:55:39: MQTT publish: topic 'zigbee2mqtt/Moes TRV Jardin', payload '{"battery_low":false,"boost_timeset_countdown":0,"child_lock":"UNLOCK","comfort_temperature":20,"current_heating_setpoint":20,"eco_temperature":17,"error_status":0,"frost_protection":"OFF","heating_stop":"OFF","holiday_start_stop":"2021/01/01 01:01 | 2021/01/01 01:01","holiday_temperature":10,"linkquality":160,"local_temperature":20.7,"local_temperature_calibration":1,"online":"ON","open_window":"OFF","open_window_temperature":15,"preset":"auto","schedule":{"schedule":"00:00/17 07:00/21 09:00/19 19:00/20 23:00/17 24:00/17","week_day":"sunday"},"schedule_friday":"00:00/17 06:00/21 08:30/19 19:00/20 23:00/17 24:00/17","schedule_monday":"00:00/17 06:00/21 08:30/19 19:00/20 23:00/17 24:00/17","schedule_saturday":"00:00/17 07:00/21 09:00/19 19:00/20 23:00/17 24:00/17","schedule_sunday":"00:00/17 07:00/21 09:00/19 19:00/20 23:00/17 24:00/17","schedule_thursday":"00:00/17 06:00/21 08:30/19 19:00/20 23:00/17 24:00/17","schedule_tuesday":"00:00/17 06:00/21 08:30/19 19:00/20 23:00/17 24:00/17","schedule_wednesday":"00:00/17 06:00/21 08:30/19 19:00/20 23:00/17 24:00/17","system_mode":"heat","update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":null,"working_day":"mon_fri+sat+sun"}'

git-tail-d, 2023-10-15:
same issue with Tuya Devices 
https://github.com/Koenkk/zigbee2mqtt/discussions/19122
https://github.com/Koenkk/zigbee2mqtt/issues/15306

MahrWe, 2023-10-15:
Probably, another related issue with Tuya devices

#19240

Koenkk, 2023-10-15:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/19240

closed 2023-10-15 by Koenkk