Koenkk/zigbee2mqtt #19299: Onesti Products AS easyCodeTouch_v1 doesnt show correct battery ststus

stefancn, 2023-10-14:
### What happened?

When the batteries are "empty"  , eq. 20% left ... the doorlock battery (batteryPercentageRemaining) stills shows 100%...  Today my lock blinked red, signaling urgent battery change.   Is this bug fixable. ?

Regards  /Stefan

### Zigbee2MQTT version

1.33.1-1

### Adapter firmware version

7.1.1.0 build 273

### Adapter

HA Skyconnect

stefancn, 2024-02-18:
After latest update, it shows 200% !!!

msandvold, 2024-03-03:
https://github.com/Koenkk/zigbee2mqtt/issues/21182
Same here.

msandvold, 2024-03-15:
check this out, waiting for an answer.
[https://github.com/Koenkk/zigbee-herdsman-converters/pull/6940](url)

stefancn, 2024-03-16:
Thanks... but all other info seems to work, door open/close last used pin... remote opening work. It is only the battery status that is wierd.

msandvold, 2024-04-01:
It is now fixed with Version 1.36.1 :)

stefancn, 2024-04-08:
Thanks, but do you have any idea why it now is stucked at 100%...

closed 2024-11-06 by github-actions[bot]