Koenkk/zigbee2mqtt #19308: Tuya leak sensor TS0207 not reporting battery

Suxsem, 2023-10-16:
### What happened?

Hi bought a couple Aubess IH-K665 (that seems to be a rebranded TS0207, vendor _TZ3000_k4ej3ww2).

It seems to me that device ONLY communicates when leak status changes.

The "last seen" attribute in the Z2M dashboard is always "N/A" (despite having enabled availability and added last_seen: ISO_8601_local to the mqtt configuration. Availability says "online" but I don't know on what it is based.

I tried with this but after pressing apply I always get a timeout error.
[image]

I enabled Z2M logging about an hour ago and so far I didn't get any log about any communication with the device, I will monitor the next 24 hours.

Battery says always 100%, I'm afraid to not get any low battery notification and thus ends with an offline device.

Thank you

### Zigbee2MQTT version

1.33.1

### Adapter firmware version

7.3.1.0 build 0

### Adapter

EZSP v12

buddy31130, 2023-11-05:
Mine is always showing low battery even with new ones

Suxsem, 2023-11-21:
They started to working, I think the issue was in my zigbee network. Sorry for the false reporting

closed 2023-11-21 by Suxsem

DwarfDecker, 2024-10-18:
I've to devices in to different houses, one repport 0% for weeks and the other get not battery level at all.

Note: Zigbee2MQTT version
[1.40.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.40.1)

shawnhaywood, 2024-10-19:
Reviving this post.  I purchased the same device and have the exact same issue.  Device not providing any kind of update unless leak detected.

stevenengland, 2025-03-09:
related: https://github.com/Koenkk/zigbee2mqtt/issues/21611