Koenkk/zigbee2mqtt #19310: [Feature request]: S520567 position states [Stopped/Opening/Closing]

Matssa56, 2023-10-16:
### Is your feature request related to a problem? Please describe

This is to open the issue #16827 again.

Today we have two states : closed, opened. The idea is to add opening and closing. We know that the official Wiser app reports these informations (see screenshot [in #16827](https://github.com/Koenkk/zigbee2mqtt/issues/16827#issuecomment-1483417802) ).

### Describe the solution you'd like

Motor_state or moving or running or openning or closing state to sned to Home Assistant

### Describe alternatives you've considered

I tried using template covers in Home Assistant and they don't seem to give my the opening or closing, even if I supply the %. I was thinking of making some templates / automations in order to set the opening or closing if n,ew_position > old_position or something like that but we will have at least 1s delay and not sure that it will really work well.

### Additional context

I have several devices and willing to test stuff out but I need help understanding what to do.
Here are the reports from the clusters from Zigbee2mqtt : 
[image]

Matssa56, 2023-10-16:
Something that we can do is maybe see what changes in terms of cluster or feedback when triggering the openning or closing of the motor, not sure how I can do that though ^^'

Pepeanuts, 2023-12-14:
Any update for this feature request ?

closed 2024-07-12 by github-actions[bot]