Koenkk/zigbee2mqtt #19315: Aqara Xiaomi RTCGQ11LM pairs and then never reconnects

Ltek, 2023-10-16:
### What happened?

Pairs fine, reads all attributes then never connects again. Looks like it break the link to the coordinator and never reconnects.. no matter where I place the motion sensor. I have 8 of these these same sensors connected to Conbee and working for 2+ years all over my house. The radios are in the same location.

### What did you expect to happen?

communicate and update status of all parameters

### How to reproduce it (minimal and precise)

Pair the device.

### Zigbee2MQTT version

1.33.0

### Adapter firmware version

 EZSP v8 6.10.3.0 build 297

### Adapter

Sonoff

### Debug log

Debug log is empty.

[image]
[image]
[image]

Rujith, 2023-11-20:
Have the same issue. Pairs fine, but then status is never updated. The same Aqara motion sensor works fine in ZHA

therealpeterduke, 2023-11-21:
Hey,

I had the same problem with the motion detector and the MCCGQ11LM.
The solution was for me (it has been running for a few days without any problems):
Disconnect the device, start pairing mode and keep pressing the small button during this until the interview is completed.
I did this several times in a row on different days until the connection remained stable.

The strange thing is that the devices do not connect to the coordinator that is closer, but to routers that are in the network.

I have two other MCCGQ11LMs that worked from the start without any problems.

Zigbee2MQTT version 1.33.2 
Coordinator-Type EZSP v8

Good luck!

Rujith, 2023-11-22:
> Hey,
>
> I had the same problem with the motion detector and the MCCGQ11LM. The solution was for me (it has been running for a few days without any problems): Disconnect the device, start pairing mode [...]

Still didn't work, I will keep trying. Thanks for your help

Ltek, 2023-11-22:
@Koenkk would love to have this fixed... its holding me back from migrating away from Conbee. Looks like this is a pervasive problem

Koenkk, 2023-11-22:
This is probably related to the EZSP firmware you are running, I suggest to try a different fw.

essegi65, 2023-11-30:
> > This is probably related to the EZSP firmware you are running, I suggest to try a different fw.
> 
> I'm running the stock Sonoff dongle, did know there was an alternative... please suggest something specific with a link... I'd have no idea what works/doesnt.

I have the same problem with coordinator SLZB-06 FW: 2.0.8, Z2M ver: 1.33.2.1, HAASSOS: 2023.11.6
Always with xiaomi device: RTCGQ01LM.
once the paring is done the status is not updated and the detection does not work

Geoffrey54, 2023-12-23:
Same here, paring is fine but losing connexion after. The sonoff coordinator is near the device. I will try to disconnect and reconnect it several time…

Rujith, 2024-01-02:
> Still didn't work, I will keep trying. Thanks for your help

Worked finally after several retries, reboots of HA, Z2M and battery removals etc. Not sure what helped, but finally it paired again.

Ltek, 2024-07-01:
This is a problem with Z2M... i'm on the latest FW and Z2M and it simply doesnt like these devices. I have 6 of them. They all pair fine, and quickly, with Conbee/Deconz... even v1 of Conbee device. I recently updated to Conbee III and it works with them also.

bchevreau, 2024-07-16:
Agreed with Ltek here, it seems related either to Z2M or ember firmware devices, and it became worse with last versions of Z2M. My 2 Aqara Sensors (the original versions) suddenly stop reporting, but they don't show as offline or anything. They look just fine when looking at them and their exposes in Z2M, but they are in fact silently dead. The only way to repair is to do a Permit All and maintain the pair button on the Aqara device, then click it several times after the 3 "flashes" announcing start of pairing... Then they start to report state instantly, as if re-paired. Z2M never throws an error (despite maybe a power-outage-count going up? which is nonsense for a battery-powered device).

Ltek, 2024-07-16:
@bchevreau thanks for the supporting data. Z2M has become 100x less stable and less 'able' in the past several months. 

I gave upp trying to move devices from Deconz to Z2M as none of them worked.  Z2M has been show-stopper buggy with every device I've tried : Lightify LEDs (mains powered), two models of Aqara motion sensors, Aqara humidity sensors, Bosch motion, and more.  Oddly Z2M works fantastic with rechargeable Tuya roller motors ("covers").

the developer seems more focused on adding new device support than making devices actually work properly or fully... all the devices I tried are "supported" -- ah, nope ;-)

example... Bosch motion sensors are impossible on Z2M... they only partly pair (motion parameter doesnt pair/enable 90% of the time) and when they do, they disconnect and never reconnect almost immediately after pairing, even when only 6 feet away in open air. Yet Z2M updates 'last seen' frequently.

I'm no longer trying to migrate to Z2M, I've moved a dozen devices (which didn't work proper on Z2M) back to conbee/deconz and they work as expected. I only have a few sensors (which I paired on a very old version of Z2M prior to ember fw) on Z2M and 6 tuya motors. 

I'll check back in 6 months and see if it it worth trying again. There are some fundamental design differences with Z2M vs Deconz where I think Z2M is far less 'non expert friendly' and also makes device support far more difficult. Deconz separates each device into a separate config file (DDF) so if I want to add a 'like' device I can just copy/clone the DDF or add the model # to it inside. I made some very small and easy suggestions for Z2M to make things easier to do similar but they were dismissed

Z2M has potential... but I think he's too focused on trees and not getting a view of what is needed for the forest.

asferreira, 2024-08-11:
I'm not sure I'm having the exact same problem, but what was described is very similar to what I'm facing. 
After pairing the motion sensor it remains silent for a long time, not reporting any motion detection. The difference is that the sensor communicates with the coordinator from time to time.
I have removed and paired the sensor many times, checked battery and signal level and everything is OK, but result is the same. The only situation that communication always occurs is when I short press the device's button. 
I using a Sonoff USB Dongle P as coordinator with fw version 20230507 (zStack3x0) and Z2M version 1.39.1.

Ltek, 2025-01-04:
Bummpppers

closed 2025-03-15 by github-actions[bot]