Koenkk/zigbee2mqtt #19317: Reporting tab interface completely broken after 1.33.1 update

DmitriK, 2023-10-17:
### What happened?

After upgrading to 1.33.1, my VINDSTYRKA devices were no longer updating at the rate that I had previously configured them. Upon going to the reporting tab, I noticed that all the clusters were listed twice, once with my original settings, and once with what appears to be the default. It also appears to be impossible to fix. I can disable all the extra entries except for pm25measurement, where I first have to change the attribute to measuredValueIkea, otherwise I get a `Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'ConfigureReporting 0x385cfbfffea36292/1 pm25Measurement([{"attribute":"measuredValue","minimumReportInterval":60,"maximumReportInterval":65535,"reportableChange":2}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'INVALID_DATA_TYPE')'`
However, attempting to change any of the settings results in more duplicates returning, though they seem to match my settings.
Screenshot of what this looks like after any attempt at editing:
[image]

### What did you expect to happen?

A) zigbee2mqtt does not erase my previous reporting settings
B) I can actually use the reporting tab to adjust the settings.

### How to reproduce it (minimal and precise)

Unsure what the minimum requirement is. As mentioned above, I'm using VINDSTYRKA devices (don't know if it's specific to them), and simply trying to set reporting settings.

### Zigbee2MQTT version

1.33.1

### Adapter firmware version

0x26580700

### Adapter

ConBee2/RaspBee2

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/12923846/log.txt)

sanyafifa, 2023-10-31:
I connected my new sensor E2112 yesterday - and there is no data from it except LQI.

Zigbee2MQTT version
1.33.1 dev

alexh3o, 2023-11-08:
Hi,
After noticing my ZLinky did'nt update properly for some values, I noticed the same problem in reporting page.
Some duplicates can be deleted, others not, and when I try to change a value (min rep interval or min rep change), a duplicate reporting is added again instead of changing the adequate line.
This is on a raspberry pi 4 and HA supervised:
Core 2023.11.1
Supervisor 2023.11.0
Frontend 20231030.
Zigbee2mqtt 1.33.2

[image]

Koenkk, 2023-11-08:
@alexh3o could you provide your data/database.db?

alexh3o, 2023-11-08:
Here is the part for ZLinky

`{"id":58,"type":"Router","ieeeAddr":"0x00158d000966cc3e","nwkAddr":43723,"manufId":4151,"manufName":"LiXee","powerSource":"Mains (single phase)","modelId":"ZLinky_TIC\u0000","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":83,"inClusterList":[0,3,1794,2820,2817,65382,1],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"ZLinky_TIC\u0000","manufacturerName":"LiXee","powerSource":1,"zclVersion":3,"appVersion":12,"stackVersion":2,"hwVersion":1,"dateCode":"20230214","swBuildId":"4000-0013"}},"liXeePrivate":{"attributes":{"linkyMode":0,"currentTarif":"BASE\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","warnDPS":0}},"haElectricalMeasurement":{"attributes":{"rmsCurrentMax":90,"apparentPower":1950,"rmsCurrent":7}},"seMetering":{"attributes":{"meterSerialNumber":{"type":"Buffer","data":[48,51,49,55,54,50,48,52,53,49,49,53]},"activeRegisterTierDelivered":"TH..","currentSummDelivered":[0,39755333]}},"haMeterIdentification":{"attributes":{"availablePower":60}}},"binds":[{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c82c76","endpointID":1},{"cluster":2817,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c82c76","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c82c76","endpointID":1},{"cluster":65382,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c82c76","endpointID":1}],"configuredReportings":[{"cluster":1794,"attrId":0,"minRepIntval":60,"maxRepIntval":900,"repChange":100},{"cluster":2820,"attrId":1288,"minRepIntval":60,"maxRepIntval":900,"repChange":1},{"cluster":2820,"attrId":1295,"minRepIntval":60,"maxRepIntval":900,"repChange":1},{"cluster":65382,"attrId":5,"minRepIntval":60,"maxRepIntval":900,"repChange":1},{"cluster":65382,"attrId":5,"minRepIntval":60,"maxRepIntval":900,"repChange":1,"manufacturerCode":null},{"cluster":1794,"attrId":0,"minRepIntval":60,"maxRepIntval":900,"repChange":10,"manufacturerCode":null},{"cluster":2820,"attrId":1295,"minRepIntval":15,"maxRepIntval":900,"repChange":1,"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":12,"stackVersion":2,"hwVersion":1,"dateCode":"20230214","swBuildId":"4000-0013","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1745911956},"lastSeen":1699468392019,"defaultSendRequestWhen":"immediate"}`

I can send the whole database.db is it's useful, should I do it here or through a more private channel ?

seblu, 2023-12-01:
I'm also affected by this. In my case, duplication occurs when I'm updating values in the Reporting tab.

seblu, 2023-12-01:
Same issue with freshly released `1.34.0`. Despite no error reported, it looks like it happens when the original value cannot be changed.

After updating `activePower` `min rep change` value from 10 to 1:
- A new line is added at the bottom with the requested value.
- The old line remain with its value field updated to the requested value.
[image]

After a page reload, value is back and new row is still here:
- The original line value has changed to the original value. As if it has not changed.
- The new line remain with the requested value.
[image]

I can remove the new line. On another device, the same behavior was fixed by repairing (removing/pairing).

alexh3o, 2023-12-01:
Hi, if you remove and repair your device, you can configure reporting without having duplicates after ?

emandtf, 2023-12-01:
Just updated to 1.34.0 and now (without doing anything) many "entities" in database.db are duplicated and reported with default/wrong Reporting times/repo_changes.

Those dupplicated entities are the same that was previously affected by this duplication bug when we was trying to update their values.

Koenkk, 2023-12-03:
Pushed a fix! After upgrading, click `Apply` on the configured reportings to get rid of the duplicates.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-12-03 by Koenkk

Koenkk, 2023-12-04:
Can you try to execute `rm -f tsconfig.tsbuildinfo` in the z2m folder first?

emandtf, 2023-12-05:
Different error but unable to run Z2M. 

> zigbee2mqtt@1.34.0-dev start
> node index.js
> Building Zigbee2MQTT... (hash changed), failed
> Error: Command failed: npm run build
> at ChildProcess.exithandler (node:child_process:419:12)
> at ChildProcess.emit (node:events:513:28)
> at maybeClose (node:internal/child_process:1091:16)
> at Socket.<anonymous> (node:internal/child_process:449:11)
> at Socket.emit (node:events:513:28)
> at Pipe.<anonymous> (node:net:321:12)

Koenkk, 2023-12-05:
What if you execute `npm run build`?

nplawes, 2023-12-19:
```
docker exec -it z2m grep version package.json
  "version": "1.34.0",
```
I appear to be running the "fixed" version, but pressing "Apply" does not fix the issue for me (even after a page refresh)

Also tried on a device without duplicates... I now have duplicates :-(

seblu, 2023-12-20:
Same here. I tried few days ago and duplication is still here.

Koenkk, 2023-12-21:
@nplawes 

> Also tried on a device without duplicates... I now have duplicates :-(

Could you provide the data/database.db entry of this device?

nplawes, 2023-12-21:
I'm not sure which version broke this for me.

It also happens for SPLZB-134 smart plugs.

Looking at the data from the database I see that the duplicated entries all have "manufacturerCode": null

Thanks

/nick

{
  "id": 18,
  "type": "EndDevice",
  "ieeeAddr": "0x00178801032ab969",
  "nwkAddr": 11099,
  "manufId": 4107,
  "manufName": "Philips",
  "powerSource": "Battery",
  "modelId": "SML001",
  "epList": [
    1,
    2
  ],
  "endpoints": {
    "1": {
      "profId": 49246,
      "epId": 1,
      "devId": 2128,
      "inClusterList": [
        0
      ],
      "outClusterList": [
        0,
        3,
        4,
        6,
        8,
        768,
        5
      ],
      "clusters": {},
      "binds": [],
      "configuredReportings": [],
      "meta": {}
    },
    "2": {
      "profId": 260,
      "epId": 2,
      "devId": 263,
      "inClusterList": [
        0,
        1,
        3,
        1030,
        1024,
        1026
      ],
      "outClusterList": [
        25
      ],
      "clusters": {
        "genPowerCfg": {
          "attributes": {
            "batteryPercentageRemaining": 45
          }
        },
        "msTemperatureMeasurement": {
          "attributes": {
            "measuredValue": 1745
          }
        },
        "msIlluminanceMeasurement": {
          "attributes": {
            "measuredValue": 15635
          }
        },
        "msOccupancySensing": {
          "attributes": {
            "48": 2,
            "occupancy": 0,
            "pirOToUDelay": 0
          }
        }
      },
      "binds": [
        {
          "cluster": 1,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b00238da250",
          "endpointID": 1
        },
        {
          "cluster": 1024,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b00238da250",
          "endpointID": 1
        },
        {
          "cluster": 1026,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b00238da250",
          "endpointID": 1
        },
        {
          "cluster": 1030,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b00238da250",
          "endpointID": 1
        }
      ],
      "configuredReportings": [
        {
          "cluster": 1,
          "attrId": 33,
          "minRepIntval": 900,
          "maxRepIntval": 3600,
          "repChange": 1
        },
        {
          "cluster": 1030,
          "attrId": 0,
          "minRepIntval": 0,
          "maxRepIntval": 3600,
          "repChange": 0
        },
        {
          "cluster": 1026,
          "attrId": 0,
          "minRepIntval": 300,
          "maxRepIntval": 3600,
          "repChange": 10
        },
        {
          "cluster": 1024,
          "attrId": 0,
          "minRepIntval": 60,
          "maxRepIntval": 3600,
          "repChange": 200
        },
        {
          "cluster": 1,
          "attrId": 33,
          "minRepIntval": 900,
          "maxRepIntval": 3600,
          "repChange": 1,
          "manufacturerCode": null
        }
      ],
      "meta": {}
    }
  },
  "appVersion": 2,
  "stackVersion": 1,
  "hwVersion": 1,
  "dateCode": "20190219",
  "swBuildId": "6.1.1.27575",
  "zclVersion": 1,
  "interviewCompleted": true,
  "meta": {
    "configured": 2
  },
  "lastSeen": 1703149418656,
  "defaultSendRequestWhen": "immediate"
}

emandtf, 2023-12-21:
> I'm not sure which version broke

Are you sure you're running 1.34-dev version?
My duplicated values were fixed just by pressing "Apply" on those values, and then refresh the web page.

nplawes, 2023-12-21:
I don't see a docker tag for 1.34-dev. There is a latest-dev, which hopefully is stable enough to not break my live environment :-)

And indeed that does seem to fix matters, with 'apply' and refresh page.

Also fixes the annoying persistent pop-ups, which I was going to report on next :-)

Thanks

/nick

emandtf, 2023-12-21:
> I don't see a docker tag for 1.34-dev. There is a latest-dev, which hopefully is stable enough to not break my live environment

Don't try on a production environment. Dev branch could be often broken and doesn't allow to start.

nplawes, 2023-12-21:
Too late :-)

I do generally avoid using dev branches, and will switch back to the stable branch when it's next updated.

Although, I can obviously switch back to 'latest' now, if there is no incompatibility introduced by 'latest-dev'.

Fortunately I do have daily backups, so can revert in an emergency.

/nick

Koenkk, 2023-12-22:
Ah I forgot this fix did not made it into 1.34.0, fix will be included in the 1 January release.

nplawes, 2023-12-22:
That sounds great... The dev release seems fine at the moment, so I will switch back to the mainstream release next year.

Thanks!

/nick