Koenkk/zigbee2mqtt #19318: Z2M don't start after last dev-Update: Error: Cannot find module './dist/util/settings'

Burki24, 2023-10-17:
### What happened?

Make last Update to dev from 16.10.2023. Running without any problems. Then starting Z2M and got the error calld in title.

> Oct 17 05:24:09 Z2M-EG systemd[1]: Started zigbee2mqtt.
> Oct 17 05:24:10 Z2M-EG npm[873170]: > zigbee2mqtt@1.33.1-dev start
> Oct 17 05:24:10 Z2M-EG npm[873170]: > node index.js
> Oct 17 05:24:11 Z2M-EG npm[873186]: Error: Cannot find module './dist/util/settings'
> Oct 17 05:24:11 Z2M-EG npm[873186]: Require stack:
> Oct 17 05:24:11 Z2M-EG npm[873186]: - /opt/zigbee2mqtt/index.js
> Oct 17 05:24:11 Z2M-EG npm[873186]:     at Function.Module._resolveFilename (node:internal/modules/cjs/loader:1048:15)
> Oct 17 05:24:11 Z2M-EG npm[873186]:     at Function.Module._load (node:internal/modules/cjs/loader:901:27)
> Oct 17 05:24:11 Z2M-EG npm[873186]:     at Module.require (node:internal/modules/cjs/loader:1115:19)
> Oct 17 05:24:11 Z2M-EG npm[873186]:     at require (node:internal/modules/helpers:130:18)
> Oct 17 05:24:11 Z2M-EG npm[873186]:     at start (/opt/zigbee2mqtt/index.js:90:22)
> Oct 17 05:24:11 Z2M-EG systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
> Oct 17 05:24:11 Z2M-EG systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
> Oct 17 05:24:11 Z2M-EG systemd[1]: zigbee2mqtt.service: Consumed 1.999s CPU time.
> Oct 17 05:24:21 Z2M-EG systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 135.
> Oct 17 05:24:21 Z2M-EG systemd[1]: Stopped zigbee2mqtt.
> Oct 17 05:24:21 Z2M-EG systemd[1]: zigbee2mqtt.service: Consumed 1.999s CPU time.

### What did you expect to happen?

I'm not sure what happens. A friend of me has the same issue after last update.

### Zigbee2MQTT version

1.33.

### Adapter firmware version

Original EZSP

### Adapter

sonoff-zigbee-3-0-usb-dongle-plus-e

Koenkk, 2023-10-17:
Can you try executing a `npm run build` after switching to the dev branch?

T-IOOO, 2023-10-17:
Having the same issue

Oct 17 10:38:23 pi systemd[1]: zigbee2mqtt.service: Consumed 1.748s CPU time.
Oct 17 10:38:34 pi systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 92.
Oct 17 10:38:34 pi systemd[1]: Stopped zigbee2mqtt.service - zigbee2mqtt.
Oct 17 10:38:34 pi systemd[1]: zigbee2mqtt.service: Consumed 1.748s CPU time.
Oct 17 10:38:34 pi systemd[1]: Started zigbee2mqtt.service - zigbee2mqtt.
Oct 17 10:38:35 pi npm[13522]: > zigbee2mqtt@1.33.1-dev start
Oct 17 10:38:35 pi npm[13522]: > node index.js
Oct 17 10:38:35 pi npm[13534]: Error: Cannot find module './dist/util/settings'
Oct 17 10:38:35 pi npm[13534]: Require stack:
Oct 17 10:38:35 pi npm[13534]: - /opt/zigbee2mqtt/index.js
Oct 17 10:38:35 pi npm[13534]:     at Function.Module._resolveFilename (node:internal/modules/cjs/loader:1077:15)
Oct 17 10:38:35 pi npm[13534]:     at Function.Module._load (node:internal/modules/cjs/loader:922:27)
Oct 17 10:38:35 pi npm[13534]:     at Module.require (node:internal/modules/cjs/loader:1143:19)
Oct 17 10:38:35 pi npm[13534]:     at require (node:internal/modules/cjs/helpers:119:18)
Oct 17 10:38:35 pi npm[13534]:     at start (/opt/zigbee2mqtt/index.js:90:22)
Oct 17 10:38:35 pi systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Oct 17 10:38:35 pi systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.

npm run build is not doing anything.

Koenkk, 2023-10-17:
And with `rm tsconfig.tsbuildinfo && npm run build`?

okastl, 2023-10-17:
I had the same problem...

> And with `rm tsconfig.tsbuildinfo && npm run build`?

This solves it, thanks!

On Windows 
```
del tsconfig.tsbuildinfo
npm run build
```

T-IOOO, 2023-10-18:
> And with `rm tsconfig.tsbuildinfo && npm run build`?

sudo rm tsconfig.tsbuildinfo && sudo npm run build

It worked! 👍

Burki24, 2023-10-18:
> rm tsconfig.tsbuildinfo && npm run build

It also worked for me, many thanks.

closed 2023-10-18 by Burki24

Koenkk, 2023-10-18:
if you do a 'npm run build' again, dies it work?

Burki24, 2023-10-19:
> npm run build

Yes, without problems.

Koenkk, 2023-10-20:
@Burki24 so what happens when you now execute:

```bash
git pull
npm run build
```

Burki24, 2023-10-21:
> @Burki24 so what happens when you now execute:
> 
> ```shell
> git pull
> npm run build
> ```

Now it runs well. Thanks a lot.

jsassner, 2024-01-07:
I got this error with 1.35.1 (ef84ff86fed89d0957961580f92541b20546b951). Doing npm run build fixed it. I was running 1.35.0 previously (dev version, don't know exactly what git commit)