Koenkk/zigbee2mqtt #19322: LEDVANCE 4058075729100 OTA Error

tomiss288, 2023-10-17:
### What happened?

Hello,
OTA checked and inform about new firmware for LEDVANCE 4058075729100 led lamp but during updated occurs error : Update of 'Korytarz Lampka Mala' failed (The value of "offset" is out of range. It must be >= 0 and <= 188380. Received 188384).
I found similar problem but was not resolved. 
Could you pls check it ?

### Zigbee2MQTT version

1.33.1

### Adapter firmware version

20230922

### Adapter

zStack3x0

### Debug log

Zigbee2MQTT:info  2023-10-17 17:34:02: MQTT publish: topic 'zigbee2mqtt/Korytarz Lampka Mala', payload '{"brightness":254,"device":{"applicationVersion":0,"friendlyName":"Korytarz Lampka Mala","hardwareVersion":3,"ieeeAddr":"0xf0d1b8be24080ce8","manufacturerID":4489,"manufacturerName":"LEDVANCE","model":"4058075729100","networkAddress":9810,"powerSource":"Mains (single phase)","softwareBuildID":"03143660","stackVersion":2,"type":"Router","zclVersion":8},"linkquality":76,"state":"ON","update":{"installed_version":51656288,"latest_version":52442720,"state":"available"},"update_available":true}'
Zigbee2MQTT:info  2023-10-17 17:34:02: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'Korytarz Lampka Mala' failed (The value of \"offset\" is out of range. It must be >= 0 and <= 188380. Received 188384)","meta":{"device":"Korytarz Lampka Mala","status":"update_failed"},"type":"ota_update"}'

ilwalti, 2024-04-15:
Still not resolved. Any hints?

PWnet, 2024-12-10:
Same problem here with a TS0601_thermostat.

info 2024-12-10 14:35:01z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Checking if update available for 'TRV Bagno PT'","meta":{"device":"TRV Bagno PT","status":"checking_if_available"},"type":"ota_update"}'
info 2024-12-10 14:35:06z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Failed to check if update available for 'TRV Bagno PT' (The value of \"offset\" is out of range. It must be >= 0 and <= 76386. Received 76388)","meta":{"device":"TRV Bagno PT","status":"check_failed"},"type":"ota_update"}'
info 2024-12-10 14:35:06z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"TRV Bagno PT"},"error":"Failed to check if update available for 'TRV Bagno PT' (The value of \"offset\" is out of range. It must be >= 0 and <= 76386. Received 76388)","status":"error","transaction":"gzwc4-1"}'
error 2024-12-10 14:35:06z2m: Failed to check if update available for 'TRV Bagno PT' (The value of "offset" is out of range. It must be >= 0 and <= 76386. Received 76388)

closed 2025-02-17 by github-actions[bot]