Koenkk/zigbee2mqtt #19324: Namron thermostat outlet plug - Add support for thermostat mode

slippern1, 2023-10-17:
### What happened?

The Namron thermostat outlet plug (4512749-N) does support thermostat mode, but this is not added in z2m.

[image]

[image]

Can someone add this support?

### Zigbee2MQTT version

1.33.1

### Adapter firmware version

20210708

### Adapter

zStack3x0

ok-67, 2023-10-18:
I have this plugg. 
Using ZHA it is reported as a thermostat outlet plug and it is possible to set the wanted temperature. However out of the box it only operates as a regular on/off plug, but when setting the proprietary attribute at 0x1000 to 1 it operates nicely as a thermostat outlet plug.

I have tried to make an external converter file for it, but for some strange reason the external_converters: option is automatically removed from my configuration file.

exentrik, 2023-10-30:
I also didnt get this device to work as thermostat in zha , and tried z2m. But here I can confirm its not registring as thermostat/climate

@ok-67 I went back to ZHA, which attribute did you set, i see there is 2 called 0x1000.

- LightLink (Endpoint id: 1, Id: 0x1000, Type: in)
- LightLink (Endpoint id: 1, Id: 0x1000, Type: out)

which of these do i alter.?   screenshot would be nice. thank you.
[image]

ok-67, 2023-11-01:
[image]

exentrik, 2023-11-08:
> [image]

Thank you @ok-67  The problem was zha toolkit had a write problem in its current version (worked a week later).

I copied the existing driver and made an external converter for Z2M , and then tried to add code for thermostat without success.

Kjelljk, 2024-05-09:
Trying to write to cluster 0x0000 and attribute 0x1000 with dev console, but have not figured it out. Any tips how to write to this attribute? 

After setting attribute to "1" for HVAC mode, and rebooting the device it shall be in thermostat mode. I can then try to make a profile for this mode.

Kjelljk, 2024-05-20:
> The dew console dont support this. I have made some code that do work. However I am new to type script and the code are not polished enough to be included, in short I am waiting for someone to write a modern extend for the thermostat module, then I will update my code and hopefully it can be included in Z2M

Share?

ok-67, 2024-05-21:
[plugg.zip](https://github.com/Koenkk/zigbee2mqtt/files/15387050/plugg.zip)
This is the plugg.js that I use as an external converter for the namron plugs in my homeassistant installations. This works OK, but for some of my plugs I had to reconfigure them after enabling thermostat mode before I could access all the parameters.

On my computer I am running a separate Z2M installation under Ubuntu for development purposes. There I have a different version that I am working on. There I am using modern extend for some of the parameters, but not for the thermostat parameter. I hope someone will make a modern extend for the thermostat so I can implement that one also. Then I hope it should be appropriate for an "official" version.

closed 2025-01-12 by github-actions[bot]