Koenkk/zigbee2mqtt #19327: [New device support]: Zigbee model 'TS004F' and manufacturer name '_TZ3000_nuombroo'

konistehrad, 2023-10-17:
### Link

https://www.aliexpress.us/item/3256805646640443.html?spm=a2g0o.order_list.order_list_main.30.50971802p4CMa0&gatewayAdapt=glo2usa

### Database entry

{"zclVersion":3,"epList":[1,2,3,4],"ieeeAddr":"0x70ac08fffe4c1c58","powerSource":"Battery","defaultSendRequestWhen":"immediate","modelId":"TS004F","appVersion":65,"manufName":"_TZ3000_nuombroo","nwkAddr":38778,"dateCode":"","interviewCompleted":true,"manufId":4098,"stackVersion":0,"lastSeen":1697566763986,"hwVersion":1,"id":24,"meta":{"configured":-64804543},"endpoints":{"1":{"inClusterList":[0,1,3,4,6,4096],"devId":260,"binds":[{"cluster":1,"deviceIeeeAddress":"0x00212effff097ee8","type":"endpoint","endpointID":1},{"cluster":6,"deviceIeeeAddress":"0x00212effff097ee8","type":"endpoint","endpointID":1}],"configuredReportings":[{"cluster":1,"maxRepIntval":62000,"attrId":33,"repChange":0,"manufacturerCode":null,"minRepIntval":3600}],"meta":{},"epId":1,"outClusterList":[25,10,3,4,5,6,8,4096],"profId":260,"clusters":{"genOnOff":{"attributes":{"tuyaOperationMode":1}},"manuSpecificTuya_3":{"attributes":{}},"genBasic":{"attributes":{"zclVersion":3,"powerSource":3,"modelId":"TS004F","65503":"\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0012\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0012\u0000\u0000\u0000\u0000\u0012","appVersion":65,"dateCode":"","65534":0,"stackVersion":0,"manufacturerName":"_TZ3000_nuombroo","65506":31,"65508":1}},"genPowerCfg":{"attributes":{"batteryVoltage":30,"batteryPercentageRemaining":200}}}},"4":{"inClusterList":[],"binds":[{"cluster":6,"deviceIeeeAddress":"0x00212effff097ee8","type":"endpoint","endpointID":1}],"configuredReportings":[],"meta":{},"epId":4,"outClusterList":[],"clusters":{}},"2":{"inClusterList":[],"binds":[{"cluster":6,"deviceIeeeAddress":"0x00212effff097ee8","type":"endpoint","endpointID":1}],"configuredReportings":[],"meta":{},"epId":2,"outClusterList":[],"clusters":{}},"3":{"inClusterList":[],"binds":[{"cluster":6,"deviceIeeeAddress":"0x00212effff097ee8","type":"endpoint","endpointID":1}],"configuredReportings":[],"meta":{},"epId":3,"outClusterList":[],"clusters":{}}},"type":"EndDevice"}

### Comments

I took the existing TS004F converter and expanded it to both include the new `_TZ3000_nuombroo` fingerprint, but also expose the command mode for allowing group association.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: tuya.fingerprint('TS004F', ['_TZ3000_nuombroo']),
    model: 'TS004F',
    vendor: 'TuYa',
    description: 'Wireless switch with 4 buttons',
    exposes: [
        e.battery(), 
        e.enum('operation_mode', ea.ALL, ['command', 'event']).withDescription(
                'Operation mode: "command" - for group control, "event" - for clicks'),
        e.action(['on', 'off', 'brightness_step_up', 'brightness_step_down', '1_single', '1_double', '1_hold', '2_single', '2_double', '2_hold',
            '3_single', '3_double', '3_hold', '4_single', '4_double', '4_hold'])],

    fromZigbee: [fz.battery, fz.tuya_on_off_action, fz.tuya_operation_mode, fz.command_on, fz.command_off, fz.command_step, fz.command_move],
    toZigbee: [tz.tuya_operation_mode],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await endpoint.read('genBasic', [0x0004, 0x000, 0x0001, 0x0005, 0x0007, 0xfffe]);
        await endpoint.write('genOnOff', {'tuyaOperationMode': 1});
        await endpoint.read('genOnOff', ['tuyaOperationMode']);
        try {
            await endpoint.read(0xE001, [0xD011]);
        } catch (err) {/* do nothing */}
        await endpoint.read('genPowerCfg', ['batteryVoltage', 'batteryPercentageRemaining']);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
        for (const ep of [1, 2, 3, 4]) {
            // Not all variants have all endpoints
            // https://github.com/Koenkk/zigbee2mqtt/issues/15730#issuecomment-1364498358
            if (device.getEndpoint(ep)) {
                await reporting.bind(device.getEndpoint(ep), coordinatorEndpoint, ['genOnOff']);
            }
        }
	await reporting.batteryPercentageRemaining(endpoint);
    },
};

module.exports = definition;
```

Koenkk, 2023-10-17:
Could you make a pr to add support for this device?

konistehrad, 2023-10-17:
Sorry, actions list should also include `brightness_move_up` and `brightness_move_down`, but otherwise this converter looks complete. Recommend pushing some of this back to the converter I took this from to allow operation mode modification: https://github.com/Koenkk/zigbee-herdsman-converters/blob/c9b8f3172cb11ea0ca36440f8956eda582182df7/src/devices/tuya.ts#L2005

Koenkk, 2023-10-17:
> Are you object to modifying the existing converter with the additional parts, or should I just hammer in the manufacturer name for the PR?

Please create a new definition (don't modify existing one)

closed 2023-10-20 by Koenkk