Koenkk/zigbee2mqtt #19332: [New device support]: lumi.remote.b186acn03 (Xiaomi Aqara T1 single rocker wireless remote switch)

jeong840, 2023-10-18:
### Link

https://smartstore.naver.com/aqarakorea/products/5712908829

### Database entry

{"id":19,"type":"EndDevice","ieeeAddr":"0x54ef4410001ddd8c","nwkAddr":39275,"manufId":4447,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.remote.b186acn03","epList":[1,2,3,4,5,6],"endpoints":{"1":{"profId":260,"epId":1,"devId":259,"inClusterList":[0,3,1],"outClusterList":[3,6],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.remote.b186acn03","appVersion":29,"manufacturerName":"LUMI","powerSource":3,"zclVersion":3,"stackVersion":2,"hwVersion":1,"dateCode":"20200324","swBuildId":"2019\u0000www."}},"genPowerCfg":{"attributes":{"batteryVoltage":32}},"aqaraOpple":{"attributes":{"247":{"type":"Buffer","data":[1,33,168,12,3,40,25,4,33,168,19,5,33,58,0,6,36,1,0,0,0,0,8,33,29,1,10,33,72,53,12,32,1,100,16,0]},"252":0}},"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"epId":2,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"epId":3,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"epId":4,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"5":{"epId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"6":{"epId":6,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":29,"stackVersion":2,"hwVersion":1,"dateCode":"20200324","swBuildId":"2019\u0000www.","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1697616457154,"defaultSendRequestWhen":"immediate"}

### Comments

I bought this from a Korean aqara official seller and it is certified under the name WXKG05LM.

jeong840, 2023-10-28:
I configured the external converter as shown below and it works without any problems. I will try to add code to check the battery status.

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        zigbeeModel: ['lumi.remote.b186acn03'],
        model: 'WXKG05LM',
        vendor: 'Xiaomi',
        description: 'Aqara T1 wireless switch',
        fromZigbee: [fz.xiaomi_on_off_action, fz.xiaomi_multistate_action],
        toZigbee: [],
        exposes: [e.action(['single','double','hold'])],
};

module.exports = definition;`

closed 2023-10-28 by jeong840

Koenkk, 2023-10-29:
Does battery work with: https://gist.github.com/Koenkk/be39ac8d30be8f5bab5f25fab98d1210 ?

jeong840, 2023-10-30:
> Does battery work with: https://gist.github.com/Koenkk/be39ac8d30be8f5bab5f25fab98d1210 ?

It works perfectly. Thank you 👍

Koenkk, 2023-10-30:
Added, thanks!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)