Koenkk/zigbee2mqtt #19334: [New device support]: CK-BL702-SWP-01(7020)

sixhills, 2023-10-18:
### Link

https://www.ebay.co.uk/itm/314855773062

### Database entry

{"id":20,"type":"Router","ieeeAddr":"0x7cb94c7736db0000","nwkAddr":15251,"manufId":4742,"manufName":"eWeLink","powerSource":"Unknown","modelId":"CK-BL702-SWP-01(7020)","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":9,"inClusterList":[0,3,4,5,6,4096,64599,64529],"outClusterList":[25,2820,10],"clusters":{"64529":{"attributes":{"28676":0,"28677":238803,"28678":0}},"genBasic":{"attributes":{"modelId":"CK-BL702-SWP-01(7020)","manufacturerName":"eWeLink","powerSource":0,"zclVersion":8,"appVersion":1,"stackVersion":0,"hwVersion":0,"dateCode":"20210610-BL","swBuildId":"1.0.1"}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00238dad65","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b00238dad65","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0,"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":0,"hwVersion":0,"dateCode":"20210610-BL","swBuildId":"1.0.1","zclVersion":8,"interviewCompleted":true,"meta":{"configured":1466202208},"lastSeen":1697636915231,"defaultSendRequestWhen":"immediate"}

### Comments

This is a Woolley smart plug with energy monitoring.

I tried to follow the "Support new devices" page but I'm unsure how to specify the "fromZigbee" and "toZigbee" lists.

Can you please include this device?

### External converter

```shell
// Incomplete attempt at an external converter
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['CK-BL702-SWP-01(7020)'],
    model: 'CK-BL702-SWP-01(7020)',
    vendor: 'eWeLink',
    description: 'BL702 Smart plug',
    fromZigbee: [fz.on_off, fz.command_on_state, fz.command_off_state, fz.electrical_measurement],
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.switch()],
    // The configure method below is needed to make the device reports on/off state changes
    // when the device is controlled manually through the button on it.
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        await reporting.onOff(endpoint);
    },
};

module.exports = definition;
```

ELKER54, 2023-10-21:
I would love to have this device supported. I just got 3 of these and hope that they will work soon.

Axelf-alex, 2023-10-25:
Yes please got 3 would like to get them working too, they are not much different from already supported one BSD29
 @Koenkk can we get your expertise on this please would really appreciate it.

englishteeth, 2023-10-25:
I've it running with a local _external converter_ with the following definition, cribbed and copied from other devices, so no idea if it's the best way. 
I have been trying to work out the power management data before I even thought about a PR, but it might keep you going.

```
const definition = {
    zigbeeModel: ['CK-BL702-SWP-01(7020)'],
    model: 'BSD29',
    vendor: 'Wooley',
    description: 'Zigbee 3.0 smart plug 13A (UK version)',
    fromZigbee: [fz.on_off_skip_duplicate_transaction],
    toZigbee: [tz.on_off],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        await reporting.onOff(endpoint);
    },
    exposes: [e.switch()],
};
```

Koenkk, 2023-10-25:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/2414b6ba5f911f02a67c7bc07b2b6d81

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

Axelf-alex, 2023-10-25:
I knew you would fix it straight away @Koenkk thank you so much absolute legend :) . 

The only problem is I can only turn it on and off these ones have power monitoring function I don't know what to include in the config to enable this 

[image]

Weirdly this device is 
CK-BL702-SWP-01(7020) but on the plug plastic and booklet mentions BSD29 lol
So not quite the same I would say. 

Any way around this? 

Thank you so much once again

Koenkk, 2023-10-25:
When in the frontend, you click on the device -> configure button (yellow refresh icon), does this succeed?

Axelf-alex, 2023-10-25:
I have added it couple of times just in case pairing was a bit flaky but made no difference then pressed the yello "Reconfigure" made no difference again :( 
Got errors tho 
1st Screenshot when added ---- 2nd Whne "Reconfigure" 
[image]
[image]

sixhills, 2023-10-25:
I've also tried the new external converter and I'm seeing the same effect as @Axelf-alex: pressing the yellow 'Reconfigure' button results in a timeout.

sixhills, 2023-10-25:
I'm not sure if the following helps, but I recorded this in the log when first connecting the device using the eternal converter listed at the start of this thread:
Oct 22 19:34:34 <host> `Zigbee2MQTT:info  2023-10-22 19:34:34: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"BL702 Smart plug",

"exposes":[{"features":[{"access":7,"description":"On/off state of the switch",
"label":"State","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],
"type":"switch"},{"access":1,"description":"Sum of consumed energy",
"label":"Energy","name":"energy","property":"energy","type":"numeric","unit":"kWh"},
{"access":1,"description":"Instantaneous measured power",
"label":"Power","name":"power","property":"power","type":"numeric","unit":"W"},
{"access":1,"description":"Measured electrical potential value",
"label":"Voltage","name":"voltage","property":"voltage","type":"numeric","unit":"V"},
{"access":1,"description":"Instantaneous measured electrical current",
"label":"Current","name":"current","property":"current","type":"numeric","unit":"A"},
{"access":1,"description":"Link quality (signal strength)",
"label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],
"model":"CK-BL702-SWP-01(7020)",

"options":[{"access":2,"description":"State actions will also be published as 'action' when true (default false).",
"label":"State action","name":"state_action","property":"state_action","type":"binary","value_off":false,"value_on":true},
{"access":2,"description":"Calibrates the power value (percentual offset), takes into effect on next report of device.",
"label":"Power calibration","name":"power_calibration","property":"power_calibration","type":"numeric"},
{"access":2,"description":"Number of digits after decimal point for power, takes into effect on next report of device.",
"label":"Power precision","name":"power_precision","property":"power_precision","type":"numeric","value_max":3,"value_min":0},
{"access":2,"description":"Calibrates the current value (percentual offset), takes into effect on next report of device.","label":"Current calibration","name":"current_calibration","property":"current_calibration","type":"numeric"},
{"access":2,"description":"Number of digits after decimal point for current, takes into effect on next report of device.",
"label":"Current precision","name":"current_precision","property":"current_precision","type":"numeric","value_max":3,"value_min":0},
{"access":2,"description":"Calibrates the voltage value (percentual offset), takes into effect on next report of device.",
"label":"Voltage calibration","name":"voltage_calibration","property":"voltage_calibration","type":"numeric"},
{"access":2,"description":"Number of digits after decimal point for voltage, takes into effect on next report of device.",
"label":"Voltage precision","name":"voltage_precision","property":"voltage_precision","type":"numeric","value_max":3,"value_min":0},
{"access":2,"description":"Number of digits after decimal point for energy, takes into effect on next report of device.",
"label":"Energy precision","name":"energy_precision","property":"energy_precision","type":"numeric","value_max":3,"value_min":0},
{"access":2,"description":"Calibrates the energy value (percentual offset), takes into effect on next report of device.",
"label":"Energy calibration","name":"energy_calibration","property":"energy_calibration","type":"numeric"}],"supports_ota":false,"vendor":"eWeLink"},

"friendly_name":"0x7cb94c77d5860000","ieee_address":"0x7cb94c77d5860000","status":"successful","supported":true},
"type":"device_interview"}'
...
Oct 22 19:34:34 <host> npm[17739]: Zigbee2MQTT:info  2023-10-22 19:34:34: Configuring '0x7cb94c77d5860000'
Oct 22 19:34:34 <host> npm[17739]: Zigbee2MQTT:info  2023-10-22 19:34:34: Successfully configured '0x7cb94c77d5860000'`

Axelf-alex, 2023-10-26:
I tried to read and try to do things from the GUI of Z2m I noticed in the code there is 

        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);
        await reporting.onOff(endpoint);
        await reporting.readEletricalMeasurementMultiplierDivisors(endpoint);
        await reporting.activePower(endpoint);
        await reporting.rmsVoltage(endpoint);
        await reporting.rmsCurrent(endpoint);

But nothing comes up automatically after adopting device I tried it manually its goes well until I get to the reporting part.

[image]

At the reporting part the readEletricalMeasurementMultiplierDivisors(endpoint) is missing also I tried the rmsVoltage and activePower and it fails 

[image]

[image]

Any help as always would be greatly appreciated. @Koenkk I think if we fix the haElectricalMeasurement we might fix the reporting. 

Thank you in advance for all the help and support :)

Koenkk, 2023-10-27:
Can you try removing `await reporting.rmsVoltage(endpoint);`

sixhills, 2023-10-27:
I've tried that and it no longer reports a failure to configure.

Clicking the "Reconfigure" button generates:

Oct 27 19:44:06 pi42 npm[10603]: Zigbee2MQTT:info  2023-10-27 19:44:06: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"WL1_plug"},"status":"ok","transaction":"ybk5z-3"}'

so that's welcome progress, but there's still no reporting of power, either in the log or on the dashboard.

Thanks for sticking with this - it's much appreciated.

dangnguyen0906, 2023-10-28:
i got the same issues of tracking power measuring . Please help to add this devices as supported in next version of z2m

Koenkk, 2023-10-28:
Does something shows up in the frontend -> device -> reporting tab?

Axelf-alex, 2023-10-28:
@Koenkk unfortunately nothing comes up
[image]

sixhills, 2023-10-28:
Just for info: for people struggling with this smart plug, I took a risk on very similar devices from Aliexpress:
[https://www.aliexpress.com/item/1005005362261536.html](url)
They arrived in the UK in 6 days and "just work" with Z2M.  They identify as Tuya BSD29 devices and power monitoring works fine.

Koenkk, 2023-10-29:
@Axelf-alex that indeed doesn't look good, based on what you provided in https://github.com/Koenkk/zigbee2mqtt/issues/19334#issuecomment-1781347783 (manually setting the reporting also fails), I guess the device doesn't support it or there is some non standard way of setting this up. To check both, you need the original hub and check if power measurements work there, if yes, you can [sniff](https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html) the traffic which reveals how to setup the reporting correctly.

Axelf-alex, 2023-10-29:
@Koenkk Thank you so much for your help and support with this. I guess expensive e-waste then..!! 😂
 I guess if ZHA works with it can we not sniff the traffic from there? Will try couple of things and if I manage to find a solution will post here

sixhills, 2023-10-29:
At least they can be used as MQTT-controlled switches, so not a total loss. I wondered whether to buy a Sonoff hub ( which it claims to work with) to investigate further, but it's probably not worth it when there are other power-monitoring plugs that work.
My thanks also to @Koenkk for all his help.

englishteeth, 2023-10-29:
Not fully investigated yet, but I was looking at the data for the unhandled messages and have this draft converter:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    who_knows: {
        cluster: '64529',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            if (utils.hasAlreadyProcessedMessage(msg, model)) return;
            const lookup = [
                {key: '28678', name: 'power', factor: 'acPower'},
                {key: '28677', name: 'voltage', factor: 'acVoltage'},
                {key: '28676', name: 'current', factor: 'acCurrent'},
            ];
            const payload = {};
            for (const entry of lookup) {
                if (msg.data.hasOwnProperty(entry.key)) {
                    const property = utils.postfixWithEndpointName(entry.name, msg, model, meta);
                    const value = msg.data[entry.key] / 1000;
                    payload[property] = utils.calibrateAndPrecisionRoundOptions(value, options, entry.name);
                }
            }
            return payload;
        },
    },
};

const definition = {
    zigbeeModel: ['CK-BL702-SWP-01(7020)'],
    model: 'BSD29',
    vendor: 'Woolley',
    description: 'Zigbee 3.0 smart plug 13A (UK version)',
    fromZigbee: [fz.on_off_skip_duplicate_transaction, fzLocal.who_knows],
    toZigbee: [tz.on_off],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        await reporting.onOff(endpoint);
        device.powerSource = 'Mains (single phase)';
        device.save();
    },
    exposes: [e.power(), e.current(), e.voltage(), e.switch()],
};

module.exports = definition;

```
Totally non-standard, but the numbers output look plausible.

As said above, they still work as switches and repeaters. :)

Koenkk, 2023-10-29:
@englishteeth I wasn't aware that the device reports on a non-standard cluster, your code looks good, if others can confirm it works correctly you can make a PR.

sixhills, 2023-10-29:
@englishteeth Thanks so much for that code.  I don't understand it but I'm delighted to report that it works!
The Z2M log is reporting:

`Oct 29 19:06:02 pi42 npm[10579]: Zigbee2MQTT:info  2023-10-29 19:06:02: MQTT publish: topic 'zigbee2mqtt/WL1_plug', payload '{"current":0.08,"linkquality":138,"power":9.37,"state":"ON","voltage":236.58}'
`

The dashboard reports the same values.  The reported values are extremely plausible, so I think you've fixed the problem.  Thank you so much.

sixhills, 2023-10-29:
@englishteeth One last query: it looks as though it should also report energy used so far in kWh.  Have you found a key that would extract that value?  If that's possible, it would be great to have a way of resetting energy to zero but I appreciate that that may be a step too far.  Thanks again for your excellent work on this.

Axelf-alex, 2023-10-29:
@englishteeth You sir deserve a medal absolute legend thank you so much it was amazingly well. I tested even a hair dryer and the numbers look plausible. 

Next Z2M patch will probably have this in support for other users to have an easier time 😊

englishteeth, 2023-10-30:
Thanks for the kind words and number sanity checks, I'll look into how to raise a PR for this as it stands.

@sixhills - I only found the '64529' cluster because it was reporting by default so, so far I've not been able to get the plug to do anything else. There is another curious cluster '64599' but it might really need a compatible hub  to find out what's going on.

Koenkk, 2023-10-30:
@englishteeth added the device, will be supported out of the box in the 1 November release.

Delliaf, 2023-10-31:
This is the same, external converter is working
[image]
[image]

https://sl.aliexpress.ru/p?key=qQpBUci

sixhills, 2023-11-01:
Closing as this device is now supported by Z2M.

closed 2023-11-01 by sixhills

Mikes-HAAS, 2023-11-23:
Works for me thats is all I needed as bought a pack of 4.
Regards Mike

treemonkey83, 2024-01-01:
Has anyone experienced excessive publishing from these plugs. It's reporting it's state every few seconds. Looking at the debug level log it's pushing

Received Zigbee message from 'LR wall', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0

Over and over. Denounce of any time doesn't seem anything.

Tazman12345, 2024-01-17:
Hello
i tryed to setup the external converter from Koenkk and restart z2Mqtt. I'm an absolute beginner and apparently I messed up. z2mqtt no longer starts and shows the following errors:

Error: Cannot find module 'const fz = require('zigbee-herdsman-converters/converters/fromZigbee'); const tz = require('zigbee-herdsman-converters/converters/toZigbee'); const exposes = require('zigbee-herdsman-converters/lib/exposes'); const reporting = require('zigbee-herdsman-converters/lib/reporting'); const legacy = require('zigbee-herdsman-converters/lib/legacy'); const extend = require('zigbee-herdsman-converters/lib/extend'); const ota = require('zigbee-herdsman-converters/lib/ota'); const tuya = require('zigbee-herdsman-converters/lib/tuya'); const utils = require('zigbee-herdsman-converters/lib/utils'); const globalStore = require('zigbee-herdsman-converters/lib/store'); const e = exposes.presets; const ea = exposes.access;  const definition = {     zigbeeModel: ['CK-BL702-SWP-01(7020)'],     model: 'BSD29',     vendor: 'Wooley',     description: 'Zigbee 3.0 smart plug 13A (UK version)',     fromZigbee: [fz.on_off_skip_duplicate_transaction, fz.electrical_measurement],     toZigbee: [tz.on_off],     configure: async (device, coordinatorEndpoint, logger) => {         const endpoint = device.getEndpoint(1);         await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);         await reporting.onOff(endpoint);         await reporting.readEletricalMeasurementMultiplierDivisors(endpoint);         await reporting.activePower(endpoint);         await reporting.rmsVoltage(endpoint);         await reporting.rmsCurrent(endpoint);     },     exposes: [e.switch(), e.power(), e.voltage(), e.current(),], };;  module.exports = definition;'
Require stack:
- /opt/zigbee2mqtt/dist/util/utils.js
- /opt/zigbee2mqtt/dist/util/settings.js
- /opt/zigbee2mqtt/index.js
    at Function.Module._resolveFilename (node:internal/modules/cjs/loader:1028:15)
    at Function.Module._load (node:internal/modules/cjs/loader:873:27)
    at Module.require (node:internal/modules/cjs/loader:1100:19)
    at require (node:internal/modules/cjs/helpers:119:18)
    at Object.getExternalConvertersDefinitions (/opt/zigbee2mqtt/lib/util/utils.ts:171:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/opt/zigbee2mqtt/lib/extension/externalConverters.ts:12:20)
    at new Controller (/opt/zigbee2mqtt/lib/controller.ts:84:58)
    at start (/opt/zigbee2mqtt/index.js:106:18)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)

Can I somehow get this to work again?
Thank you in advance for your help