Koenkk/zigbee2mqtt #19342: Xiaomi/Aqara Thermostat SRTS-A01: Setting new local temperature turns thermostat to "heat", but state still reported as "off"

Ra72xx, 2023-10-19:
### What happened?

Thermostat is turned off. 
Setting a new local temperature with Z2M ui or HA turns the radiator to "heat" (as visible on the display).
However, both Z2M and consequently HA still report the radiator state as "off", which is wrong.
Firmware of the thermostat is 0.0.0_01030

### What did you expect to happen?

Setting the local temperature when thermostat is "off" should either
- set the local temperature but keep the thermostat off nevertheless, or
- set the local temperature and update the thermostat state to "heat" to reflect the real hardware state
The first variant is preferable IMHO, as it allows better integration with "window off" functions in combination with heating schedules as provided by HA custom components.

### How to reproduce it (minimal and precise)

Turn the thermostat off and set a new local temperature.

### Zigbee2MQTT version

1.33.1

### Adapter firmware version

7.1.1.0 build 273

### Adapter

SkyConnect (EZSP v9)

Ra72xx, 2023-10-21:
Looking further into it, there seems to be a problem with feedback of some changes in Z2M.
As said above, setting a temperature turns the thermostat on, this is not recognized by Z2M.
Under some conditions which I don't really understand doing changes on the thermostat is also not recognized by Z2M.

helgek, 2023-11-23:
Having a similar issue, randomly one of my devices would not respond to commands coming from Z2M, results in a timeout error:https://github.com/Koenkk/zigbee2mqtt/issues/13993#issuecomment-1824743866 - coordinator is the highly praised SLZB-0 (based on stable chipset CC2652P  recommended by Z2M, not the experimental EFR32MG21 chipset).

Sugaroverdose, 2024-01-26:
SRTS-A01 does not provides actual valve state according to https://github.com/Koenkk/zigbee2mqtt/issues/13993#issuecomment-1744604199 (i didn't check whether it does not actually have valve position data), meanwhile, converter maps "system_mode" to stuff which in aquara software is most definitely called "antifreeze" which has "heat" and "off" options, from what i see "off" should actually be called "automatic"(because it does not forces valve to be closed if it's conditions to be opened are met) while "heat" forces valve to get fully open whatever temperatures is set, so this issue is about assumingly wrong converter mapping

closed 2024-08-27 by github-actions[bot]