Koenkk/zigbee2mqtt #19343: [Wrong device]:

Jordo-o, 2023-10-19:
### Link

https://miboxer.com/product/rgb-led-controller-zigbee-3-0

### Model

FUT037Z

### Description

RGB Led controller

### Vendor

MiBoxer

### Picture (link)

miboxer.com/wp-content/uploads/fut037z.jpg

### Database entry

{"id":4,"type":"Router","ieeeAddr":"0xa4c13840d0e4c2a1","nwkAddr":43705,"manufId":4417,"manufName":"_TZ3210_dbilpfqk","powerSource":"Mains (single phase)","modelId":"TS0501B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[3,4,5,6,4096,8,768,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0501B","manufacturerName":"_TZ3210_dbilpfqk","powerSource":1,"zclVersion":3,"appVersion":98,"stackVersion":0,"hwVersion":1,"dateCode":""}},"lightingColorCtrl":{"attributes":{}},"genLevelCtrl":{"attributes":{"currentLevel":254}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{"onLevelSupported":false}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":98,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1697699568440,"defaultSendRequestWhen":"immediate"}

cajonKA, 2023-10-27:
+1 from me
Can confirm the above mentioned MiBoxer Mi-Light Led RGB Controller is shown as Tuya TS0503B in Zigbee2MQTT

Koenkk, 2023-11-08:
Can you provide a screenshot of the z2m frontend -> device -> about tab with z2m 1.33.2?

Jordo-o, 2023-11-08:
Is this what you mean?

[image]

Koenkk, 2023-11-08:
Fixed!

Changes will be available in the dev branch tomorrow. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-11-08 by Koenkk

AnderVentura, 2024-03-21:
I'm getting this error using Zigbee2MQTT 1.36.0-1 in Home Assistant 2024.3.1. Any way I can fix it?