Koenkk/zigbee2mqtt #19345: Miboxer FUT03xZ devices seem to have problems with action command

Jordo-o, 2023-10-19:
### What happened?

The devices (in my case FUT036Z and FUT037Z) seem to have problems with the ON state command.
After being turned off for a while, the devices respond in a strange way to the ON command.

After turning it on, they immediately go off again. Although, according to the device state they are still turned on to the set brightness. You have to send a brightness value again or even turn it off first before it turns on again.
After reviewing the log, it seems that the MQTT topic is also triggered twice, once ON and once OFF.

It looks like the light stays on when sending only the brightness and not the state ON command.
But I am not completely sure about the fix yet.

### What did you expect to happen?

The light should stay on after the first command.

### How to reproduce it (minimal and precise)

1. Wait for a few hours... unfortunately, the device doesn't show this problem shortly after successfully turning it on.
2. Send on command via the zigbee2mqtt frontend or for example GoogleHome via OpenHAb or HomeAsisstant.
3. See the light turning off again.

### Zigbee2MQTT version

1.33.1-1

### Adapter firmware version

20221226

### Adapter

Slaesh's CC2652RB Stick

### Debug log

debug 2023-10-19 09:56:08: Received MQTT message on 'zigbee2mqtt/RGB TV (1)/set' with data '{"state":"ON"}'
debug 2023-10-19 09:56:08: Publishing 'set' 'state' to 'RGB TV (1)'
info  2023-10-19 09:56:08: MQTT publish: topic 'zigbee2mqtt/RGB TV (1)', payload '{"brightness":206,"color":{"x":0.422,"y":0.52},"color_mode":"xy","color_power_on_behavior":"previous","do_not_disturb":true,"last_seen":"2023-10-19T07:56:08.986Z","linkquality":98,"state":"OFF"}'
info  2023-10-19 09:56:08: MQTT publish: topic 'zigbee2mqtt/RGB TV (1)', payload '{"brightness":206,"color":{"x":0.422,"y":0.52},"color_mode":"xy","color_power_on_behavior":"previous","do_not_disturb":true,"last_seen":"2023-10-19T07:56:08.986Z","linkquality":98,"state":"ON"}'

lux73, 2023-10-22:
i can second this - few Month before i buy FUT039Z RGB+CCT LED Controller

i have some serious Problems with it - state Command leave untrue as you mentoined

BUT this Device also blocks my whole Zigbee Network after few Weeks - so no Commands to other Zigbee Devices could made

in my opinion these Miboxer Devices are crap, sorry...  they are cheap but unreliable

Jordo-o, 2023-10-27:
I indeed noticed that it doesn't matter to leave the state command away. Although, it looks a bit better without then with the state command. 

But I have 4 of them and the don't block my network. But maybe that's because all the devices can take another route to the controller. 

But indeed, they are cheap and a bit crappy....
For now I will send the command twice then 🤷🏽

emandtf, 2023-11-04:
I'm using 6 FUT036Z from 2 years without any issue (beside the "_do_not_disturb_" feature that not always works leaving the dimmer ON after an AC power outage)

Jordo-o, 2023-11-04:
Okay, that's weird I think. 
Maybe different firmware or something? Only, I can't see the version of these devices.... :/

emandtf, 2023-11-04:
> Maybe different firmware or something? Only, I can't see the version of these devices.... :/
If you are referring to multiple device types, I doubt firmware is the cause.
Are you sure to not have some Homeassistant (or whichever smart software) Automation that does that?

Jordo-o, 2023-11-04:
There you have a good point 😅
I also ruled out a power problem. 

I have had it with different platforms, homeassistant, OpenHAB and even with the Z2M UI. So I really can't find the problem. But it only happens after the lights being off for a longer while. 

Solution is pressing the on button twice, but it is kinda annoying 😅

joconcepts, 2023-11-06:
I have the same issues with them which already caused me to order another LED dimmer model (which is not yet here). But it is very annoying, I'm using it with a motiondetector and another one with homekit using homebridge and both show the same weird behavior (turning on, then instantly turning off, etc).

If the new dimmer will fix the issue I will replace them, however I'm very interested in knowing the reason for that behavior.

Jordo-o, 2023-11-27:
> I recently purchased a FUT037Z device, but it is not compatible with Z2M. Where can I find an external converter?
> 

The controller should be supported.
Maybe you can reset the controller (button on top) and try to connect it again.

alexsahka, 2023-11-27:
> The controller should be supported. Maybe you can reset the controller (button on top) and try to connect it again.

I bought three FUT037Z devices, and they all behave identically.

Jordo-o, 2023-11-28:
Well, then its not the device. 
Can you maybe check your router firmware version and Z2M version?

alexsahka, 2023-11-28:
All of my devices are working correctly on the Z2M hub except for the FUT037Z. 

Zigbee2MQTT version
1.33.2-dev commit: [ef00a64](https://github.com/Koenkk/zigbee2mqtt/commit/ef00a64)

Coordinator type
zStack3x0

Coordinator revision
20220103

alexsahka, 2023-11-28:
Does yours look like this?

[image]

alexsahka, 2023-11-28:
> The controller should be supported. Maybe you can reset the controller (button on top) and try to connect it again.

I don't see the FUT037Z on the supported list of devices

[image]

paranoiq1, 2024-01-24:
I have similar problem with my two FUT039Z. I turn them on, then they immediately go off. State remains 'on'. Neither brightness change, nor turning them on once more will help, but yesterday they both started to work after few hours of not touching.
I run zigbee2mqtt 1.35.0-1

Jordo-o, 2024-01-24:
> Does yours look like this? [...]

Sorry, for the late reply 😅

But no, mine was without WW and CW. Note that you have de FUT037Z+. Mine is without +. That's probably the problem. 

I'm afraid that you need to make converter yourself.

Jordo-o, 2024-01-24:
> I have similar problem with my two FUT039Z. I turn them on, then they immediately go off. State remains 'on'. Neither brightness change, nor turning them on once more will help, but yesterday they both started to work after few hours of not touching. I run zigbee2mqtt 1.35.0-1

I have fixed it by sending the on command twice. First one without a color/temperature. Second one with color and correct brightness (Few ms delay). 

Isn't great, but it works.

paranoiq1, 2024-01-24:
> I have fixed it by sending the on command twice.

Have you tried turning off DND mode? I read that this actually should help, because those devices have this "functionality" that after power outage you need to click twice to turn them on

Jordo-o, 2024-01-24:
> > I have fixed it by sending the on command twice.
> 
> Have you tried turning off DND mode? I read that this actually should help, because those devices have this "functionality" that after power outage you need to click twice to turn them on

Yes, DND mode has been off for a while now. Still doesn't help :(

Jordo-o, 2024-01-29:
Currently I have requested support from MiBoxer.
There are not so fast with a solution, but hopefully, it's on its way :)

paranoiq1, 2024-01-29:
Ok, so in my case turning light on twice did not help, but turning it on and then setting new brightness did  the trick or at least I hope it did

Jordo-o, 2024-01-29:
> Ok, so in my case turning light on twice did not help, but turning it on and then setting new brightness did the trick or at least I hope it did

That should also work indeed. It looks like there is some bug with the device not handling a ON command and brightness/color together. So, sending with a few milliseconds delay should do the trick, although it's not so nice.

I'm afraid they won't come with a solution.
But if they do, I will let you all know :)

Jordo-o, 2024-02-03:
I have had a lot of email contact with MiBoxer. They have tested some things on their side and I have tested some things with my controllers. 

But unfortunately, they conclude that the problem is in the gateway :(. Soo, I'm afraid they can't help us any further.

xeroski, 2024-03-12:
Hi. I've probably found a reason why this is happening. If Miboxer is turned off and I send the turn-off command again, it turns off immediately after turning it on. This may happen if we send a global command to turn off all lights and Moboxer is already turned off

Jordo-o, 2024-03-13:
So, it will sort of remember that it was turned off, while being off?
Sounds like a useless feature to me haha, but if that's the case, it should be possible to fix.

Jordo-o, 2024-03-18:
I have tested some of my devices and can confirm that this is indeed can be the problem.
When I turned off my lights once, the light turned on normally. When I turned it off twice (or more) it went off directly after receiving the ON command...

Hope MiBoxer can help us witht he problem...

keteflips, 2024-09-08:
> I have tested some of my devices and can confirm that this is indeed can be the problem. When I turned off my lights once, the light turned on normally. When I turned it off twice (or more) it went off directly after receiving the ON command...
> 
> Hope MiBoxer can help us witht he problem...
> 
> **Edit:** MiBoxer support keeps telling me that our gateway is not compatible and that they won't give us any firmware updates. The only advice I get is buying a MiBoxer gateway....

Any news?

I thinks my problem its firmware related https://github.com/Koenkk/zigbee2mqtt/issues/23793

Jordo-o, 2024-09-08:
> 
> Any news?
> 
> I thinks my problem its firmware related #23793

No news, unfortunately. MiBoxer won't help us and tells us to buy an MiBoxer gateway to "fix" the problem. They say it doesn't happen with their test setup. 

You have to make a workaround outside zigbee2mqtt to fix the problem... It looks like the problem is in the firmware of the controller.

xeroski, 2024-09-08:
That's why I won't buy anything from Miboxer again. Let them live in their own closed world

0BanZai0, 2024-11-01:
Oh no, I'm stuck with a Miboxer controller that I can't change... I have the exact same problem.

Isn't there a way to "ping" the device frequently from z2m to receive the correct state? 

I might also be willing to buy the Gateway for the FW update to solve this...

0BanZai0, 2024-11-12:
https://github.com/user-attachments/assets/b9a1b9f9-7aaa-4b0a-b9c8-d5852162a8b4

I'm in contact with miboxer. But they can't figure out where the problem is.

This is what I wrote them right now together with a attached video.

```
...attached you’ll find a video that shows what’s happening.

1.	First press of my Zigbee switch: nothing happens with the light. But the status oft he light in „Zigbee2mqtt“ switches to „off“.
2.	Second press of the switch: the light turns on for a split second. In „Zigbee2mqtt“ the light status is changed to „on“.
3.	Third and following presses will cycle this behaviour.

This shows that the state of the light is not given to Z2M correctly as it should.
I can only leave this state, when I manually change the brightness.

It will then happen again, when the lamp gets the „turn off“ command two or more times in a row.
```

The Miboxer Gateway didn't help, as there was no FW update for the controllers.

I'm thinking about two possible workarounds:

1. Enable the "availability" check in Z2M for this lamp.
2. Build some kind of automation, that will change the brighness setting somehow after the "on" command is sent... somehow

Help here would be greatly appreciated :)

0BanZai0, 2024-11-15:
I've got a response from miboxer that I want to share with you.
This does not solve the issue, but at least confirms it.
I'd be very helpful for good ideas on how to build a workaround / script.

`I asked the engineer to record the operation video for you, please see the attachment.

There is no problem with ON and OFF.

But when you turn it off and then adjust the brightness to 0, and then adjust it to another brightness, the light will go out. This is a hardware problem, and we will upgrade it later.

The current solution is not to automatically set the brightness to 0 and then set another brightness. Avoiding such operations will not cause this problem.
`

0BanZai0, 2024-11-21:
I think I've found the solution after cross reading this: https://github.com/Koenkk/zigbee2mqtt/issues/22030

It seems that having set a "transition" setting for the FUT036Z causes the problem.
Maybe the controller has a problem with storing the last known brightness and state somehow.

Solution is to leave the "transition" field blank in Z2M.

@Jordo-o @keteflips - would be great if you could verify this :)

0BanZai0, 2024-11-24:
I've cheered too soon... It's back. Kind of different, but back.
I cannot reproduce it yet. Maybe it's triggered by some kind of "good night" automation.
I'll report back :)

0BanZai0, 2024-11-28:
@Jordo-o  - I've got it! It is the transition setting. 
Not just in the device config itself, but in the group config as well.
I've had this device as a part of a Z2M group that I use to switch off all lights on a floor together.
Because I'm using Homekit a lot and was annoyed that homekit does not allow smooth dimming for groups, I've put the "transition: x"-setting in the configuration.yaml of Z2M.
This group-transition setting was the one that caused the bug on my Miboxer device again.

So my verdict:
- Don't use any transition setting in the device settings (leave it blank, not zero)
- Don't put the device in a group with a transition setting or remove the setting from the group

Is there any way to "hardcode" this in Z2M to ignore any transition settings so that I can keep it in my group?

Jordo-o, 2024-11-28:
@0BanZai0 

I have tested it too, and it looks like it is working for my devices as well! I had indeed changed the transition setting to "empty" instead of 0 and also removed transitions from HA. 

The devices still does some transition on its own, without sending any transition to it, so I'm happy with the solution this way. 

Took some time to get it working, but it looks like we finally got a solution. I still have to test the rest of the controllers, but it looks good! 😁

AMArosendaal, 2025-02-10:
Hi guys, 

I also had this annoying problem, and thought i did something wrong in the automation.
Also i could not reproduce it at first. So i've been living with this unreliable behavior for over a year.
Now i switched over from Philips Hue bridge to an general Zigbee hub and the problem became more apparent. 

After reading this post i can confirm and reproduce it was cause by using **transition**. 
Apparently when using a transition it counts down to brightness 0. Then after turning the light on, it quickly shuts off again.

I removed al forms of transition in my automations, and until now it seems to behave correct. Fingers crossed.

keteflips, 2025-02-25:
Now mine its recogniced as Tuya [TS0505B_1](https://www.zigbee2mqtt.io/devices/TS0505B_1.html#tuya-ts0505b_1)

0BanZai0, 2025-02-26:
> Now mine its recogniced as Tuya [TS0505B_1](https://www.zigbee2mqtt.io/devices/TS0505B_1.html#tuya-ts0505b_1)

How? Mine still is listed as FUT036Z

johndf, 2025-04-07:
I’m also getting TS0505B instead of FUT037Z.  Any suggestions on how to force it?  Repairing doesn’t solve it.

Jordo-o, 2025-04-07:
Sometimes they are indeed losted as TS0505B, they are just rebrands after all. 
Everything should work fine as TS0505B, but it should be fixed since this release https://github.com/Koenkk/zigbee-herdsman-converters/pull/6439. 
Maybe you can remove the device and re-pair it.

paranoiq1, 2025-04-30:
I have two FUT039Z and for some reason I cannot pair them anymore after moving to new hub and repair with old hub. Even if after many retries their pair correctly they are unreliable and I have problem to even turn them on. Thinking about switching to other controller but not sure what would be reliable and support 5 channel. Though they do worka on Tuya hub

Zigbee2MQTT version
[2.2.1]
Coordinator type
zStack3x0
Coordinator revision
20240710

closed 2025-07-07 by github-actions[bot]