Koenkk/zigbee2mqtt #19351: [Wrong device]: AVATTO Tuya ZigBee Smart Thermostat Low Power Battery Version

Liionboy, 2023-10-19:
### Link

https://www.aliexpress.com/item/1005005822896810.html?spm=a2g0o.order_list.order_list_main.5.2efd1802haJd6J

### Model

ZWT198

### Description

Thermostat

### Vendor

AVATTO

### Picture (link)

https://ae01.alicdn.com/kf/S9d1b77a3456642c3986d50e9b788c8f9p/AVATTO-Tuya-ZigBee-Smart-Thermostat-Low-Power-Battery-Version-Water-Gas-Boiler-Temperature-Controller-Alexa-Google.jpg_80x80.jpg

### Database entry

{"id":5,"type":"EndDevice","ieeeAddr":"0x3425b4fffe4c52d1","nwkAddr":4994,"manufId":4098,"manufName":"_TZE200_viy9ihs7","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"+\u001f�,i","65506":31,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE200_viy9ihs7","stackVersion":0,"dateCode":"","zclVersion":3,"appVersion":65,"powerSource":3}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1440156621},"lastSeen":1697740964265,"defaultSendRequestWhen":"immediate"}

### Notes

AVATTO Thermostat is seen like ONNDO smart thermostat and Temperature Calibration don't work:

[image]

[image]

Liionboy, 2023-10-20:
Is the same like this : https://www.zigbee2mqtt.io/devices/X5H-GB-B.html

Only this is power by battery

Koenkk, 2023-10-20:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/c511c3857ee9ab474923b228bce07e1d

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

Liionboy, 2023-10-20:
Something is wrong because i have set Temperature when to start and now all thermostat start Heat even temperature is with +2

Liionboy, 2023-10-20:
This is error when i try to modify temperature or something on thermostat:

[image]

Liionboy, 2023-10-20:
This is thermostat:

https://www.aliexpress.com/item/1005005822896810.html?spm=a2g0o.order_list.order_list_main.5.1a961802dTJSX1

If you want i can connect 1 on Tuya and tell me how i can extract data from Tuya application

Koenkk, 2023-10-21:
That means this thermostat works differently. To add support for this:
- Figure out the datapoints: https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html
- Create a new converter, you can take a look at existing thermostat converters to start with (e.g. [this one](https://github.com/Koenkk/zigbee-herdsman-converters/blob/ffbfd19989cee66fae9a2df959b6dce435c528c8/src/devices/tuya.ts#L2705)

Liionboy, 2023-10-21:
I extract all of them but now i dont know how to put in a convert file :D :

{"1":"Switch","2":"Set temperature","3":"Current temperature","4":"Mode","9":"Child lock","11":"Fault alarm","15":"Set temperature upper limit","16":"Lower Temp","19":"Temperature correction","101":"Working status","102":"Frost protection function","103":"Reset","104":"Workday setting","107":"Output startup temperature difference","109":"Weekly procedure","110":"Backlight brightness"}

I will try or maybe you can help me faster then me :dancers:

Liionboy, 2023-10-21:
On Temperature calibration is from 0.1, 0.2, 0.3 ....... not and i think this is the problem and with -0.1, -0.2, -0.3 adn is from -9.9 to 9.0

Liionboy, 2023-10-21:
I don't knoow from where i can take function for convert

Liionboy, 2023-10-21:
i did a external convertor and change name:

[image]

Maybe on future will be implement all of expose because there are a lot in Tuya Application the list is below in 1 comment from me.

I saw in HASS Temp calibration is work, only in z2m are wrong number. Expose are same like that ONNDO Thermostat after i search on google only different is this is on Battery .

Thanks

Koenkk, 2023-10-22:
Once ready, please submit a pull request to update https://github.com/Koenkk/zigbee-herdsman-converters/blob/ca799050c761ff8268e57227db36e6beffe8d35c/src/devices/tuya.ts#L3927

Liionboy, 2023-10-25:
i dont know how to do a pull request so i will paste code here:

fingerprint: tuya.fingerprint('TS0601', ['_TZE200_viy9ihs7']),
model: 'ZWT198',
vendor: 'TuYa',
description: 'AVATTO Battery Wall-Mount Thermostat',
fromZigbee: [tuya.fz.datapoints],
toZigbee: [tuya.tz.datapoints],
onEvent: tuya.onEvent({timeStart: '2000'}),
configure: tuya.configureMagicPacket,
exposes: [
    e.child_lock(),
    e.climate()
        .withSystemMode(['off', 'heat'], ea.STATE_SET)
        .withPreset(['auto', 'manual', 'temporary program'])
        .withSetpoint('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
        .withRunningState(['idle', 'heat'], ea.STATE)
        .withLocalTemperature(ea.STATE)
        .withLocalTemperatureCalibration(-9.9, 9.9, 0.1, ea.STATE_SET),
    e.binary('frost_protection', ea.STATE_SET, 'ON', 'OFF')
        .withDescription('Antifreeze function'),
    e.numeric('upper_temp', ea.STATE_SET).withUnit('°C').withValueMax(95)
        .withValueMin(15).withValueStep(1).withPreset('default', 60, 'Default value')
        .withDescription('Maximum upper temperature'),
    e.numeric('deadzone_temperature', ea.STATE_SET).withUnit('°C').withValueMax(10)
        .withValueMin(0.5).withValueStep(0.5).withPreset('default', 1, 'Default value')
        .withDescription('The delta between local_temperature and current_heating_setpoint to trigger Heat'),
    // Not working yet:
    // e.enum('schedule_mode', ea.STATE_SET, ['disabled','weekday/sat+sun','weekday+sat/sun','7day'])
    //     .withDescription('Schedule mode')
],
meta: {
    tuyaDatapoints: [
        [1, 'system_mode', tuya.valueConverterBasic.lookup({'heat': true, 'off': false})],
        [2, 'current_heating_setpoint', tuya.valueConverter.divideBy10],
        [3, 'local_temperature', tuya.valueConverter.divideBy10],
        [4, 'preset', tuya.valueConverterBasic.lookup({'auto': tuya.enum(0), 'manual': tuya.enum(1), 'temporary program': tuya.enum(2)})],
        [9, 'child_lock', tuya.valueConverter.lockUnlock],
        [15, 'upper_temp', tuya.valueConverter.divideBy10],
        [19, 'local_temperature_calibration', tuya.valueConverter.divideBy10],
        [101, 'running_state', tuya.valueConverterBasic.lookup({'heat': tuya.enum(1), 'idle': tuya.enum(0)})],
        [102, 'frost_protection', tuya.valueConverter.onOff],
        [104, 'schedule_mode', tuya.valueConverterBasic.lookup(
            {'disabled': 0, 'weekday/sat+sun': 1, 'weekday+sat/sun': 2, '7day': 3},
        )],
        [107, 'deadzone_temperature', tuya.valueConverter.divideBy10],
        // These are the schedule values in bytes, 8 periods in total (4 bytes per period).
        // For each period:
        // 1st byte: hour
        // 2nd byte: minute
        // 3rd, 4th bytes: temperature multiplied by 10
        // Last 2 periods are ignored if schedule_mode is 7day. When schedule_mode is disabled,
        // scheduling can't be configured at all.
        // For example, if schedule_mode is weekday/sat+sun and this byte array is received:
        // [6,10,1,144,8,10,0,170,11,40,0,170,12,40,0,170,17,10,0,230,22,10,0,170,8,5,0,200,23,0,0,160]
        // Then the schedule is:
        // Mon-Fri: 6:10 --> 40C, 8:10 --> 17C, 11:40 --> 17C, 12:40 --> 17C, 17:10 --> 23C, 22:10 --> 17C
        // Sat-Sun: 8:05 --> 20C, 23:00 --> 16C
        // I wasn't able to find a proper converter or to create one, so i commented it out
        // [109, 'dp109', tuya.valueConverter.raw],

        // unmapped DPs, still need to figure out what they do
        // [103, 'dp103', tuya.valueConverter.trueFalse1],
        // [105, 'dp105', tuya.valueConverter.trueFalse1],
        // [110, 'backlight_mode', tuya.valueConverter.raw],
        // [111, 'dp111', tuya.valueConverter.trueFalse1]
    ],
},
};

joaofl, 2023-11-09:
I have the same device, and could help with the PR. But I'm actually unable to pair it. How do you enter pairing mode?

Liionboy, 2023-11-10:
> I have the same device, and could help with the PR. But I'm actually unable to pair it. How do you enter pairing mode?

You hold down arrow to enter in pair mode until wi fi sign blink

Liionboy, 2023-11-13:
> I have the same device, and could help with the PR. But I'm actually unable to pair it. How do you enter pairing mode?

did you succed to connect?

joaofl, 2023-11-14:
> > I have the same device, and could help with the PR. But I'm actually unable to pair it. How do you enter pairing mode?
> 
> did you succed to connect?

Yes, thank you. I was amazed that I could not find that info anywhere.

After connecting in have some controls working out of the box. But I noticed that I am loosing the date/time right after I set it in the device. That was the first issue I found. 

Now I have to dig a bit deeper in this thread and gather the missing pieces

Liionboy, 2023-11-16:
> Yes, thank you. I was amazed that I could not find that info anywhere.
>
> After connecting in have some controls working out of the box. But I noticed that I am loosing the date/time right after I set [...]

yes, same here always when restart z2m data will be lose, and as you can see can't use all function from this thermostat like on instructions, is detected like  ONNDO smart thermostat  not like AVATTO but for me i saw is work now only for heat and i can calibrate temperature from HASS not from z2m because there are not working

Liionboy, 2023-11-16:
[image]

joaofl, 2023-11-16:
But if it is detected as onddo, it probably has the same id? That could make things harder if they behave differently. Does the ONddo works fully? Do you know?

Liionboy, 2023-11-16:
> But if it is detected as onddo, it probably has the same id? That could make things harder if they behave differently. Does the ONddo works fully? Do you know?

i dont have that model, only this from AVATTO

Liionboy, 2023-11-16:
i see this is ONNDO ----> https://www.ebay.com/itm/334559054299

joaofl, 2023-11-16:
It looks exactly the same, right? Aren't they running the same firmware?

Liionboy, 2023-11-16:
> It looks exactly the same, right? Aren't they running the same firmware?

i dont know because i don't have that from ONNDO this from AVATTO is on battery

andrejsoucek, 2023-11-17:
@Liionboy I'm using you converter but the calibration still can't be set through the z2m. Also do you know what is the reason of the time and day reset? I can help with MR and code but don't know what exactly is needed

andrejsoucek, 2023-11-18:
@Koenkk are you able to help here? I tried the `hy_set_time_request` code from this post https://github.com/Koenkk/zigbee2mqtt/issues/10489#issuecomment-1312817478 and now I'm trying 
` onEvent: tuya.onEventSetTime,` but after z2m restart the device always has time set to Thursday 0:00.  

Can I somehow debug the device if it is asking for the date & time?

Koenkk, 2023-11-19:
Can you try with both ` onEvent: tuya.onEvent({timeStart: '2000'}),` and ` onEvent: tuya.onEvent({timeStart: '1970'}),`, also try to re-pair the device after that? (to force it to request a new time)

andrejsoucek, 2023-11-19:
> Can you try with both ` onEvent: tuya.onEvent({timeStart: '2000'}),` and ` onEvent: tuya.onEvent({timeStart: '1970'}),`, also try to re-pair the device after that? (to force it to request a new time)

Thanks! 1970 works and the time is now synced!
I have now noticed another issue - the debug log is flooded with
`Debug 2023-11-19 15:25:24Received Zigbee message from 'Termostat', type 'commandMcuVersionResponse', cluster 'manuSpecificTuya', data '{"seq":8475,"version":255}' from endpoint 1 with groupID 0
Debug 2023-11-19 15:25:24No converter available for 'ZWT198' with cluster 'manuSpecificTuya' and type 'commandMcuVersionResponse' and data '{"seq":8475,"version":255}'`

Koenkk, 2023-11-19:
The ignore is fine, could you make a pr to add support for this device?

andrejsoucek, 2023-11-19:
> The ignore is fine, could you make a pr to add support for this device?

Sure, I can, but there is a possibility of 2 devices with the same fingerprint.. 
The current [tuya.js](https://github.com/Koenkk/zigbee-herdsman-converters/blob/6ce404d203b0946b7eaefe7abfc211568c55774a/src/devices/tuya.ts#L4091) is probably mentioning [this device](https://www.ebay.com/itm/334559054299). 
We have [this device](https://www.aliexpress.com/item/1005005822896810.html) with the same fingerprint as defined in the tuya.js. So my PR would probably break the owners of ONNDO thermostat...

Liionboy, 2023-11-19:
With battery will be great if we discover how to configure

Liionboy, 2023-11-19:
> @Liionboy I'm using you converter but the calibration still can't be set through the z2m. It displays the values which is set on the device, tho. Also do you know what is the reason of the time and day reset? I can help with MR and code but don't know what exactly is needed

that was made only to change name from ONNDO to AVATTO nothing more, with time was the same problem for me but i didn't pay attention , i saw after every restart time was reseted, with calibration is work from HA but when HA is restarted same problem you will se on zero calibration value

andrejsoucek, 2023-11-19:
> that was made only to change name from ONNDO to AVATTO nothing more, with time was the same problem for me but i didn't pay attention , i saw after every restart time was reseted, with calibration [...]

The issue with the time was fixed in the MR above, you can just update your external converter based on my [comment](https://github.com/Koenkk/zigbee2mqtt/issues/19351#issuecomment-1817871180). Calibartion is ok after restart, even after repairing. Looks like the value saved in the termostat is correctly read by z2m.

Liionboy, 2023-11-20:
there are a lot to work with this thermostat , we need to find brightness and scheduled then battery percent

marcelhoogantink, 2023-11-23:
Hi all,

While also working on improvements on the Avatto ZWT198, I came across this threat.

Yesterday I did a PR for it, and it is now in progress.
Brightness, Scheduling are covered. Scheduling format is split into workday-schedule (6 items) and holiday schedule ( 2items), including a format check.
I did not find anything with battery-info, but there are two non-documented boolean datapoints (105 and 111) which can also be set, so these contains possibly no battery info.

Find below a screen-shot:

[image]

Liionboy, 2023-11-23:
i think this will be present in next version

marcelhoogantink, 2023-12-03:
> @marcelhoogantink I see your update published in the last version. Any idea why the exposed properties in the z2m are still the same as before (as with the custom converter mentioned here). I commented out the converter from the configuration.yaml, deleted the backup json file, restarted z2m and nothing. Still the old properties displayed, nothing new.

You could try to remove the device and pair it again, or do a factory reset on the device.
Did you clear the cash (f5)?

Imbanon, 2023-12-26:
FYI for Deadzone temperature, you can type in 0.1 as a minimum (unlike min being 0.5 on the thermostat setting) giving the thermostat a far better precision with holding the set temperature

maybe this should also be changed in code for the slider

[image]

Koenkk, 2023-12-26:
@andrejsoucek no caching is done, restarting z2m is enough. To confirm this, try changing the `description` and see if it has changed in the z2m frontend.

andrejsoucek, 2023-12-28:
> @andrejsoucek no caching is done, restarting z2m is enough. To confirm this, try changing the `description` and see if it has changed in the z2m frontend.

I tried, it still looked like it loaded the external converter (which was still there as the .js file but not in the configuration.yaml)

closed 2024-07-28 by github-actions[bot]

pilot173, 2024-11-12:
> > Can you try with both ` onEvent: tuya.onEvent({timeStart: '2000'}),` and ` onEvent: tuya.onEvent({timeStart: '1970'}),`, also try to re-pair the device after that? (to force it to request a new time)
> 
> Thanks! 1970 works and the time is now synced! I have now noticed another issue - the debug log is flooded with `Debug 2023-11-19 15:25:24Received Zigbee message from 'Termostat', type 'commandMcuVersionResponse', cluster 'manuSpecificTuya', data '{"seq":8475,"version":255}' from endpoint 1 with groupID 0 Debug 2023-11-19 15:25:24No converter available for 'ZWT198' with cluster 'manuSpecificTuya' and type 'commandMcuVersionResponse' and data '{"seq":8475,"version":255}'` And it seems I fixed it by adding `fz.ignore_tuya_set_time` to fromZigbee array. Not sure if it is ok or it conflicts with something?

Hi,

I have a problem with time sync in my Thermostat ZWT198. After pairing with Z2M looks good, but next day the time is random on the thermostat and I cant make synchro.

Can U send me step by step how to fix this? I'm newbie :)

Thanks a lot!

marcospm92, 2024-11-27:
Hello @marcelhoogantink , sorry to write in such an old topic, but I am facing some issues with this thermostat and would like to know if you are also having the same issues…
When setting local_temperature_calibration to a negative value, is it working fine for you? It starts fine for me, but after some time it gets converted to a big number, such as 6552 or similar, having a lot of errors in Home Assistant

pilot173, 2024-11-28:
> Hello @marcelhoogantink , sorry to write in such an old topic, but I am facing some issues with this thermostat and would like to know if you are also having the same issues… When setting [...]

Same issue

marcelhoogantink, 2024-11-28:
Sorry, i do not use this device anymore.

peins94, 2024-12-02:
> Hello @marcelhoogantink , sorry to write in such an old topic, but I am facing some issues with this thermostat and would like to know if you are also having the same issues… When setting [...]

You are probably best to raise a seperate issue for this

marcelhoogantink, 2024-12-02:
Again, i do not use this device anymore

peins94, 2024-12-02:
> Again, i do not use this device anymore

Was replying to someone else :)