Koenkk/zigbee2mqtt #19354: No devices found when pairing was permitted [PiZero/Pi3B+]

x86Hephestus, 2023-10-20:
### What happened?

I am trying to create a low-power z2mqtt coordinator using a Pi Zero W running 32-bit Pi OS Lite Bookworm. I mostly followed [the instruction provided](https://www.zigbee2mqtt.io/guide/installation/01_linux.html#installing). I had downloaded and installed Node 16.20.0 and NPM 8.19.4 from here (https://unofficial-builds.nodejs.org/download/release/v16.20.0/) as that is the exact version my production z2mqtt instance runs on. I confirmed installation and versions are correct. After downloading, building, and starting z2mqtt on the Pi Zero I do not see any errors in the logs. It starts, the frontend works, and it connects to my mqtt broker.

However, when I click Permit join and place a CentraLite 4257050-ZHAC into pairing mode no devices are discovered by z2mqtt. 

Thinking this might be a node/npm problem, I tried with major versions 18 and 20 with no success.

I removed modemmanager and disabled hciuart but issue is persisting.

Looking in dmesg I do not see any power or USB errors.

I then took the same TI CC2531 USB I have been using in my testing and attached it to a Pi 3B+. In testing I tried running both 32 and 64 bit Pi OS Lite Bookworm. I installed Node/NPM versions 18 and 20 from official sources. Behavior was the same as Pi Zero.

I then flashed HomeAssistantOS on the Pi 3B+ and, using the same dongle, was able to pair to the CentraLite via ZHA. I then set permit join on my production network and was able to pair the CentraLite as well.

Without anything obvious in the logs I am at a bit of a loss.

### What did you expect to happen?

Install zigbee2mqtt and once it is confirmed running be able to pair a known good zigbee device via permit pairing.

### How to reproduce it (minimal and precise)

RPi Zero W running latest 32-bit Pi OS Lite Bookworm
SSH into Pi Zero

wget https://unofficial-builds.nodejs.org/download/release/v16.20.0/node-v16.20.0-linux-armv6l.tar.xz
tar -xvf node-v16.20.0-linux-armv6l.tar.xz
cd node-v16.20.0-linux-armv6l/
sudo cp -R * /usr/local/

node --version && npm --version

sudo apt-get update && sudo apt-get install -y git

sudo mkdir /opt/zigbee2mqtt
sudo chown -R ${USER}: /opt/zigbee2mqtt
git clone --depth 1 https://github.com/Koenkk/zigbee2mqtt.git /opt/zigbee2mqtt
cd /opt/zigbee2mqtt
npm ci
npm run build

create configuration.yaml and populate with correct values

npm start

log in via front end and permit pairing.

### Zigbee2MQTT version

1.33.1 commit: 7e63039

### Adapter firmware version

20211115

### Adapter

TI CC2531 USB CDC

### Debug log

[Debug Logs.txt](https://github.com/Koenkk/zigbee2mqtt/files/13049310/Debug.Logs.txt)

x86Hephestus, 2023-10-20:
I took my production zigbee network offline and tried it's radio (a SONOFF Zigbee 3.0 Dongle-P ZB Dongle running Coordinator type: zStack3x0 / Coordinator Revision: 20220219) with the Pi Zero W and Node 16.2.0/NPM8.19.4 and everything worked perfectly right off the bat. Must be something with how z2mqtt works with TI CC2531. I know it's old but it is still officially supported right?

lux73, 2023-10-20:
CC2531 is crap and not worth to deal with it

closed 2024-05-18 by github-actions[bot]