Koenkk/zigbee2mqtt #19356: Ikea Shortcut Button E1812 send Null

MorganMLGman, 2023-10-20:
### What happened?

Ikea shortcut button E1812 is sending Null as action value instead of correct value.

[image]

### What did you expect to happen?

Correct action, press, hold, release, etc.

### How to reproduce it (minimal and precise)

1. Join device to the network
2. Button firmware 24.4.6
3. Press a button

### Zigbee2MQTT version

1.33.1

### Adapter firmware version

20221226

### Adapter

Sonoff Zigbee 3.0 CC2652P

danieledwardgeorgehitchcock, 2023-10-20:
This may be the correct behaviour for the device.. [See here](https://www.zigbee2mqtt.io/guide/faq/#why-is-the-action-property-always-empty)

MorganMLGman, 2023-10-20:
But HA also doesn't see state change, it is always "action: none" and if I remember correctly some time ago the behaviour was like shown there [IKEA E1812](https://www.zigbee2mqtt.io/devices/E1812.html#action-enum). But it was some time ago and me moving out, and now in new place I can't get it to work.

bazb, 2024-01-11:
I have the same issue. Broke my buttons.

closed 2024-08-10 by github-actions[bot]