Koenkk/zigbee2mqtt #19359: TI0001 Keeps losing connection

nielsnieland2, 2023-10-20:
### What happened?

After installing entire house with Livolo TI0001 and setting it up, everything worked for a while but now they are losing connections, not updating status one by one. Already 3 are not updating anymore. Tried reconfiguring, reinstalling etc 

### What did you expect to happen?

I just spend €€€€ on livolo because it looks nice.. I really need some help to get it to work

### Zigbee2MQTT version

1.33.1

### Adapter firmware version

20210708

### Adapter

Sonoff zigbee 3.0 dongle plus

### Debug log

error 2023-10-20 21:18:24: Publish 'set' 'state' to 'Noah Slaapkamer Licht' failed: 'Error: Command 0x00124b002174616e/6 genOnOff.toggle({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":0,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))'
debug 2023-10-20 21:18:24: Error: Command 0x00124b002174616e/6 genOnOff.toggle({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":0,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:415:23)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
info  2023-10-20 21:18:24: MQTT publish: topic 'zigbee2mqtt/Gang Lamp WC', payload '{"last_seen":"2023-10-20T19:18:24.977Z","linkquality":135,"state_bottom_left":null,"state_bottom_right":null,"state_left":null,"state_right":null}'
info  2023-10-20 21:18:25: MQTT publish: topic 'zigbee2mqtt/Gang Lamp WC', payload '{"last_seen":"2023-10-20T19:18:24.977Z","linkquality":135,"state_bottom_left":"OFF","state_bottom_right":null,"state_left":null,"state_right":null}'
debug 2023-10-20 21:18:25: Received Zigbee message from 'Gang Lamp WC', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,235,193,54,35,0,75,18,0,7,1,0,1,0,11],"type":"Buffer"}' from endpoint 6 with groupID 0
info  2023-10-20 21:18:25: MQTT publish: topic 'zigbee2mqtt/Gang Lamp WC', payload '{"elapsed":605,"last_seen":"2023-10-20T19:18:25.120Z","linkquality":135,"state_bottom_left":"OFF","state_bottom_right":"OFF","state_left":"OFF","state_right":"OFF"}'
info  2023-10-20 21:18:25: MQTT publish: topic 'zigbee2mqtt/Gang Lamp WC', payload '{"last_seen":"2023-10-20T19:18:25.250Z","linkquality":135,"state_bottom_left":null,"state_bottom_right":null,"state_left":null,"state_right":null}'
info  2023-10-20 21:18:25: MQTT publish: topic 'zigbee2mqtt/Gang Lamp WC', payload '{"last_seen":"2023-10-20T19:18:25.250Z","linkquality":135,"state_bottom_left":null,"state_bottom_right":"OFF","state_left":null,"state_right":null}'
debug 2023-10-20 21:18:25: Received Zigbee message from 'Gang Lamp WC', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,235,193,54,35,0,75,18,0,7,1,0,1,0,11],"type":"Buffer"}' from endpoint 6 with groupID 0
info  2023-10-20 21:18:25: MQTT publish: topic 'zigbee2mqtt/Gang Lamp WC', payload '{"elapsed":282,"last_seen":"2023-10-20T19:18:25.403Z","linkquality":135,"state_bottom_left":"OFF","state_bottom_right":"OFF","state_left":"OFF","state_right":"OFF"}'
info  2023-10-20 21:18:25: MQTT publish: topic 'zigbee2mqtt/Gang Lamp WC', payload '{"last_seen":"2023-10-20T19:18:25.403Z","linkquality":135,"state_bottom_left":null,"state_bottom_right":null,"state_left":null,"state_right":"OFF"}'
de

mirozv, 2024-01-19:
I have the same issue. I have to re-pair the switches afterward. But then the problem repeats.

activatorr, 2024-01-28:
Same, it works after reboot about 5 minutes, and then goes non working

activatorr, 2024-01-28:
@Koenkk please, some attention on this problem, it nothing one, from 1.35 many problems goes in z2m, last near ideally working is 1.33.1

botany112, 2024-03-09:
it works great here.  no problem. i only made a automation that enable and disable pairing at zigbee2mqtt every 4 hours.

closed 2024-10-06 by github-actions[bot]