Koenkk/zigbee2mqtt #19361: [New device support]:

wols89, 2023-10-20:
### Link

https://aliexpress.ru/item/1005005770731714.html

### Database entry

{"id":5,"type":"EndDevice","ieeeAddr":"0xa4c138e49df54f23","nwkAddr":59756,"manufId":4417,"manufName":"_TZE200_p3dbf6qs","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffecfba41","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","interviewCompleted":true,"meta":{"configured":821693351},"lastSeen":1697819079943,"defaultSendRequestWhen":"immediate"}

### Comments

Hello,
There is a thermostatic valve, presumably me167. I saw it on Github, they seemed to talk about its support. In fact, when connecting, z2m writes that the device is not supported.
Then I installed an external converter from the author @twhittock
twhittock - https://github.com/twhittock/avatto_me167.
The thermostat has been identified, even the photo is correct. However, none of the functions are active. The Z2M does not see the temperature from the sensor.
I also found the code for _TZE200_p3dbf6qs on Github, tried to replace “dataPoints: {”, in this case z2m does not start.
Tnx! 

[image]
[image]
[image]

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const tuyaLocal = {
  dataPoints: {
    me167Mode: 2,
    me167HeatingSetpoint: 4,
    me167LocalTemp: 5,
    me167ChildLock: 7,
    me167Heating: 3,
    me167Schedule1: 28,
    me167Schedule2: 29,
    me167Schedule3: 30,
    me167Schedule4: 31,
    me167Schedule5: 32,
    me167Schedule6: 33,
    me167Schedule7: 34,
    me167ErrorCode: 35,
    me167FrostGuard: 36,
    me167AntiScaling: 39,
    me167TempCalibration: 47,
  },
};

const fzLocal = {
  me167_thermostat: {
    cluster: 'manuSpecificTuya',
    type: ['commandDataResponse', 'commandDataReport'],
    convert: (model, msg, publish, options, meta) => {
        const result = {};

        function weeklySchedule(day, value) {
          // byte 0 - Day of Week (0~7 = Wed ~ Tue) ???
          // byte 1 - hour ???
          // byte 2 - minute ???
          // byte 3 - Temp (temp = value )
          // byte 4 - Temperature (temp = value / 10)

          const weekDays=[ 'wed', 'thu', 'fri', 'sat', 'sun','mon', 'tue'];
[... 237 lines cut ...]
        tzLocal.me167_thermostat_system_mode,
        tzLocal.me167_thermostat_schedule,
        tzLocal.me167_thermostat_calibration,
        tzLocal.me167_thermostat_anti_scaling,
        tzLocal.me167_thermostat_frost_guard,
    ],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [
      e.child_lock(),
      exposes.switch().withState('anti_scaling', true).withDescription('Anti Scaling feature is ON or OFF'),
      exposes.switch().withState('frost_guard', true).withDescription('Frost Protection feature is ON or OFF'),
      exposes.climate().withSetpoint('current_heating_setpoint', 5, 35, 1)
                     .withLocalTemperature()
                     .withSystemMode(['auto','heat','off'])
                     .withRunningState(['idle', 'heat'], ea.STATE)
                     .withLocalTemperatureCalibration(-10, 10, 1, ea.STATE_SET)
    ],
};

module.exports = definition;
```

Manni51, 2023-10-22:
I'm using:
ME167 (AVATTO) modelID: TS0601, manufacturerName:_TZE200_6rdj8dzm
ME168 (AVATTO) modelID: TS0601, manufacturerName:_TZE200_p3dbf6qs.
Both are supported.
Using "system mode" auto on ME168 causes strange values of "Current heating setpoint"

wols89, 2023-10-22:
> Я использую: ME167 (AVATTO) ID модели: TS0601, имя производителя: _TZE200_6rdj8dzm ME168 (AVATTO) ID модели: TS0601, имя производителя: _TZE200_p3dbf6qs. Оба поддерживаются. Использование автоматического «системного режима» на ME168 приводит к странным значениям «Текущей уставки нагрева».

I wonder why it doesn't work for me then?  The z2m versions are all the latest.  Now I installed version 1.32.2-1 zigbee2mqtt.  As you can see, it doesn’t work either out of the box or with the converter. 
 Do you have any external converter for this trv?

Manni51, 2023-10-22:
I don't use an external converter. You might get the converter of ME167 manufacturerName:_TZE200_6rdj8dzm, change manufacturerName to _TZE200_p3dbf6qs and try this as external converter.

TheBelgarion, 2023-10-26:
I also have the same model. I wanted to connect it to HomeAssistant via zigbee2MQTT.
I got identified nicely but on HA with MQTT Integration is see all the useless entities but the important climate once are all missing ( also checked for hidden entries)

[image]
[image]

anyway to fix what or check why the most important informations are not on HA?

TY

TheBelgarion, 2023-10-28:
I just saw that error in HA: 
2023-10-28 19:14:44.536 ERROR (MainThread) [homeassistant.util.logging] Exception in async_discover when dispatching 'mqtt_discovery_new_climate_mqtt': ({'action_template': "{% set values = {None:None,'idle':'off','heat':'heating','cool':'cooling','fan_only':'fan'} %}{{ values[value_json.running_state] }}", 'action_topic': 'zigbee2mqtt/TRV03', 'availability': [{'topic': 'zigbee2mqtt/bridge/state', 'value_template': '{{ value_json.state }}'}, {'topic': 'zigbee2mqtt/TRV03/availability', 'value_template': '{{ value_json.state }}'}], 'availability_mode': 'all', 'current_temperature_template': '{{ value_json.local_temperature }}', 'current_temperature_topic': 'zigbee2mqtt/TRV03', 'device': {'identifiers': ['zigbee2mqtt_0xa4c138853938d9cf'], 'manufacturer': 'AVATTO', 'model': 'Thermostatic radiator valve (ME168)', 'name': 'TRV03'}, 'json_attributes_topic': 'zigbee2mqtt/TRV03', 'max_temp': '35', 'min_temp': '5', 'mode_command_topic': 'zigbee2mqtt/TRV03/set/system_mode', 'mode_state_template': '{{ value_json.system_mode }}', 'mode_state_topic': 'zigbee2mqtt/TRV03', 'modes': ['auto', 'heat', 'off'], 'name': None, 'object_id': 'trv03', 'origin': {'name': 'Zigbee2MQTT', 'sw': '1.33.1', 'url': 'https://www.zigbee2mqtt.io'}, 'temp_step': 1, 'temperature_command_topic': 'zigbee2mqtt/TRV03/set/current_heating_setpoint', 'temperature_state_template': '{{ value_json.current_heating_setpoint }}', 'temperature_state_topic': 'zigbee2mqtt/TRV03', 'temperature_unit': 'C', 'unique_id': '0xa4c138853938d9cf_climate_zigbee2mqtt', 'platform': 'mqtt'},)

olili, 2023-10-30:
> I'm using: ME167 (AVATTO) modelID: TS0601, manufacturerName:_TZE200_6rdj8dzm ME168 (AVATTO) modelID: TS0601, manufacturerName:_TZE200_p3dbf6qs. Both are supported. Using "system mode" auto on ME168 causes strange values of "Current heating setpoint"

I am facing the same issue. Setpoint ist sporadically set to 25°C. Any hint?

TheBelgarion, 2023-10-31:
I updated to latest Version 2023.10 of HA and now the values are all there so climate seems to be rather new  stuff

closed 2024-05-29 by github-actions[bot]