Koenkk/zigbee2mqtt #19369: no OTA endpoint available

Holger237237, 2023-10-21:
### What happened?

Error "Failed to find endpoint which support OTA cluster" showed when I click on check updates for Xiaomi QBKG04LM using OTA.

### What did you expect to happen?

upgrade the firmware successfully

### How to reproduce it (minimal and precise)

Checkng for avaiable Updates in frontend-OTA
device :
Modell / Firmware-Date / Firmware-Version
Tuya-BSD29 / 20210625 / 500+TZSKT31BS102

### Zigbee2MQTT version

1.33.1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff Zigbee Stick 3.0 EZSP v8

### Debug log

2023-10-21 13:55:38MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"ZSteckdose01"},"error":"Failed to check if update available for 'ZSteckdose01' (Failed to find endpoint which support OTA cluster)","status":"error","transaction":"e1ynr-5"}'
Error 2023-10-21 13:55:38Failed to check if update available for 'ZSteckdose01' (Failed to find endpoint which support OTA cluster)
Info 2023-10-21 13:55:39MQTT publish: topic 'zigbee2mqtt/ZSteckdose01', payload '{"child_lock":"UNLOCK","current":0,"energy":0,"indicator_mode":"off/on","last_seen":"2023-10-18T19:36:51.126Z","linkquality":108,"power":0,"power_outage_memory":"restore","state":"ON","voltage":228}'

Koenkk, 2023-10-22:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-10-22 by Koenkk

Holger237237, 2023-10-24:
There aren't any changes by using dev-version of today. Same error

Koenkk, 2023-10-24:
what zigbee herdsman converters version shows under the frontend, settings, about page?

Holger237237, 2023-10-24:
15.101.0