Koenkk/zigbee2mqtt #19370: [Feature request]: Candeo C202/HK-DIM-A missing OTA support.

bfayers, 2023-10-21:
### Is your feature request related to a problem? Please describe

Recently bought a Candeo C202 (showing up as HK-DIM-A, though.) and noticed there is no OTA support in Z2M. 

They have published a firmware update on their website (https://candeo.io/support/support-c202/support-c202-home-assistant/) with instructions for applying it under ZHA, with a note that they would be releasing "A new firmware with zigbee2mqtt for Start Up behaviour will be released in September 2023." (which hasn't arrived yet) 

### Describe the solution you'd like

Add OTA Support for this device

### Describe alternatives you've considered

N/A	

### Additional context

N/A

Koenkk, 2023-10-22:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-10-22 by Koenkk

Chriswright16, 2023-11-01:
The Auto update on 2 of my C202s has bricked them, now all become unknown. any ideas? Was firmware 11 from 1

bfayers, 2023-11-02:
> The Auto update on 2 of my C202s has bricked them, both devices have disappeared from Z2M. any ideas? Was firmware 11 from 1

Out of interest do (did) your C202s show up as C202 or HK-DIM-A?

anders3408, 2023-11-02:
> The Auto update on 2 of my C202s has bricked them, both devices have disappeared from Z2M. any ideas? Was firmware 11 from 1

Same here. The one i had also just updated and is now dissappered in Z2M.

`Device 'Køkken' was updated from '{"dateCode":"20220523","softwareBuildID":"1.00"}' to 'null'
IMQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"from":{"date_code":"20220523","software_build_id":"1.00"},"id":"Køkken","to":null},"status":"ok","transaction":"xpr85-1"}'`

Chriswright16, 2023-11-02:
Shows up as HK-DIM-A

yes id agree its probably wise to disable for now.

Unfortunately im 3000miles from home so can't physically check my switches for the batch codes

hoombar, 2024-03-23:
I'm trying to manually update my Candeo dimmers with an OTA file that I grabbed off of their site. It ticks the boxes for what is supported (it has a 7 on the back of the light switch, and it has a supported older firmware to update from, so I don't think it'll brick my device).

When I try to manually do it [manually with a local OTA file](https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#local-ota-index-and-firmware-files), it just tells me that the dimmer switch doesn't support OTA updates.

Does the above reverted change block me from being able to do this?

thestraycat, 2024-10-06:
@bfayers - Did anything ever come of this? I'm in the same boat with supported candeo dimmers and no method to update them! (and i need a few features from the new update...)

bfayers, 2024-10-06:
> @bfayers - Did anything ever come of this? I'm in the same boat with supported candeo dimmers and no method to update them! (and i need a few features from the new update...)

I think the issue was that you _can_ do OTA on these, but they all identify the same but only some date codes (physically on the device) can actually OTA. 

If you're in the right date codes according to candeo [here](https://support.candeo.io/portal/en/kb/articles/c202-c202n-latest-zha-firmware#Compatible_devices) you could probably do it with a local override like mentioned above but you'd also need a custom converter with the ota cluster for the device.

You can look at the commit from further up in this issue that [added OTA originally](https://github.com/Koenkk/zigbee-herdsman-converters/blob/b211cc4c1be5f9dde201e511ff3f50716ed25bbc/src/devices/candeo.ts) if you want to try it