Koenkk/zigbee2mqtt #19372: all my SONOFF ZBMINI-L stopped to work

acaonweb, 2023-10-21:
### What happened?

Stopped to send commands, only get the state.
Initially I've tought one was broken, so I've bought another one, after installing the brand new, it's failed interview and pairing
So:
1) the old one stopped to works, reading only the state
2) the new one is recognized, added but fails the interview, and doesn't works

### What did you expect to happen?

it works as before

### How to reproduce it (minimal and precise)

install and pair

### Zigbee2MQTT version

1.33.0

### Adapter firmware version

20220219

### Adapter

Zzh! CC2652R Multiprotocol RF Stick

Tonio16, 2023-10-22:
Hello

First things would be to update your firmware to the 20221226 version and the last  version of zigbee2mqtt.

Antoine

acaonweb, 2023-10-22:
I've updated to 1.33.1 but I'm not sure to update the firmware because everything is working fine. 
It's strictly necessary?

Tonio16, 2023-10-22:
`because everything is working fine.
It's strictly necessary?`
Ok but why did you open this issue in that case?

Personnaly I would apply the firmware update of 20221226. Not the 20230520, which is facing some issues.

N3rdix, 2023-10-23:
I just updated from 1.33.0 to 1.33.1 and my ZBMINI-L still work fine (with a Sonoff adapter), so I assume it's not an issue with the Z2M update

acaonweb, 2023-10-29:
2023-10-29 09:41:10Publish 'set' 'state' to '0x00124b0024c07134' failed: 'Error: Command 0x00124b0024c07134/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 21546 - 1 - 152 - 6 - 11 after 10000ms)'
2023-10-29 09:41:22Failed to interview '0x00124b0024c07134', device has not successfully been paired

acaonweb, 2023-10-29:
this is the log I have with another zigbee https://www.zigbee2mqtt.io/devices/ZBMINI-L.html#sonoff-zbmini-l

acaonweb, 2023-10-31:
> Are you certain that no interference exists? I had a similar experience from one day to the other because my neighbor started building his Zigbee network on the same channel as me. Switching the channel (luckily in his environment with only few devices) solved the issue...

Well, I can't be sure about interference, but I've many devices and I've problem only with two, and with the same type (Sonoff without neutral). But how can I change the channel? I've a raspberry, and raspbian

Geoffrey54, 2023-12-10:
I’ve got the same issue here, i just update z2m in version 1.34.0-1 (was in 1.33.1-1) and i have got the state of the device but the on off doesn’t work anymore.

Geoffrey54, 2023-12-15:
> I’ve got the same issue here, i just update z2m in version 1.34.0-1 (was in 1.33.1-1) and i have got the state of the device but the on off doesn’t work anymore.

After doing nothing it rework again...

acaonweb, 2024-01-14:
the same for me, it's working again, I've only update to 1.34.0

closed 2024-01-14 by acaonweb