Koenkk/zigbee2mqtt #19377: Zigbee2MQTT crashes with RBSH-RTH0

Davc0m, 2023-10-22:
### What happened?

Zigbee2MQTT crashes when using [room thermostat II 230 V](https://www.bosch-smarthome.com/de/de/produkte/geraete/raumthermostat-fussbodenheizung/) (floor heating) RBSH-RTH0

`error 2023-10-22 12:05:35: Failed to configure 'Fussbodenheizung Wohnzimmer ', attempt 1 (Error: Read 0x001e5e090296a24e/1 hvacUserInterfaceCfg([16395,16442,16443], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4617,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:421:28)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:563:22)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/bosch.ts:813:13)
    at Configure.configure (/app/lib/extension/configure.ts:117:13)
    at Immediate.<anonymous> (/app/lib/extension/configure.ts:69:17))`

### Zigbee2MQTT version

1.33.1-1

### Adapter firmware version

20231005

### Adapter

ZBDongle-E - SONOFF Sonoff Zigbee 3.0 USB Dongle Plus

### Debug log

`debug 2023-10-22 12:05:29: Received Zigbee message from 'Fussbodenheizung Wohnzimmer ', type 'attributeReport', cluster 'hvacThermostat', data '{"occupiedHeatingSetpoint":2250}' from endpoint 1 with groupID 0
debug 2023-10-22 12:05:30: Received Zigbee message from 'Tv Steckdose', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":110}' from endpoint 1 with groupID 0
info  2023-10-22 12:05:30: MQTT publish: topic 'zigbee2mqtt/Tv Steckdose', payload '{"energy":2.7,"last_seen":"2023-10-22T12:05:30+02:00","linkquality":152,"power":110,"power_on_behavior":"previous","state":"ON","update":{"installed_version":671115824,"latest_version":671115824,"state":"idle"},"update_available":null}'
info  2023-10-22 12:05:30: MQTT publish: topic 'zigbee2mqtt/Fussbodenheizung Wohnzimmer ', payload '{"boost":"OFF","child_lock":"UNLOCK","display_brightness":7,"display_onTime":10,"display_ontime":10,"humidity":55.31,"last_seen":"2023-10-22T12:05:29+02:00","linkquality":136,"local_temperature":22.6,"local_temperature_calibration":0,"occupied_cooling_setpoint":5,"occupied_heating_setpoint":22.5,"pi_heating_demand":58,"running_state":"heat","setpoint_change_source":"externally","system_mode":"heat","window_open":"OFF"}'
debug 2023-10-22 12:05:31: Received Zigbee message from 'Fussbodenheizung Wohnzimmer ', type 'attributeReport', cluster 'hvacThermostat', data '{"localTemp":2250}' from endpoint 1 with groupID 0
info  2023-10-22 12:05:31: MQTT publish: topic 'zigbee2mqtt/Fussbodenheizung Wohnzimmer ', payload '{"boost":"OFF","child_lock":"UNLOCK","display_brightness":7,"display_onTime":10,"display_ontime":10,"humidity":55.31,"last_seen":"2023-10-22T12:05:31+02:00","linkquality":136,"local_temperature":22.5,"local_temperature_calibration":0,"occupied_cooling_setpoint":5,"occupied_heating_setpoint":22.5,"pi_heating_demand":58,"running_state":"heat","setpoint_change_source":"externally","system_mode":"heat","window_open":"OFF"}'
info  2023-10-22 12:05:32: MQTT publish: topic 'zigbee2mqtt/Fussbodenheizung Wohnzimmer ', payload '{"boost":"OFF","child_lock":"UNLOCK","display_brightness":7,"display_onTime":10,"display_ontime":10,"humidity":55.31,"last_seen":"2023-10-22T12:05:32+02:00","linkquality":132,"local_temperature":22.5,"local_temperature_calibration":0,"occupied_cooling_setpoint":5,"occupied_heating_setpoint":22.5,"pi_heating_demand":58,"running_state":"heat","setpoint_change_source":"externally","system_mode":"heat","window_open":"OFF"}'
debug 2023-10-22 12:05:32: Received Zigbee message from 'Tv Steckdose', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info  2023-10-22 12:05:32: MQTT publish: topic 'zigbee2mqtt/Tv Steckdose', payload '{"energy":2.7,"last_seen":"2023-10-22T12:05:32+02:00","linkquality":152,"power":110,"power_on_behavior":"previous","state":"ON","update":{"installed_version":671115824,"latest_version":671115824,"state":"idle"},"update_available":null}'
debug 2023-10-22 12:05:34: Received Zigbee message from 'Tv Steckdose', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":126}' from endpoint 1 with groupID 0
info  2023-10-22 12:05:34: MQTT publish: topic 'zigbee2mqtt/Tv Steckdose', payload '{"energy":2.7,"last_seen":"2023-10-22T12:05:34+02:00","linkquality":152,"power":126,"power_on_behavior":"previous","state":"ON","update":{"installed_version":671115824,"latest_version":671115824,"state":"idle"},"update_available":null}'
info  2023-10-22 12:05:35: MQTT publish: topic 'zigbee2mqtt/Fussbodenheizung Wohnzimmer ', payload '{"boost":"OFF","child_lock":"UNLOCK","display_brightness":7,"display_onTime":10,"display_ontime":10,"humidity":55.31,"last_seen":"2023-10-22T12:05:34+02:00","linkquality":136,"local_temperature":22.5,"local_temperature_calibration":0,"occupied_cooling_setpoint":5,"occupied_heating_setpoint":22.5,"pi_heating_demand":58,"running_state":"heat","setpoint_change_source":"externally","system_mode":"heat","window_open":"OFF"}'
debug 2023-10-22 12:05:35: Received Zigbee message from 'Fussbodenheizung Wohnzimmer ', type 'readResponse', cluster 'hvacThermostat', data '{"localTemperatureCalibration":0}' from endpoint 1 with groupID 0
info  2023-10-22 12:05:35: MQTT publish: topic 'zigbee2mqtt/Fussbodenheizung Wohnzimmer ', payload '{"boost":"OFF","child_lock":"UNLOCK","display_brightness":7,"display_onTime":10,"display_ontime":10,"humidity":55.31,"last_seen":"2023-10-22T12:05:35+02:00","linkquality":136,"local_temperature":22.5,"local_temperature_calibration":0,"occupied_cooling_setpoint":5,"occupied_heating_setpoint":22.5,"pi_heating_demand":58,"running_state":"heat","setpoint_change_source":"externally","system_mode":"heat","window_open":"OFF"}'
debug 2023-10-22 12:05:35: Received Zigbee message from 'Fussbodenheizung Wohnzimmer ', type 'readResponse', cluster 'hvacThermostat', data '{"16391":1,"16450":0,"16451":0}' from endpoint 1 with groupID 0
info  2023-10-22 12:05:35: MQTT publish: topic 'zigbee2mqtt/Fussbodenheizung Wohnzimmer ', payload '{"boost":"OFF","child_lock":"UNLOCK","display_brightness":7,"display_onTime":10,"display_ontime":10,"humidity":55.31,"last_seen":"2023-10-22T12:05:35+02:00","linkquality":136,"local_temperature":22.5,"local_temperature_calibration":0,"occupied_cooling_setpoint":5,"occupied_heating_setpoint":22.5,"pi_heating_demand":58,"running_state":"heat","setpoint_change_source":"externally","system_mode":"heat","window_open":"OFF"}'
debug 2023-10-22 12:05:35: Received Zigbee message from 'Fussbodenheizung Wohnzimmer ', type 'readResponse', cluster 'hvacUserInterfaceCfg', data '{"keypadLockout":0}' from endpoint 1 with groupID 0
info  2023-10-22 12:05:35: MQTT publish: topic 'zigbee2mqtt/Fussbodenheizung Wohnzimmer ', payload '{"boost":"OFF","child_lock":"UNLOCK","display_brightness":7,"display_onTime":10,"display_ontime":10,"humidity":55.31,"last_seen":"2023-10-22T12:05:35+02:00","linkquality":132,"local_temperature":22.5,"local_temperature_calibration":0,"occupied_cooling_setpoint":5,"occupied_heating_setpoint":22.5,"pi_heating_demand":58,"running_state":"heat","setpoint_change_source":"externally","system_mode":"heat","window_open":"OFF"}'
debug 2023-10-22 12:05:35: Received Zigbee message from 'Fussbodenheizung Wohnzimmer ', type 'readResponse', cluster 'hvacUserInterfaceCfg', data '{"16442":10,"16443":7}' from endpoint 1 with groupID 0
info  2023-10-22 12:05:35: MQTT publish: topic 'zigbee2mqtt/Fussbodenheizung Wohnzimmer ', payload '{"boost":"OFF","child_lock":"UNLOCK","display_brightness":7,"display_onTime":10,"display_ontime":10,"humidity":55.31,"last_seen":"2023-10-22T12:05:35+02:00","linkquality":132,"local_temperature":22.5,"local_temperature_calibration":0,"occupied_cooling_setpoint":5,"occupied_heating_setpoint":22.5,"pi_heating_demand":58,"running_state":"heat","setpoint_change_source":"externally","system_mode":"heat","window_open":"OFF"}'
error 2023-10-22 12:05:35: Failed to configure 'Fussbodenheizung Wohnzimmer ', attempt 1 (Error: Read 0x001e5e090296a24e/1 hvacUserInterfaceCfg([16395,16442,16443], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufact [... cut ...] ,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:421:28)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:563:22)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/bosch.ts:813:13)
    at Configure.configure (/app/lib/extension/configure.ts:117:13)
    at Immediate.<anonymous> (/app/lib/extension/configure.ts:69:17))`

eberlemartin, 2023-11-25:
Hi There, I'm also having problems with the BTH-RM230Z. My z2m doesn't crash but i dont like how this device is implemented because of serveral reasons:

- Its name is wrong, the 230V mains powered thermostat that goes on your wall (not on a radiator) has BTH-RM230Z printed on the back. Its difficult not to confuse this with the battery operated device that looks exactly the same.
- It throws many errors when pairing and you have to "initialize" some values manually to get them working.
- State is always "heating".

Im fixing all this and will create a PR. Hopefully you will be able to pair it when its fixed.

Davc0m, 2023-12-29:
After the update, the system now works properly. Thank you