Koenkk/zigbee2mqtt #19382: [New device support]: TRADFRI bulb E27 WW G95 CL 470lm

JHDE, 2023-10-22:
### Link

https://www.ikea.com/nl/nl/p/tradfri-led-lamp-e27-470-lumen-smart-draadloos-dimbaar-warm-wit-transparant-bol-90539072/

### Database entry

{"id":65,"type":"Router","ieeeAddr":"0x385cfbfffe8f7ed4","nwkAddr":12798,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb E27 WW G95 CL 470lm","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,4096,64599,64636],"outClusterList":[25],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":17,"stackVersion":106,"hwVersion":1,"dateCode":"20220729","swBuildId":"1.1.006","zclVersion":8,"interviewCompleted":true,"meta":{},"lastSeen":1697984982048,"defaultSendRequestWhen":"immediate"}

### Comments

Technical description from IKEA model is LED2102G3.
This model is supported by Zigbee2Mqtt, but the bulb is not recognized as LED2102G3 but as TRADFRI bulb E27 WW G95 CL 470lm.
So Zigbee2mqtt pairs this device as unsupported.

Koenkk, 2023-10-22:
This device should be detected correctly in the next z2m version on 1 November.

closed 2023-10-22 by Koenkk

DennisvDijk, 2023-12-21:
This still occurs..

waterkrow, 2024-01-07:
it still unsupported.

Koenkk, 2024-01-07:
Should be supported in 1.35.1, if not, provide your data/database.db entry

freedomanonyme, 2024-03-22:
[image]

hell1337hound, 2024-03-26:
Do you have to remove it and re-add it? Or re-interview it for it to become supported?

Koenkk, 2024-03-26:
@freedomanonyme what shows up in the z2m -> settings -> about page, can you copy the model number? The one that starts with `TRADFRI bulb`

freedomanonyme, 2024-03-26:
Hi ! The model number is : TRADFRI bulb E26 WW G95 CL 450lm. I tried into Z2Mqtt Edge (dev build) and it works.

Koenkk, 2024-03-27:
Then it will be supported in the 1 April relesae.

freedomanonyme, 2024-03-27:
> Then it will be supported in the 1 April release.

Wow! Beautiful! I will give a try and I will keep you in touch! Thanks!

freedomanonyme, 2024-04-03:
Hi! I must apologize! It was an ID-01-T code. Zigbee2mqtt works fine for all my entities as of 2021, so I don't spend a lot of time maintaining it even when I new stuff, until the last Ikea bulb light. If it works, don't touch it my mother said. I realize today that I was on Daniel Welch's repository with version 1.18. I added your repository today and the bulb works great. I feel cheap about it. Thank you so much!