Koenkk/zigbee2mqtt #19383: [New device support]:  Add Support for another TS0601 Thermostat

Endorian88, 2023-10-22:
### Link

https://de.aliexpress.com/item/1005005997839493.html

### Database entry

{"id":40,"type":"EndDevice","ieeeAddr":"0xa4c138e3a0ebf720","nwkAddr":45823,"manufId":4417,"manufName":"_TZE204_g2ki0ejr","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65534":0,"modelId":"TS0601","manufacturerName":"_TZE204_g2ki0ejr","powerSource":3,"zclVersion":3,"appVersion":73,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00212effff096503","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":73,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1440156621},"lastSeen":1697995490290,"defaultSendRequestWhen":"immediate"}

### Comments

Zigbee2MQTT
Current version: 1.33.1-1 via Home Assistant Addon

Datapoints via Tuya IoT

`{"2":"Mode",
"4":"Set temperature",
"5":"Current temperature",
"6":"Battery capacity",
"7":"Child lock",
"9":"Set temperature ceiling",
"10":"The lower limit of temperature",
"14":"Window check",
"16":"Window temp",
"17":"Window time",
"18":"Backlight brightness",
"19":"Factory data reset",
"21":"Holiday temperature",
"24":"Home temp",
"25":"Leave temp",
"28":"Week program",
"29":"Week program Tuesday",
"30":"Week program Wednesday",
"31":"Week program Thursday",
"32":"Week program Friday",
"33":"Week program Saturday",
"34":"Week program Sunday",
"35":"Fault alarm",
"36":"Frost protection",
"37":"Rapid warming",
"38":"Rapid heating countdown",
"39":"Switch Scale",
"47":"Temperature correction",
"48":"Valve testing",
"49":"State of the valve",
"101":"111"}`

oxcid3, 2023-10-23:
Thanks for posting this. - I have a handful of these that are currently not showing up correctly and don't seem to match any of the custom quirks of other similar devices.

enrikkko, 2023-10-28:
Огромное спасибо ) я именно эти купил термоголовки . пока не понял как использовать вашу инструкцию что бы добавить устройство к себе в умный дом . пока не совсем понимаю, как вы это сделали ? подскажите пожалуйста

frederick9372, 2023-11-02:
Same problem with the same device, even with update 1.33.2

Randomizen, 2023-11-02:
I have this device too. It does not work with z2m. Please add support for this thermostat

oxcid3, 2023-11-10:
I think someone has posted a converter that works with Zigbee2MQTT here: https://github.com/zigpy/zha-device-handlers/issues/2677

but if anyone can help with a ZHA quirk based on this that would be really appreciated.

Steefph, 2023-11-10:
Yes, correct, that is my converter. Thought I was here and not there

Koenkk, 2023-11-10:
@Steefph could you make a pr to add support for this device?

Steefph, 2023-11-10:
@Koenkk, will try to do so on Sunday. It will be a first for me so hopefully no issues :)

wormvortex, 2023-11-10:
Will this work with ZHA currently or not?

Oxydation, 2023-11-19:
@Steefph were you able to set values? I added your converter and found the devices without further hassle. However it can read values but if I set e.g. the current setpoint it does not change the value on the device itself. Tested on three devices. Only manually changing it on the device does work. Also mode changes and all other settings are not working. The device is set to manual mode.

Steefph, 2023-11-19:
@Oxydation I've 2 devices and it "works". I need to give the command several times with the external converter. Haven't had time this week to look further into it.

Leopoll, 2023-11-23:
> I need to give the command several times with the external converter

Me too.
I'm waiting for a fix. Thank you very much for the work done.
But the most important thing is that the heater can operate from an external sensor. Can this be added?

Leopoll, 2023-11-24:
Error: Invalid value for number.termostat_batarei_holiday_temperature: 150 (range 5.0 - 30.0)
Invalid datapoint?
And, when you change Local temperature calibration, local temperature must change too. But it not changed.

alnospam64, 2023-11-26:
I got this working properly by updating to the latest development version.
Question to the other users: does this TRV send a status update every 20 seconds or so?

Oxydation, 2023-11-26:
> I got this working properly by updating to the latest development version.
> Question to the other users: does this TRV send a status update every 20 seconds or so?

Yes, and it's a couple of messages at once

Leopoll, 2024-04-15:
Please, who can help fix the error in the integration of this device? Or should I now write to another section of the forum?
As I already wrote, changing the local temperature calibration with Z2M does not work correctly.

> Info 2024-04-16 00:53:07MQTT publish: topic 'zigbee2mqtt/Термостат батареи старый', payload '{"backlight":0,"battery":100,"boost_heating":"OFF","boost_timeset_countdown":null,"child_lock":"UNLOCK","comfort_temperature":26,"current_heating_setpoint":20,"eco_temperature":20,"error_status":0,"factory_reset":"SET","frost_protection":"OFF","holiday_temperature":20,"linkquality":127,**"local_temperature":27.5,"local_temperature_calibration":-1,**"max_temperature":null,"max_temperature_limit":35,"min_temperature":null,"min_temperature_limit":5,"open_window_temperature":20,"preset":"manual","running_state":"idle","schedule":null,"schedule_friday":"06:00/20.0 11:30/21.0 13:30/22.0 17:30/23.0","schedule_monday":"06:00/20.0 11:30/21.0 13:30/22.0 17:30/23.0","schedule_saturday":"06:00/24.0 12:00/23.0 14:30/22.0 17:30/21.0","schedule_sunday":"06:00/19.0 12:30/20.0 14:30/21.0 18:30/20.0","schedule_thursday":"06:00/20.0 11:30/21.0 13:30/22.0 17:30/23.0","schedule_tuesday":"06:00/20.0 11:30/21.0 13:30/22.0 17:30/23.0","schedule_wednesday":"06:00/20.0 11:30/21.0 13:30/22.0 17:30/23.0","system_mode":"heat","valve":null,"window_detection":"OFF"}'
> Info 2024-04-16 00:53:38MQTT publish: topic 'zigbee2mqtt/Термостат батареи старый', payload '{"backlight":0,"battery":100,"boost_heating":"OFF","boost_timeset_countdown":null,"child_lock":"UNLOCK","comfort_temperature":26,"current_heating_setpoint":20,"eco_temperature":20,"error_status":0,"factory_reset":"SET","frost_protection":"OFF","holiday_temperature":20,"linkquality":127,**"local_temperature":35,"local_temperature_calibration":1,**"max_temperature":null,"max_temperature_limit":35,"min_temperature":null,"min_temperature_limit":5,"open_window_temperature":20,"preset":"manual","running_state":"idle","schedule":null,"schedule_friday":"06:00/20.0 11:30/21.0 13:30/22.0 17:30/23.0","schedule_monday":"06:00/20.0 11:30/21.0 13:30/22.0 17:30/23.0","schedule_saturday":"06:00/24.0 12:00/23.0 14:30/22.0 17:30/21.0","schedule_sunday":"06:00/19.0 12:30/20.0 14:30/21.0 18:30/20.0","schedule_thursday":"06:00/20.0 11:30/21.0 13:30/22.0 17:30/23.0","schedule_tuesday":"06:00/20.0 11:30/21.0 13:30/22.0 17:30/23.0","schedule_wednesday":"06:00/20.0 11:30/21.0 13:30/22.0 17:30/23.0","system_mode":"heat","valve":null,"window_detection":"OFF"}'

closed 2024-11-13 by github-actions[bot]