Koenkk/zigbee2mqtt #19388: [Feature request]: Ability to set the room to be ingested into Home-Assistant directly in Z2M

samip5, 2023-10-23:
### Is your feature request related to a problem? Please describe

I'm annoyed that I cannot currently set the area/room the device is in, so that it would go to it's right place in Home-Assistant directly

### Describe the solution you'd like

I would like a way to tell Home-Assistant in MQTT discovery in which room/area it should be placed into.

### Describe alternatives you've considered

N/A

### Additional context

I'm not sure if Home-Assistant supports this yet, but it would be useful.

MiyHun, 2023-11-15:
Hi,

It will be a great idea for me too.

Thanks

samip5, 2024-05-14:
Not STAKES

acommuni, 2024-11-24:
It would be great to have this feature, it's avalaible in ZHA

samip5, 2025-01-26:
Not really stale maybe?

samip5, 2025-05-28:
Nope?