Koenkk/zigbee2mqtt #19389: Vimar 02973.B NOT Supported

lsimonini982, 2023-10-23:
### What happened?

When I try to pair a Vimar 02973.B the system tell me that is not supported but Vimar 02973.B is listed as supported device

### What did you expect to happen?

I'm expectinng to pair Vimar 02973.B without any problem

### How to reproduce it (minimal and precise)

Try to pair Vimar 02973.B to Zigbee2MQTT network 

### Zigbee2MQTT version

 1.33.1

### Adapter firmware version

20221226

### Adapter

CC2652P 

### Debug log

[45df7312_zigbee2mqtt_2023-10-23T07-15-11.145Z.log](https://github.com/Koenkk/zigbee2mqtt/files/13067198/45df7312_zigbee2mqtt_2023-10-23T07-15-11.145Z.log)

Koenkk, 2023-10-23:
Can you provide the data/database.db entry of your device?

lsimonini982, 2023-10-23:
Hi @Koenkk here you go. Thanks
[database.zip](https://github.com/Koenkk/zigbee2mqtt/files/13078641/database.zip)

Koenkk, 2023-10-24:
What is the ieeeAddress of the 02973.B?

lsimonini982, 2023-10-24:
Here you go:
0xf4ce36c927174d9d

Luca

Koenkk, 2023-10-25:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/8b55d344a8652ef408ebb5342e5e47a4

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

lsimonini982, 2023-10-26:
Hi @Koenkk it works. Thanks for your support.

Koenkk, 2023-10-27:
Integrated the fix, you can remove the ext converter after updating to the next release.

closed 2023-10-27 by Koenkk

diegoflyer, 2024-02-08:
@lsimonini982 can you help me? This thermostat is bluetooth and zigbee technology, did you switch to zigbee to have it running in zigbee2mqtt?
On the other hand, if you have switched to zigbee are you able to use the PID functionality?
Or is there a way to connect to zigbee2mqtt in bluetooth thanks to some coordinatore that can manage also bluetooth device?