Koenkk/zigbee2mqtt #19392: Moes ZTS-EU_2gang not reporting manual activation to HA

tomesik, 2023-10-23:
### What happened?

After upgrading to Zigbee2MQTT to 1.33.1-1 HomeAssistant seems not to getting info from manual activation/deactivation of switch via button on it (device is working as usual but HA is not notified). Logs zigbee2mqtt are empty also. But I can activate and deactivate switch via zigbee.

### What did you expect to happen?

migration to 1.33.1-1

### How to reproduce it (minimal and precise)

just push button on switch

### Zigbee2MQTT version

1.33.1-1

### Adapter firmware version

20210128

### Adapter

Sonoff Zbdongle USB Zigbee 3.0 CC2652P

Palthe, 2023-10-23:
I have the same problem with a number of my devices, restarting HAS has no effect. Once the power of the device that does not respond is on and off again, or after a period of time, the responce in HAS works again.

The error log:

Zigbee2MQTT:error 2023-10-23 22:36:56: Error: CommandResponse 0x001788010cd9d740/2 genOta.queryNextImageResponse({"status":152}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)
Zigbee2MQTT:error 2023-10-23 22:36:57: Publish 'set' 'state' to 'back garden - at the house - light switch' failed: 'Error: Command 0xa4c1383b78913732/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)'
Zigbee2MQTT:error 2023-10-23 22:36:57: Publish 'set' 'state' to 'back garden - at the house - light switch' failed: 'Error: Command 0xa4c1383b78913732/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)'
Zigbee2MQTT:error 2023-10-23 22:37:00: Publish 'set' 'state' to 'back garden - at the house - light switch' failed: 'Error: Command 0xa4c1383b78913732/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 14009 - 1 - 21 - 6 - 11 after 10000ms)'

tomesik, 2023-11-19:
Full reinstall of z2mqtt solved the problem. However now all works, but there is problem in automations-traces devices are unknown entities in HA.

closed 2024-06-17 by github-actions[bot]