Koenkk/zigbee2mqtt #19402: [Wrong device]: TuYa TS0207_repeater

IdefixRC, 2023-10-24:
### Link

https://expo.tuya.com/product/753050

### Model

MYQ-ZRP01

### Description

Repeater

### Vendor

Tuya

### Picture (link)

https://www.zigbee2mqtt.io/images/devices/TS0207_repeater.jpg

### Database entry

n/a

### Notes

The repeater is discovered as Water leak detector.
Looking at the Zigbee2MQTT DB I can see that product code 0207 shows up as both repeater (https://www.zigbee2mqtt.io/devices/TS0207_repeater.html) and as various Water leak detectors (https://www.zigbee2mqtt.io/supported-devices/#s=0207). Is there a way to force the correct device type given the code is the same?

Thanks

Koenkk, 2023-10-24:
please provide the data/database.db entry of your device

grobihh, 2023-10-25:
@IdefixRC Did you check this issue already?
https://github.com/Koenkk/zigbee2mqtt/issues/19196

IdefixRC, 2023-10-25:
OK. Need to find out where to get this

IdefixRC, 2023-10-25:
Hi, I hope this helps:

`{"id":4,"type":"Router","ieeeAddr":"0x943469fffe546e2f","nwkAddr":45917,"manufId":4098,"manufName":"_TZ3000_nlsszmzl","powerSource":"Mains (single phase)","modelId":"TS0207","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":8,"inClusterList":[0,10,3],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005","65506":22,"stackVersion":0,"dateCode":"","appVersion":64}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1698237215263,"defaultSendRequestWhen":"immediate"}`

IdefixRC, 2023-10-25:
shows up in HA as: 

Description Water leak detector
Support status Supported
IEEE Address 0x943469fffe546e2f
Network address 0xB35D
Manufacturer [TuYa](https://www.zigbee2mqtt.io/supported-devices/#v=TuYa)
Model [TS0207_water_leak_detector](https://www.zigbee2mqtt.io/devices/TS0207_water_leak_detector.html#tuya-ts0207_water_leak_detector)

Koenkk, 2023-10-25:
Will be detected correctly in the 1 November release!

closed 2023-10-25 by Koenkk