Koenkk/zigbee2mqtt #19403: Aubess WHD02 disconnected after a day, leaves immediately after joining

andreagdipaolo, 2023-10-24:
### What happened?

When I paired the device the first time it worked for at least 12 hours.
The next day I couldn't find it anymore in the Z2M web ui and I tried to pair it again.
I tried with the dev version too and I updated my coordinator to the latest firmware with no luck.
When I try to pair the device again, in the logs in debug mode I can just see it immediately **leaving**, never joining, nothing else.
I believe it's not a limit on how many devices I can have because I successfully added a couple of new sensors for testing.

### What did you expect to happen?

The device should pair and stay connected. It's a router and it's powered from mains.

### How to reproduce it (minimal and precise)

Allow joining, press the pairing button for 10 seconds, see logs.

### Zigbee2MQTT version

1.33.1 commit: 7e63039

### Adapter firmware version

7.3.1.0 build 176

### Adapter

Sonoff Dongle-E

### Debug log

```
 2023-10-24 03:27:14 Device 'Interruttore 1' left the network
 2023-10-24 03:27:14 MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Interruttore 1","ieee_address":"0xa4c13886aba19efb"},"type":"device_leave"}'
 2023-10-24 03:27:14 Device 'Interruttore 1' left the network
 2023-10-24 03:27:14 MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Interruttore 1","ieee_address":"0xa4c13886aba19efb"},"type":"device_leave"}'
```

andreagdipaolo, 2023-10-29:
Seems to be solved after a power cycle.

andreagdipaolo, 2023-10-29:
Seems to be solved after a device power cycle.

andreagdipaolo, 2023-11-01:
Disconnected again and a new power cycle won't fix it. Will investigate tomorrow. 
Meanwhile I switched again to dev branch.

andreagdipaolo, 2023-11-01:
Seems solved after switching to another channel.

closed 2023-11-01 by andreagdipaolo

Nickolay40, 2023-11-07:
Hello, which channel you use .. i have same issue with this device ?

andreagdipaolo, 2023-11-07:
I switched from 11 to 15 even though they are both free here.
I had some other problems but apparently they were fixed after stopping the container, removing the coordinator, leaving it for a minute, plugging it again and starting the container.
Adding another router near it maybe helped too.

Nickolay40, 2023-11-07:
Ok. I try to fixed it without changing channel, because if I change channel need to fix all network with all devices. just only one of 3 WHD02 has this problem.

Nickolay40, 2023-11-16:
Hello, I solved my problem without change channel ... Problem was with installetion this switch in socket box. Buttom on switch was always pushed. It caused re-pair this device every time in Z2M.  I just removed plastic bottom.