Koenkk/zigbee2mqtt #19404: GZCGQ01LM - request for change expose values

jomuc, 2023-10-24:
### Is your feature request related to a problem? Please describe

device doesn't show up in dashboard, 
because of wrong configuration of exposed values: e.illuminance() and e.illuminance_lux()

### Describe the solution you'd like

please change exposed values of device to:

`{
        exposes: [e.battery(), e.battery_voltage(), e.illuminance(), e.power_outage_count()]
  }`

### Describe alternatives you've considered

no alternatives considered

### Additional context

[image]

Koenkk, 2023-10-24:
Can you provide a screenshot of the z2m frontend -> device -> state tab?

jomuc, 2023-10-25:
Screenshot of state tab:
[image]

jomuc, 2023-10-25:
`{
    "battery": 100,
    "illuminance": 0,
    "illuminance_lux": 0,
    "last_seen": "2023-10-25T03:22:46.289Z",
    "linkquality": 163,
    "power_outage_count": 12,
    "voltage": 3068
}`

Koenkk, 2023-10-25:
> device doesn't show up in dashboard,

I'm a bit confused, in the screenshot from the OP it shows the device in the dashboard.

jomuc, 2023-10-25:
Oh sorry. I did patch the node module "zigbee-herdsman-converters" with the exposes from the my first post.
The master code or code from version 1.33.1 doesn't show the device in dashboard.
But I did not manage to get illuminance_lux shown in dashboard. It just shows illuminance, but not the lux value.
The device state itself has both values.

Koenkk, 2023-10-27:
Both `illuminance` and `illumiance_lux` should be exposed, that it shows in the dashboard once `illumiance_lux` is removed is a bug in the frontend. So please create an issue here: https://github.com/nurikk/zigbee2mqtt-frontend/issues

closed 2024-05-25 by github-actions[bot]