Koenkk/zigbee2mqtt #19405: [New device support]: ZT-LP-ZEU2S-WH-MS MOES

MiAutomations, 2023-10-24:
### Link

https://moeshouse.com/products/zigbee-wireless-self-powered-scene-switch

### Database entry

{"id":76,"type":"GreenPower","ieeeAddr":"0x00000000a3970ea3","nwkAddr":3747,"manufId":null,"modelId":"GreenPower_2","epList":[242],"endpoints":{"242":{"epId":242,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":1698149193664,"defaultSendRequestWhen":"immediate"}

### Comments

Hello 

I try to add this device but the device appears in the list as not supported yet.

Can you please add this device to a supported list devices in a new future release ? 

Thank you for your support and best regards

Chartreusito, 2023-10-27:
If you own it, you can help supporting new devices : https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html

MiAutomations, 2023-11-05:
> If you own it, you can help supporting new devices : https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html

I'm  not familiarized with the  external converter But if you can help me I can try.

Chartreusito, 2023-11-06:
I assumed you device was tuya, but first have a look at the general documentation : https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-adding-your-device
The key here is to trigger your device to identify which cluster it uses and then try to find a working configuration among the other [herdmans converter](https://github.com/Koenkk/zigbee-herdsman)

mathieuruellan, 2024-01-06:
Hello, I've tried to add default converters, but there is no converter for "commandNotification".
I'm beginning in home automation and zigbee, if you have a js converter link, i could try.

here my logs

> 
> Zigbee2MQTT:debug 2024-01-06 16:39:39: Received Zigbee message from 'switch1', type 'commandNotification', cluster 'greenPower', data '{"commandID":79,"frameCounter":163,"srcID":793774560}' from endpoint 242 with groupID null
> Zigbee2MQTT:debug 2024-01-06 16:39:39: No converter available for 'GreenPower_2' with cluster 'greenPower' and type 'commandNotification' and data '{"commandID":79,"frameCounter":163,"srcID":793774560}'
> Zigbee2MQTT:info  2024-01-06 16:39:39: MQTT publish: topic 'zigbee2mqtt/switch1', payload '{"device":{"friendlyName":"switch1","ieeeAddr":"0x000000002f5009e0","manufacturerID":null,"model":"GreenPower_2","networkAddress":2528,"type":"GreenPower"},"last_seen":"2024-01-06T16:39:39.164Z","linkquality":209}

igor-semenov, 2024-02-02:
```
фев 02 17:32:19 raspberrypi bash[19752]: Zigbee2MQTT:info  2024-02-02 17:32:19: Device '0x00000000ed5e4506' joined
фев 02 17:32:20 raspberrypi bash[19752]: Zigbee2MQTT:info  2024-02-02 17:32:20: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00000000ed5e4506","ieee_address":"0x00000000ed5e4506"},"type":"device_joined"}'
фев 02 17:32:20 raspberrypi bash[19752]: Zigbee2MQTT:info  2024-02-02 17:32:20: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x00000000ed5e4506"},"type":"device_connected"}'
фев 02 17:32:20 raspberrypi bash[19752]: Zigbee2MQTT:info  2024-02-02 17:32:20: Successfully interviewed '0x00000000ed5e4506', device has successfully been paired
фев 02 17:32:20 raspberrypi bash[19752]: Zigbee2MQTT:warn  2024-02-02 17:32:20: Device '0x00000000ed5e4506' with Zigbee model 'GreenPower_2' and manufacturer name 'undefined' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
фев 02 17:32:20 raspberrypi bash[19752]: Zigbee2MQTT:info  2024-02-02 17:32:20: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x00000000ed5e4506","ieee_address":"0x00000000ed5e4506","status":"successful","supported":null},"type":"device_interview"}'
фев 02 17:32:20 raspberrypi bash[19752]: Zigbee2MQTT:info  2024-02-02 17:32:20: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0x00000000ed5e4506","supported":false},"type":"pairing"}'
```

igor-semenov, 2024-02-04:
I've added new js file for device, but every time I press the same button, it produces different commands:
```
фев 04 23:18:53 raspberrypi bash[3066]: Zigbee2MQTT:debug 2024-02-04 23:18:53: Received Zigbee message from '0x00000000af79b9e0', type 'commandCommissioningNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[61,61,216,74,14,215,98],"type":"Buffer"}},"commandID":41,"frameCounter":1324,"gppGddLink":216,"gppNwkAddr":15677,"options":2864,"payloadSize":0,"srcID":2943990240}' from endpoint 242 with groupID 0
фев 04 23:19:01 raspberrypi bash[3066]: Zigbee2MQTT:debug 2024-02-04 23:19:01: Received Zigbee message from '0x00000000af79b9e0', type 'commandCommissioningNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[223,26,214,4,164,177,194],"type":"Buffer"}},"commandID":3,"frameCounter":1333,"gppGddLink":214,"gppNwkAddr":6879,"options":2864,"payloadSize":0,"srcID":2943990240}' from endpoint 242 with groupID 0
фев 04 23:19:04 raspberrypi bash[3066]: Zigbee2MQTT:debug 2024-02-04 23:19:04: Received Zigbee message from '0x00000000af79b9e0', type 'commandCommissioningNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[223,26,214,204,153,107,228],"type":"Buffer"}},"commandID":201,"frameCounter":1342,"gppGddLink":214,"gppNwkAddr":6879,"options":2864,"payloadSize":0,"srcID":2943990240}' from endpoint 242 with groupID 0
```

ngeissel, 2024-02-08:
I'm interrested too (since a few hours) into finding a way to handle MOE green power switch. 

Same here for commands, but since it is as switch does it really matters ?  : 
```
Debug 2024-02-08 11:44:22Received Zigbee message from '0x00000000e91c2afd', type 'commandCommissioningNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[35,64,155,35,28,185,7],"type":"Buffer"}},"commandID":155,"frameCounter":1387,"gppGddLink":155,"gppNwkAddr":16419,"options":2608,"payloadSize":0,"srcID":3910937341}' from endpoint 242 with groupID 0
Debug 2024-02-08 11:44:23Received Zigbee message from '0x00000000e91c2afd', type 'commandCommissioningNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[35,64,155,74,187,160,95],"type":"Buffer"}},"commandID":252,"frameCounter":1396,"gppGddLink":155,"gppNwkAddr":16419,"options":2608,"payloadSize":0,"srcID":3910937341}' from endpoint 242 with groupID 0
Debug 2024-02-08 11:44:27Received Zigbee message from '0x00000000e91c2afd', type 'commandCommissioningNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[35,64,153,15,212,74,125],"type":"Buffer"}},"commandID":70,"frameCounter":1405,"gppGddLink":153,"gppNwkAddr":16419,"options":2608,"payloadSize":0,"srcID":3910937341}' from endpoint 242 with groupID 0
Debug 2024-02-08 11:44:27Received Zigbee message from '0x00000000e91c2afd', type 'commandCommissioningNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[35,64,149,233,198,137,178],"type":"Buffer"}},"commandID":81,"frameCounter":1414,"gppGddLink":149,"gppNwkAddr":16419,"options":2608,"payloadSize":0,"srcID":3910937341}' from endpoint 242 with groupID 0
```

NilkOne, 2024-02-18:
Hello, any news if it possible to add it on zigbee2mqtt ?

I bought et 2-scenes et 3-scenes switch but Not supported yet

psummo, 2024-04-09:
> The fundamental thing I don't understand is whether I need any specific hardware (new USB stick maybe) to connect ZGP device? My USB stick allows me to pair device, I can see commands from it when I press buttons, but commandId is random every time I press same button, and I see no way to find out, which button of my 2-gang is pressed....

The problem seems that GreenPower devices works with high level encryption. So it's not device limitation, but software limitation.

BTW I'm here with the same request.  I test it with tuya hub with smartlife app and works really nice. I'm waiting for zigbee2mqtt support! I tried with custom adapter, but there is no way to debug it or understand how to intercept single click actions

igor-semenov, 2024-04-09:
> BTW I'm here with the same request. I test it with tuya hub with smartlife app and works really nice. I'm waiting for zigbee2mqtt support! I tried with custom adapter, but there is no way to debug it or understand how to intercept single click actions

Could you please share link to tuya hub, which you use?

Namyts, 2024-04-09:
I also bought this device recently without realising the lack of compatibility with ZGP with an existing network. I agree with @psummo that this looks like it's a software issue but it should also have a software fix. The fact that other's have gotten similar devices working in their network by adding ZGP interpreters/translators/re-publishers (eg the IKEA lightbulb, certain Phillips Hue devices) mean that's the "encrypted" signal from this switch isn't vendor locked, instead there is just missing functionality in the network which could be fulfilled by another device (maybe the device running zigbee2mqtt) (?)    
What am I misunderstanding?

psummo, 2024-04-09:
> Could you please share link to tuya hub, which you use?

I bought the hub 1 year ago. I swear that I didn't know that the hub is Moes also ahahah I noticed it now. Btw it's a normal tuya hub that you can buy on aliexpress also. The model is MoesGo ZHUB-W, Zigbee 3.0. If anybody know /want try investigate with this hub in some way in order to understand what happen with this kind of switch, I'm available

psummo, 2024-04-09:
> I also bought this device recently without realising the lack of compatibility with ZGP with an existing network. I agree with @psummo that this looks like it's a software issue but it should also [...]

The problem could be also in the zigbee dongle (I've the sonoff) at this point, I'm newbie in this so I'm not able to understand if it's only the software (zigbee2mqtt) or also the firmware

Namyts, 2024-04-09:
I also use the same dongle with the multiprotocol firmware (& docker container). Same questions too. The fact we can see the logs makes me think it's something that can be interpreted, but it also feels like there might be some missing information to translate this information into exact button presses... 

For record, below are my almost identical logs:

```
Zigbee2MQTT:debug 2024-04-09 10:49:22: Received Zigbee message from 'James Kinetic Switch', type 'commandNotification', cluster 'greenPower', data '{"commandID":126,"frameCounter":222,"srcID":1939043194}' from endpoint 242 with groupID null
Zigbee2MQTT:debug 2024-04-09 10:49:22: Skipping message, still interviewing
Zigbee2MQTT:info  2024-04-09 10:49:22: MQTT publish: topic 'zigbee2mqtt/James Kinetic Switch', payload '{"last_seen":"2024-04-09T09:49:22.916Z","linkquality":212}'
Zigbee2MQTT:info  2024-04-09 10:49:22: MQTT publish: topic 'zigbee2mqtt/James Kinetic Switch/last_seen', payload '2024-04-09T09:49:22.916Z'
Zigbee2MQTT:info  2024-04-09 10:49:22: MQTT publish: topic 'zigbee2mqtt/James Kinetic Switch/linkquality', payload '212'
Zigbee2MQTT:debug 2024-04-09 10:49:23: Received Zigbee message from 'James Kinetic Switch', type 'commandNotification', cluster 'greenPower', data '{"commandID":95,"frameCounter":231,"srcID":1939043194}' from endpoint 242 with groupID null
Zigbee2MQTT:debug 2024-04-09 10:49:23: Skipping message, still interviewing
Zigbee2MQTT:info  2024-04-09 10:49:23: MQTT publish: topic 'zigbee2mqtt/James Kinetic Switch', payload '{"last_seen":"2024-04-09T09:49:23.865Z","linkquality":210}'
Zigbee2MQTT:info  2024-04-09 10:49:23: MQTT publish: topic 'zigbee2mqtt/James Kinetic Switch/last_seen', payload '2024-04-09T09:49:23.865Z'
Zigbee2MQTT:info  2024-04-09 10:49:23: MQTT publish: topic 'zigbee2mqtt/James Kinetic Switch/linkquality', payload '210'
```

ngeissel, 2024-04-09:
> I also use the same dongle with the multiprotocol firmware (& docker container). Same questions too. The fact we can see the logs makes me think it's something that can be interpreted, but it also [...]

How did you manage to get such clean logs ? This could help us to go forward

Namyts, 2024-04-09:
I am running zigbee2mqtt in a docker container using docker compose. It looks like this:
```
services:
  zigbee2mqtt:
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt
    restart: unless-stopped
    volumes:
      - ./data:/app/data
      - /run/udev:/run/udev:ro
      - ./switch.js:/app/data/switch.js
    ports:
      - 8080:8080
    network_mode: host
    environment:
      - TZ=Europe/London
```
In the `/data/configuration.yaml` I have a few lines like this:
```
advanced:
  log_level: debug
  output: attribute_and_json
  last_seen: ISO_8601
```
I then look at the logs by using:
`docker logs -f zigbee2mqtt`

psummo, 2024-04-09:
> @ngeissel I am running zigbee2mqtt in a docker container using docker compose. It looks like this:
> 
> ```
> services:
>   zigbee2mqtt:
>     container_name: zigbee2mqtt
>     image: koenkk/zigbee2mqtt
>     restart: unless-stopped
>     volumes:
>       - ./data:/app/data
>       - /run/udev:/run/udev:ro
>       - ./switch.js:/app/data/switch.js
>     ports:
>       - 8080:8080
>     network_mode: host
>     environment:
>       - TZ=Europe/London
> ```
> 
> In the `/data/configuration.yaml` I have a few lines like this:
> 
> ```
> advanced:
>   log_level: debug
>   output: attribute_and_json
>   last_seen: ISO_8601
> ```
> 
> I then look at the logs by using: `docker logs -f zigbee2mqtt`

without any custom converter?

psummo, 2024-04-09:
@Namyts without any custom converter?

Namyts, 2024-04-09:
The switch.js file is a placeholder, sadly I don't have the knowledge yet to successfully write the converter. That's one of the reasons I am here :)

psummo, 2024-04-09:
I don't know if there is the way to log everything, I need it as debugging

psummo, 2024-04-09:
I try with this guide. 
https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html 
and
https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#instructions

And this is what I produced
[converter.js.txt](https://github.com/Koenkk/zigbee2mqtt/files/14924313/converter.js.txt)
The file is JS, but since of limitation on github I can only update txt. Btw Seems nothing is change. and I don't know why, sometimes converter is "recognized" and z2m doesn't give me error during pairing, sometimes yes.

I'm not expert and I tried only create the adapter following the guide for tuya.

Namyts, 2024-04-17:
I've messaged MoesHouse for assistance. Hopefully they can help explain the protocol better.    
I haven't been able to improve on your converter @psummo :(     
```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const {hasAlreadyProcessedMessage} = require('zigbee-herdsman-converters/lib/utils');

const my_converter = {
	cluster: 'greenPower',
	type: ['commandNotification', 'commandCommissioningNotification'],
	convert: (model, msg) => {
		const commandID = msg.data.commandID;
		if (hasAlreadyProcessedMessage(msg, model, msg.data.frameCounter, `${msg.device.ieeeAddr}_${commandID}`)) {return}
		if (commandID === 224) return;

		return {action: 'left'}
	}
}

const definition = {
    fingerprint: [
        {
            modelID: 'GreenPower_2',
            manufacturerName: undefined,
        },
    ],
    model: 'GreenPower_2',
    vendor: 'GreenPower_2',
    description: 'kinetic switch',
	meta: {multiEndpoint: true},
    fromZigbee: [tuya.fz.datapoints, fz.ignore_basic_report, my_converter],
	toZigbee: [],
	configure: tuya.configureMagicPacket,
	exposes: []
};

module.exports = definition;

```

igor-semenov, 2024-05-12:
> I've messaged MoesHouse for assistance. Hopefully they can help explain the protocol better.

Have you got any response from MoesHouse?

Namyts, 2024-05-13:
> Have you got any response from MoesHouse?

They replied the next day, but after exchanging a couple of emails they seem unwilling to help. Sample from response:

> It is recommended that you go to this website to consult professionals to get solutions---https://www.zigbee2mqtt.io/

and

> Sorry, we can't provide it for you. This zigbee is Tuya's protocol, and Tuya has not opened it to us.

I implore you to email them yourself (use their website contact form to establish comms). They response times aren't bad...    
It is a shame because if they provide sufficient information, we might be able to add compatibility for their product with an entirely new ecosystem...

mathieuruellan, 2024-05-13:
Have you asked with the right wording?

https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html#_4-mapping-the-datapoints

[image]

Namyts, 2024-05-13:
@mathieuruellan - not quite. 
Me: 
> Hi,
I am trying to integrate your device: https://moeshouse.com/products/zigbee-wireless-self-powered-scene-switch with Zigbee2MQTT but I am having some issues.
Please could you send me the UART protocol for the device? Or maybe send the Zigbee protocol?
Currently it looks like the device sends a random commandId every time I press either switch, but I am certain that with your help I can get this integration working! This will help me and many others :) 
Kind regards,

Moes Response:
> We recommend using our zigbee device on smart life/tuya smart/MOES app.
Since Home Assistant is an open source platform and cannot be used commercially, it is mainly aimed at individual enthusiast users;
Our zigbee devices can be used normally when connected to the gateway. If you use this platform, there is no guarantee that there will be any problems
It is recommended that you go to this website to consult professionals to get solutions---https://www.zigbee2mqtt.io/

Me: 
> Thanks for replying so quickly.
I understand that Tuya is the hub which Moes recommends. 
Zigbee2MQTT is also an open source tool, and compatibility is added by software developers such as myself!
https://github.com/Koenkk/zigbee-herdsman-converters/tree/master/src/devices - you can see in this page a huge list of supported products for Zigbee2MQTT... including some of Moes products https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/moes.ts !
I am trying to provide compatibility for the Moes self-powered switch (you're welcome 🙂 ), but am having a few issues because the command id sent by the device each time a switch is pressed appears to be random... 
I am contacting you to ask if you could provide some documentation around the protocol that your switch is using, and with this information I can write the code to make your switch compatible with Zigbee2MQTT (which will also make your device compatible with a LOT of other technologies! (for example: home assistant))
I hope you can help me by sending this info,
Kind regards,

Moes Response:
> Sorry, we can't provide it for you. This zigbee is Tuya's protocol, and Tuya has not opened it to us.

I welcome anybody else to send them an email to more clearly ask them for the info we'll need to resolve this issue :) Perhaps more emails would show more interest? Or it might land with a more co-operative support agent. Not too sure how we will resolve this issue without some external help, or someone with deeper understanding of ZGP.

MaxwellJK, 2024-05-18:
I wrote them just now - i'll keep you all posted.
I am not sure i have the knowledge to work on the implementation myself but i'll share here anything useful (if they sahre it with me)

MaxwellJK, 2024-05-22:
`This zigbee is Tuya's protocol, and Tuya has not opened it to us.`
Quick update - i was given the same kind of answer unfortunately
Either they don't want to share the details or it is true that this zigbee protocol is owned by Tuya and they don't have any technical details. I was thinking to reach out to Tuya and check if they are willing to share the details.

gurrier, 2024-07-08:
@Namyts Just noticed... Should the manufacturerName: undefined not have quotes?
like this...
manufacturerName: 'undefined'

CokACola44, 2024-08-14:
Hello everybody,

I also made the mistake of buying this switch without checking its compatibility with Zigbee2MQTT. I thought it was the best solution because I didn’t want to control the power by installing a connected switch on my wall. I want to be able to associate scenes with button presses and avoid using a battery to eliminate the need to change it every 3 months.

I was able to pair the device with Zigbee2MQTT, but I’m not receiving any responses or logs from it when pressing the two buttons.

Note: As of now, no device using kinetic energy harvesting technology has been added to the list of compatible devices with Zigbee2MQTT.

rdorys, 2024-08-24:
@CokACola44 
I have several kinetic switchs that are perfectly working with z2m
For example https://amzn.eu/d/4oDHkk6

I can confirm that group binding isn’t supported in zigbee green power specification (yes it is sad….)

TarasRybak, 2024-09-05:
Hi. 
Found this information, can it help with the problem?
https://community.silabs.com/s/article/what-is-zigbee-green-power-x?language=en_US

https://www.silabs.com/documents/public/user-guides/ug103-15-green-power-fundamentals.pdf
https://www.silabs.com/documents/public/user-guides/ug392-using-sl-green-power-with-ezp.pdf

gurrier, 2025-01-04:
I just tried again to pair with Z2M latest version 2.  Same behaviour as before, it pairs as an unsuppported Greenpower device and exposes nothing.  It's such a nicely designed device, it would be great to have it working.

N41R0D, 2025-01-06:
Hello there,
I've never worked on new devices support but I want to try on this product.
So i've started to read "[Support new Tuya devices](https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html#_2-adding-your-device)" and install a new "test env" to work on but I don't understand how to debug (my JS weakness).

If someone without the device but with experiences could explain how to debug [Tuya datapoints](https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html#_3-understanding-tuya-datapoints) I would try my best.

Last thing, I don't understand the link between the converter and the datapoints (the @Namyts [one](https://github.com/Koenkk/zigbee2mqtt/issues/19405#issuecomment-2061530011) by example).

Ten0, 2025-01-09:
Tuya seems to be documenting their zigbee devices now: https://developer.tuya.com/en/docs/connect-subdevices-to-gateways/Zigbee_2?id=Kcww7qppbe87m
However I couldn't find the doc for the self-powered switch there. But maybe some of the other switches documented there have the same behavior?

skuizy, 2025-02-16:
Hello,

Just made the mistake ok buying this same kinetic switch, advertized as zigbee 3.0 compliant, thinking it would be compatible with Z2M...
My understanding of the problem is that ZGP payload is encrypted and Z2M doesn't know how to handle this YET !
As I didn't see any pairing logs posted in this thread, i went ahead and it seems thers a key exchange hapenning with some bitmask shared (at 2025-02-16T18:59:37.863173339Z in the logs below). Can this be useful for decoding further frames or is this a dumb question from someone new to these subjects ?

```
zigbee2mqtt  | 2025-02-16T18:59:37.273504412Z [2025-02-16 18:59:37] debug:      zh:ember:uart:ash: <--- [FRAME type=DATA]
zigbee2mqtt  | 2025-02-16T18:59:37.273879182Z [2025-02-16 18:59:37] debug:      zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3](ackRx=3 frmTx=3)
zigbee2mqtt  | 2025-02-16T18:59:37.274417802Z [2025-02-16 18:59:37] debug:      zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3 frmNum=4](frmRx=4) Added to rxQueue
zigbee2mqtt  | 2025-02-16T18:59:37.274874386Z [2025-02-16 18:59:37] debug:      zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=5](ackRx=3)
zigbee2mqtt  | 2025-02-16T18:59:37.275922903Z [2025-02-16 18:59:37] debug:      zh:ember:ezsp: <=== [CBFRAME: ID=197:"GPEP_INCOMING_MESSAGE_HANDLER" Seq=42 Len=76]
zigbee2mqtt  | 2025-02-16T18:59:37.277007938Z [2025-02-16 18:59:37] debug:      zh:ember:ezsp: ezspGpepIncomingMessageHandler(): callback called with: [status=124], [gpdLink=210], [sequenceNumber=99], [addr={"applicationId":0,"sourceId":3977565935,"endpoint":0}], [gpdfSecurityLevel=NONE], [gpdfSecurityKeyType=NONE], [autoCommissioning=false], [bidirectionalInfo=0], [gpdSecurityFrameCounter=4294967295], [gpdCommandId=224], [mic=4294967295], [proxyTableIndex=255], [gpdCommandPayload=0289f3a1a29292cac37dbfa4e85e0a7a6ca618ee17997c6307000037647a62697061304047503030303197], [packetInfo={"senderShortId":65535,"senderLongId":"0xFFFFFFFFFFFFFFFF","bindingIndex":255,"addressIndex":255,"lastHopLqi":210,"lastHopRssi":0,"lastHopTimestamp":0}]
zigbee2mqtt  | 2025-02-16T18:59:37.279607796Z [2025-02-16 18:59:37] info:       zh:controller:greenpower: Received commissioning from '55023'
zigbee2mqtt  | 2025-02-16T18:59:37.283019897Z [2025-02-16 18:59:37] debug:      zh:controller:greenpower: Payload.Options: 58696 wasBroadcast: true
zigbee2mqtt  | 2025-02-16T18:59:37.287149726Z [2025-02-16 18:59:37] debug:      zh:ember: ~~~> [ZCL BROADCAST apsFrame={"profileId":41440,"clusterId":33,"sourceEndpoint":242,"destinationEndpoint":242,"options":4416,"groupId":65533,"sequence":0} header={"frameControl":{"reservedBits":0,"frameType":1,"direction":1,"disableDefaultResponse":true,"manufacturerSpecific":false},"transactionSequenceNumber":2,"commandIdentifier":1}]
zigbee2mqtt  | 2025-02-16T18:59:37.288944210Z [2025-02-16 18:59:37] debug:      zh:ember:ezsp: ===> [FRAME: ID=54:"SEND_BROADCAST" Seq=43 Len=54]
zigbee2mqtt  | 2025-02-16T18:59:37.289519829Z [2025-02-16 18:59:37] debug:      zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=3 frmRx=5](ackRx=3)
zigbee2mqtt  | 2025-02-16T18:59:37.313262117Z [2025-02-16 18:59:37] debug:      zh:ember:uart:ash: <--- [FRAME type=DATA]
zigbee2mqtt  | 2025-02-16T18:59:37.313800662Z [2025-02-16 18:59:37] debug:      zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4](ackRx=3 frmTx=4)
zigbee2mqtt  | 2025-02-16T18:59:37.314313912Z [2025-02-16 18:59:37] debug:      zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4 frmNum=5](frmRx=5) Added to rxQueue
zigbee2mqtt  | 2025-02-16T18:59:37.316226598Z [2025-02-16 18:59:37] debug:      zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=6](ackRx=4)
zigbee2mqtt  | 2025-02-16T18:59:37.317994879Z [2025-02-16 18:59:37] debug:      zh:ember:ezsp: <=== [FRAME: ID=54:"SEND_BROADCAST" Seq=43 Len=7]
zigbee2mqtt  | 2025-02-16T18:59:37.319283799Z [2025-02-16 18:59:37] debug:      zh:ember:ezsp: ~~~> [SENT type=BROADCAST apsSequence=172 messageTag=1 status=OK]
zigbee2mqtt  | 2025-02-16T18:59:37.326313905Z [2025-02-16 18:59:37] debug:      zh:ember:uart:ash: <--- [FRAME type=DATA]
zigbee2mqtt  | 2025-02-16T18:59:37.328118575Z [2025-02-16 18:59:37] debug:      zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4](ackRx=4 frmTx=4)
zigbee2mqtt  | 2025-02-16T18:59:37.328179981Z [2025-02-16 18:59:37] debug:      zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4 frmNum=6](frmRx=6) Added to rxQueue
zigbee2mqtt  | 2025-02-16T18:59:37.328954947Z [2025-02-16 18:59:37] debug:      zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=7](ackRx=4)
zigbee2mqtt  | 2025-02-16T18:59:37.330878559Z [2025-02-16 18:59:37] debug:      zh:ember:ezsp: <=== [CBFRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=43 Len=57]
zigbee2mqtt  | 2025-02-16T18:59:37.337651726Z [2025-02-16 18:59:37] debug:      zh:ember:ezsp: ezspIncomingMessageHandler(): callback called with: [type=BROADCAST_LOOPBACK], [apsFrame={"profileId":41440,"clusterId":33,"sourceEndpoint":242,"destinationEndpoint":242,"options":256,"groupId":0,"sequence":172}], [packetInfo:{"senderShortId":0,"senderLongId":"0xFFFFFFFFFFFFFFFF","bindingIndex":255,"addressIndex":255,"lastHopLqi":255,"lastHopRssi":0,"lastHopTimestamp":0}], [messageContents=19020148e500efd614ed840b0263070000c0c1c2c3c4c5c6c7c8c9cacbcccdcecf]
zigbee2mqtt  | 2025-02-16T18:59:37.821378672Z [2025-02-16 18:59:37] debug:      zh:controller: Green power device '{"sourceID":3977565935,"deviceID":2,"networkAddress":55023}' joined
[... cut ...]
zigbee2mqtt  | 2025-02-16T18:59:38.287414419Z [2025-02-16 18:59:38] debug:      zh:ember:ezsp: <=== [CBFRAME: ID=63:"MESSAGE_SENT_HANDLER" Seq=43 Len=22]
zigbee2mqtt  | 2025-02-16T18:59:38.288658821Z [2025-02-16 18:59:38] debug:      zh:ember:ezsp: ezspMessageSentHandler(): callback called with: [status=OK], [type=BROADCAST], [indexOrDestination=65533], [apsFrame={"profileId":41440,"clusterId":33,"sourceEndpoint":242,"destinationEndpoint":242,"options":256,"groupId":0,"sequence":172}], [messageTag=1]
```

tiagores, 2025-02-28:
+1 who bought this device without checking properly. 
I have also been tinkering with an external converter, but no luck so far.

Nerivec, 2025-03-15:
@skuizy can you provide a `debug` log after pairing, where you trigger each possible button on the device (short press, long press, etc.) so we can see what payloads come in to Z2M (doesn't matter that it says unsupported, the payloads will still be logged)?
_Go easy on how fast you do this, better to leave a few seconds pass between each press so the logs are clearer._

skuizy, 2025-03-16:
OK, so I've done the full pairing again to get a new key exchange.
This device doesn't support short/long press as far as I'm aware, just simple press.
In the folowing logs I've got consecutive press on each button and alternative press.
I'll be happy to provide more logs if needed !

Pairing :
```
zigbee2mqtt  | 2025-03-16T10:57:28.060996445Z [2025-03-16 10:57:28] debug:      zh:ember:uart:ash: <--- [FRAME type=DATA]
zigbee2mqtt  | 2025-03-16T10:57:28.062412012Z [2025-03-16 10:57:28] debug:      zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=2](ackRx=2 frmTx=2)
zigbee2mqtt  | 2025-03-16T10:57:28.063239590Z [2025-03-16 10:57:28] debug:      zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=2 frmNum=3](frmRx=3) Added to rxQueue
zigbee2mqtt  | 2025-03-16T10:57:28.064557603Z [2025-03-16 10:57:28] debug:      zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=4](ackRx=2)
zigbee2mqtt  | 2025-03-16T10:57:28.066853172Z [2025-03-16 10:57:28] debug:      zh:ember:ezsp: <=== [CBFRAME: ID=197:"GPEP_INCOMING_MESSAGE_HANDLER" Seq=49 Len=76]
zigbee2mqtt  | 2025-03-16T10:57:28.068506531Z [2025-03-16 10:57:28] debug:      zh:ember:ezsp: ezspGpepIncomingMessageHandler(): callback called with: [status=NO_SECURITY], [gpdLink=214], [sequenceNumber=19], [addr={"applicationId":0,"sourceId":1496140231,"endpoint":0}], [gpdfSecurityLevel=NONE], [gpdfSecurityKeyType=NONE], [autoCommissioning=false], [bidirectionalInfo=0], [gpdSecurityFrameCounter=4294967295], [gpdCommandId=224], [mic=4294967295], [proxyTableIndex=255], [gpdCommandPayload=0289f31adb70a88d71196ee50c03580537767de27ad5331309000037647a62697061304047503030303157], [packetInfo={"senderShortId":65535,"senderLongId":"0xFFFFFFFFFFFFFFFF","bindingIndex":255,"addressIndex":255,"lastHopLqi":214,"lastHopRssi":0,"lastHopTimestamp":0}]
zigbee2mqtt  | 2025-03-16T10:57:28.070999504Z [2025-03-16 10:57:28] info:       zh:controller:greenpower: Received commissioning from '18887'
zigbee2mqtt  | 2025-03-16T10:57:28.079238318Z [2025-03-16 10:57:28] debug:      zh:controller:greenpower: Payload.Options: 58696 wasBroadcast: true
zigbee2mqtt  | 2025-03-16T10:57:28.083011601Z [2025-03-16 10:57:28] debug:      zh:ember: ~~~> [ZCL BROADCAST apsFrame={"profileId":41440,"clusterId":33,"sourceEndpoint":242,"destinationEndpoint":242,"options":4416,"groupId":65533,"sequence":0} header={"frameControl":{"reservedBits":0,"frameType":1,"direction":1,"disableDefaultResponse":true,"manufacturerSpecific":false},"transactionSequenceNumber":4,"commandIdentifier":1}]
zigbee2mqtt  | 2025-03-16T10:57:28.084117785Z [2025-03-16 10:57:28] debug:      zh:ember:ezsp: ===> [FRAME: ID=54:"SEND_BROADCAST" Seq=50 Len=58]
zigbee2mqtt  | 2025-03-16T10:57:28.085283708Z [2025-03-16 10:57:28] debug:      zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=2 frmRx=4](ackRx=2)
zigbee2mqtt  | 2025-03-16T10:57:28.107881853Z [2025-03-16 10:57:28] debug:      zh:ember:uart:ash: <--- [FRAME type=DATA]
zigbee2mqtt  | 2025-03-16T10:57:28.108703690Z [2025-03-16 10:57:28] debug:      zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3](ackRx=2 frmTx=3)
zigbee2mqtt  | 2025-03-16T10:57:28.109991537Z [2025-03-16 10:57:28] debug:      zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3 frmNum=4](frmRx=4) Added to rxQueue
zigbee2mqtt  | 2025-03-16T10:57:28.110370363Z [2025-03-16 10:57:28] debug:      zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=5](ackRx=3)
zigbee2mqtt  | 2025-03-16T10:57:28.112221904Z [2025-03-16 10:57:28] debug:      zh:ember:ezsp: <=== [FRAME: ID=54:"SEND_BROADCAST" Seq=50 Len=10]
zigbee2mqtt  | 2025-03-16T10:57:28.113727636Z [2025-03-16 10:57:28] debug:      zh:ember:ezsp: ~~~> [SENT type=BROADCAST apsSequence=100 messageTag=2 status=OK]
zigbee2mqtt  | 2025-03-16T10:57:28.156032886Z [2025-03-16 10:57:28] debug:      zh:ember:uart:ash: <--- [FRAME type=DATA]
zigbee2mqtt  | 2025-03-16T10:57:28.156597135Z [2025-03-16 10:57:28] debug:      zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3](ackRx=3 frmTx=3)
zigbee2mqtt  | 2025-03-16T10:57:28.157333140Z [2025-03-16 10:57:28] debug:      zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3 frmNum=5](frmRx=5) Added to rxQueue
zigbee2mqtt  | 2025-03-16T10:57:28.158146551Z [2025-03-16 10:57:28] debug:      zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=6](ackRx=3)
zigbee2mqtt  | 2025-03-16T10:57:28.159177255Z [2025-03-16 10:57:28] debug:      zh:ember:ezsp: <=== [CBFRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=50 Len=69]
zigbee2mqtt  | 2025-03-16T10:57:28.160353567Z [2025-03-16 10:57:28] debug:      zh:ember:ezsp: ezspIncomingMessageHandler(): callback called with: [type=BROADCAST_LOOPBACK], [apsFrame={"profileId":41440,"clusterId":33,"sourceEndpoint":242,"destinationEndpoint":242,"options":256,"groupId":0,"sequence":100}], [packetInfo:{"senderShortId":0,"senderLongId":"0x0000000000000000","bindingIndex":255,"addressIndex":8,"lastHopLqi":255,"lastHopRssi":0,"lastHopTimestamp":1656832634}], [messageContents=19040148e500c7492d59840b0213090000c0c1c2c3c4c5c6c7c8c9cacbcccdcecf]
zigbee2mqtt  | 2025-03-16T10:57:28.616758222Z [2025-03-16 10:57:28] debug:      zh:controller: Green power device '{"sourceID":1496140231,"deviceID":2,"networkAddress":18887}' joined
zigbee2mqtt  | 2025-03-16T10:57:28.617757001Z [2025-03-16 10:57:28] debug:      zh:controller: New green power device '0x00000000592d49c7' joined
zigbee2mqtt  | 2025-03-16T10:57:28.619217641Z [2025-03-16 10:57:28] debug:      zh:controller: Creating device '0x00000000592d49c7'
[... cut ...]
zigbee2mqtt  | 2025-03-16T11:02:57.516656741Z [2025-03-16 11:02:57] info:       z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x00000000592d49c7', payload '{"last_seen":"2025-03-16T11:02:57.505Z","linkquality":222}'
zigbee2mqtt  | 2025-03-16T11:02:57.520779314Z [2025-03-16 11:02:57] info:       z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x00000000592d49c7/last_seen', payload '2025-03-16T11:02:57.505Z'
zigbee2mqtt  | 2025-03-16T11:02:57.523162419Z [2025-03-16 11:02:57] info:       z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x00000000592d49c7/linkquality', payload '222'
```

huafu, 2025-03-18:
I've got a 3 buttons device, here are my logs too if it can help.

<details>
<summary>Pairing (I hope there is no garbage in this one since I have multiple devices on my network too)</summary>

```
zigbee2mqtt  | [2025-03-18 08:48:12] debug:     zh:ember:uart:ash: <--- [FRAME type=DATA]
zigbee2mqtt  | [2025-03-18 08:48:12] debug:     zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=0](ackRx=0 frmTx=0)
zigbee2mqtt  | [2025-03-18 08:48:12] debug:     zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=0 frmNum=0](frmRx=0) Added to rxQueue
zigbee2mqtt  | [2025-03-18 08:48:12] debug:     zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=1](ackRx=0)
zigbee2mqtt  | [2025-03-18 08:48:12] debug:     zh:ember:ezsp: <=== [CBFRAME: ID=197:"GPEP_INCOMING_MESSAGE_HANDLER" Seq=135 Len=76]
zigbee2mqtt  | [2025-03-18 08:48:12] debug:     zh:ember:ezsp: ezspGpepIncomingMessageHandler(): callback called with: [status=124], [gpdLink=219], [sequenceNumber=139], [addr={"applicationId":0,"sourceId":344902069,"endpoint":0}], [gpdfSecurityLevel=NONE], [gpdfSecurityKeyType=NONE], [autoCommissioning=false], [bidirectionalInfo=0], [gpdSecurityFrameCounter=4294967295], [gpdCommandId=224], [mic=4294967295], [proxyTableIndex=255], [gpdCommandPayload=0289f35690230a93ea5f1951926f200236c7820891812a8b0400007165726837706f7840475030303031be], [packetInfo={"senderShortId":65535,"senderLongId":"0xFFFFFFFFFFFFFFFF","bindingIndex":255,"addressIndex":255,"lastHopLqi":219,"lastHopRssi":0,"lastHopTimestamp":0}]
zigbee2mqtt  | [2025-03-18 08:48:12] info:      zh:controller:greenpower: Received commissioning from '51637'
zigbee2mqtt  | [2025-03-18 08:48:12] debug:     zh:controller:greenpower: Payload.Options: 58696 wasBroadcast: true
zigbee2mqtt  | [2025-03-18 08:48:12] debug:     zh:ember: ~~~> [ZCL BROADCAST apsFrame={"profileId":41440,"clusterId":33,"sourceEndpoint":242,"destinationEndpoint":242,"options":4416,"groupId":65533,"sequence":0} header={"frameControl":{"reservedBits":0,"frameType":1,"direction":1,"disableDefaultResponse":true,"manufacturerSpecific":false},"transactionSequenceNumber":132,"commandIdentifier":1}]
zigbee2mqtt  | [2025-03-18 08:48:12] debug:     zh:ember:ezsp: ===> [FRAME: ID=54:"SEND_BROADCAST" Seq=136 Len=54]
zigbee2mqtt  | [2025-03-18 08:48:12] debug:     zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=0 frmRx=1](ackRx=0)
zigbee2mqtt  | [2025-03-18 08:48:12] debug:     zh:ember:uart:ash: <--- [FRAME type=DATA]
zigbee2mqtt  | [2025-03-18 08:48:12] debug:     zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=0](ackRx=0 frmTx=1)
zigbee2mqtt  | [2025-03-18 08:48:12] debug:     zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=0 frmNum=1](frmRx=1) Added to rxQueue
zigbee2mqtt  | [2025-03-18 08:48:12] debug:     zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=2](ackRx=0)
zigbee2mqtt  | [2025-03-18 08:48:12] debug:     zh:ember:ezsp: <=== [CBFRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=135 Len=85]
zigbee2mqtt  | [2025-03-18 08:48:12] debug:     zh:ember:ezsp: ezspIncomingMessageHandler(): callback called with: [type=BROADCAST], [apsFrame={"profileId":41440,"clusterId":33,"sourceEndpoint":242,"destinationEndpoint":242,"options":256,"groupId":0,"sequence":127}], [packetInfo:{"senderShortId":51637,"senderLongId":"0xFFFFFFFFFFFFFFFF","bindingIndex":255,"addressIndex":255,"lastHopLqi":188,"lastHopRssi":-53,"lastHopTimestamp":0}], [messageContents=1103040008b5c98e148b000000e02b0289f35690230a93ea5f1951926f200236c7820891812a8b0400007165726837706f7840475030303031be06668d]
zigbee2mqtt  | [2025-03-18 08:48:12] info:      zh:controller:greenpower: Received commissioning from '51637'
zigbee2mqtt  | [2025-03-18 08:48:12] debug:     zh:controller:greenpower: Payload.Options: 58696 wasBroadcast: true
zigbee2mqtt  | [2025-03-18 08:48:12] debug:     zh:ember: ~~~> [ZCL BROADCAST apsFrame={"profileId":41440,"clusterId":33,"sourceEndpoint":242,"destinationEndpoint":242,"options":4416,"groupId":65533,"sequence":0} header={"frameControl":{"reservedBits":0,"frameType":1,"direction":1,"disableDefaultResponse":true,"manufacturerSpecific":false},"transactionSequenceNumber":133,"commandIdentifier":1}]
zigbee2mqtt  | [2025-03-18 08:48:12] debug:     zh:ember:uart:ash: <--- [FRAME type=DATA]
zigbee2mqtt  | [2025-03-18 08:48:12] debug:     zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1](ackRx=0 frmTx=1)
zigbee2mqtt  | [2025-03-18 08:48:12] debug:     zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1 frmNum=2](frmRx=2) Added to rxQueue
zigbee2mqtt  | [2025-03-18 08:48:12] debug:     zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=3](ackRx=1)
zigbee2mqtt  | [2025-03-18 08:48:12] debug:     zh:ember:ezsp: <=== [FRAME: ID=54:"SEND_BROADCAST" Seq=136 Len=7]
zigbee2mqtt  | [2025-03-18 08:48:12] debug:     zh:ember:ezsp: ===> [FRAME: ID=54:"SEND_BROADCAST" Seq=137 Len=54]
zigbee2mqtt  | [2025-03-18 08:48:12] debug:     zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=1 frmRx=3](ackRx=1)
zigbee2mqtt  | [2025-03-18 08:48:12] debug:     zh:ember:ezsp: ~~~> [SENT type=BROADCAST apsSequence=89 messageTag=36 status=OK]
zigbee2mqtt  | [2025-03-18 08:48:12] debug:     zh:ember:uart:ash: <--- [FRAME type=DATA]
zigbee2mqtt  | [2025-03-18 08:48:12] debug:     zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1](ackRx=1 frmTx=2)
zigbee2mqtt  | [2025-03-18 08:48:12] debug:     zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1 frmNum=3](frmRx=3) Added to rxQueue
zigbee2mqtt  | [2025-03-18 08:48:12] debug:     zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=4](ackRx=1)
[... cut ...]
zigbee2mqtt  | [2025-03-18 08:55:00] debug:     z2m: Received Zigbee message from '0x00000000148ec9b5', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{},"commandID":219,"frameCounter":1217,"options":0,"payloadSize":0,"srcID":344902069}' from endpoint 242 with groupID 2948
zigbee2mqtt  | [2025-03-18 08:55:00] debug:     z2m: Skipping message, still interviewing
zigbee2mqtt  | [2025-03-18 08:55:00] info:      z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x00000000148ec9b5', payload '{"last_seen":"2025-03-18T08:55:00.669Z","linkquality":225}'
```

</details>

Nerivec, 2025-03-18:
The decryption should be supported after https://github.com/Koenkk/zigbee-herdsman/pull/1349 is merged (used these last logs as reference vectors, thanks!). However, remains finding a way to implement a converter for it, because from what I can tell, it doesn't have a fixed source ID, which is what is usually used to identify GP devices (not much else available...).

huafu, 2025-03-18:
Thanks @Nerivec!

Here are the new logs with the `:latest-dev` docker tag, lemme know if I can do anything more to test/help

<details>
<summary>Pairing</summary>

```
zigbee2mqtt  | [2025-03-18 21:07:35] debug:     zh:ember:uart:ash: <--- [FRAME type=DATA]
zigbee2mqtt  | [2025-03-18 21:07:35] debug:     zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=6](ackRx=6 frmTx=6)
zigbee2mqtt  | [2025-03-18 21:07:35] debug:     zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=6 frmNum=4](frmRx=4) Added to rxQueue
zigbee2mqtt  | [2025-03-18 21:07:35] debug:     zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=5](ackRx=6)
zigbee2mqtt  | [2025-03-18 21:07:35] debug:     zh:ember:ezsp: <=== [CBFRAME: ID=197:"GPEP_INCOMING_MESSAGE_HANDLER" Seq=197 Len=76]
zigbee2mqtt  | [2025-03-18 21:07:35] debug:     zh:ember:ezsp: ezspGpepIncomingMessageHandler(): callback called with: [status=124], [gpdLink=214], [sequenceNumber=50], [addr={"applicationId":0,"sourceId":1398503460,"endpoint":0}], [gpdfSecurityLevel=NONE], [gpdfSecurityKeyType=NONE], [autoCommissioning=false], [bidirectionalInfo=0], [gpdSecurityFrameCounter=4294967295], [gpdCommandId=224], [mic=4294967295], [proxyTableIndex=255], [gpdCommandPayload=0289f3a6b3411bbce10965ad6f32dfb198e6ed40e1e282320600007165726837706f78404750303030317a], [packetInfo={"senderShortId":65535,"senderLongId":"0xFFFFFFFFFFFFFFFF","bindingIndex":255,"addressIndex":255,"lastHopLqi":214,"lastHopRssi":0,"lastHopTimestamp":0}]
zigbee2mqtt  | [2025-03-18 21:07:35] info:      zh:controller:greenpower: [COMMISSIONING] from=30756
zigbee2mqtt  | [2025-03-18 21:07:35] debug:     zh:controller:greenpower: [PAIRING] options=58696 (appId=0 communicationMode=2) wasBroadcast=true gppNwkAddr=undefined
zigbee2mqtt  | [2025-03-18 21:07:35] debug:     zh:ember: ~~~> [ZCL BROADCAST apsFrame={"profileId":41440,"clusterId":33,"sourceEndpoint":242,"destinationEndpoint":242,"options":4416,"groupId":65533,"sequence":0} header={"frameControl":{"reservedBits":0,"frameType":1,"direction":1,"disableDefaultResponse":true,"manufacturerSpecific":false},"transactionSequenceNumber":166,"commandIdentifier":1}]
zigbee2mqtt  | [2025-03-18 21:07:35] debug:     zh:ember:ezsp: ===> [FRAME: ID=54:"SEND_BROADCAST" Seq=198 Len=54]
zigbee2mqtt  | [2025-03-18 21:07:35] debug:     zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=6 frmRx=5](ackRx=6)
zigbee2mqtt  | [2025-03-18 21:07:35] debug:     zh:ember:uart:ash: <--- [FRAME type=DATA]
zigbee2mqtt  | [2025-03-18 21:07:35] debug:     zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=7](ackRx=6 frmTx=7)
zigbee2mqtt  | [2025-03-18 21:07:35] debug:     zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=7 frmNum=5](frmRx=5) Added to rxQueue
zigbee2mqtt  | [2025-03-18 21:07:35] debug:     zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=6](ackRx=7)
zigbee2mqtt  | [2025-03-18 21:07:35] debug:     zh:ember:ezsp: <=== [FRAME: ID=54:"SEND_BROADCAST" Seq=198 Len=7]
zigbee2mqtt  | [2025-03-18 21:07:35] debug:     zh:ember:ezsp: ~~~> [SENT type=BROADCAST apsSequence=63 messageTag=25 status=OK]
zigbee2mqtt  | [2025-03-18 21:07:35] debug:     zh:ember:uart:ash: <--- [FRAME type=DATA]
zigbee2mqtt  | [2025-03-18 21:07:35] debug:     zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=7](ackRx=7 frmTx=7)
zigbee2mqtt  | [2025-03-18 21:07:35] debug:     zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=7 frmNum=6](frmRx=6) Added to rxQueue
zigbee2mqtt  | [2025-03-18 21:07:35] debug:     zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=7](ackRx=7)
zigbee2mqtt  | [2025-03-18 21:07:35] debug:     zh:ember:ezsp: <=== [CBFRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=198 Len=57]
zigbee2mqtt  | [2025-03-18 21:07:35] debug:     zh:ember:ezsp: ezspIncomingMessageHandler(): callback called with: [type=BROADCAST_LOOPBACK], [apsFrame={"profileId":41440,"clusterId":33,"sourceEndpoint":242,"destinationEndpoint":242,"options":256,"groupId":0,"sequence":63}], [packetInfo:{"senderShortId":0,"senderLongId":"0xFFFFFFFFFFFFFFFF","bindingIndex":255,"addressIndex":255,"lastHopLqi":255,"lastHopRssi":0,"lastHopTimestamp":0}], [messageContents=19a60148e50024785b53840b0232060000c0c1c2c3c4c5c6c7c8c9cacbcccdcecf]
zigbee2mqtt  | [2025-03-18 21:07:35] debug:     zh:controller: Green power device '{"sourceID":1398503460,"deviceID":2,"networkAddress":30756,"securityKey":{"type":"Buffer","data":[166,179,65,27,188,225,9,101,173,111,50,223,177,152,230,237]}}' joined
zigbee2mqtt  | [2025-03-18 21:07:35] debug:     zh:controller: New green power device '0x00000000535b7824' joined
zigbee2mqtt  | [2025-03-18 21:07:35] debug:     zh:controller: Creating device '0x00000000535b7824'
zigbee2mqtt  | [2025-03-18 21:07:35] debug:     zh:controller:database: Writing database to '/app/data/database.db'
zigbee2mqtt  | [2025-03-18 21:07:35] debug:     zh:controller:database: Writing database to '/app/data/database.db'
zigbee2mqtt  | [2025-03-18 21:07:35] debug:     zhc: Getting definitions for: sunricher.js,48,sunricher.js,47,sunricher.js,46,philips.js,418,legrand.js,19,led_trading.js,0,enocean.js,1,enocean.js,0
zigbee2mqtt  | [2025-03-18 21:07:35] debug:     zhc: Getting definitions for: sunricher.js,48,sunricher.js,47,sunricher.js,46,philips.js,418,legrand.js,19,led_trading.js,0,enocean.js,1,enocean.js,0
zigbee2mqtt  | [2025-03-18 21:07:35] debug:     zh:controller: Received payload: clusterID=33, address=30756, groupID=2948, endpoint=242, destinationEndpoint=242, wasBroadcast=true, linkQuality=214, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":fa [... cut ...]
zigbee2mqtt  | [2025-03-18 21:13:13] debug:     z2m: Received Zigbee message from '0x00000000535b7824', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{},"commandID":17,"frameCounter":1640,"options":448,"payloadSize":0,"srcID":1398503460}' from endpoint 242 with groupID 2948
zigbee2mqtt  | [2025-03-18 21:13:13] debug:     z2m: Skipping message, still interviewing
zigbee2mqtt  | [2025-03-18 21:13:13] info:      z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x00000000535b7824', payload '{"last_seen":"2025-03-18T21:13:13.351Z","linkquality":216}'

```

</details>

Nerivec, 2025-03-18:
@huafu looks as expected (you can confirm by pressing the same button multiple times, and making sure the `commandID` doesn't change).
```
type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{},"commandID":32
```

Of course the device is still missing a converter, but hopefully this should add support through generated converter (since source ID can't be relied upon to make an actual converter): https://github.com/Koenkk/zigbee-herdsman-converters/pull/9020

Nerivec, 2025-03-18:
@skuizy that latest-dev commit is before support was added (3 days ago) https://github.com/Koenkk/zigbee2mqtt/commits/dev

skuizy, 2025-03-18:
Duh !!! It was indeed outdated, I forgot to run the docker compose up command...
Here we go again on 2.1.3-dev commit: ef5b3de5e272dea03876d88c0417fd98f333a0b1.
commandID stays the same as expected !

<details>
<summary>Pairing :</summary>

```
zigbee2mqtt  | 2025-03-18T22:09:22.408519480Z [2025-03-18 22:09:22] debug:      zh:ember:uart:ash: <--- [FRAME type=DATA]
zigbee2mqtt  | 2025-03-18T22:09:22.409640109Z [2025-03-18 22:09:22] debug:      zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3](ackRx=3 frmTx=3)
zigbee2mqtt  | 2025-03-18T22:09:22.410453261Z [2025-03-18 22:09:22] debug:      zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3 frmNum=6](frmRx=6) Added to rxQueue
zigbee2mqtt  | 2025-03-18T22:09:22.411126398Z [2025-03-18 22:09:22] debug:      zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=7](ackRx=3)
zigbee2mqtt  | 2025-03-18T22:09:22.412579243Z [2025-03-18 22:09:22] debug:      zh:ember:ezsp: <=== [CBFRAME: ID=197:"GPEP_INCOMING_MESSAGE_HANDLER" Seq=34 Len=76]
zigbee2mqtt  | 2025-03-18T22:09:22.418282923Z [2025-03-18 22:09:22] debug:      zh:ember:ezsp: ezspGpepIncomingMessageHandler(): callback called with: [status=NO_SECURITY], [gpdLink=213], [sequenceNumber=61], [addr={"applicationId":0,"sourceId":3297666745,"endpoint":0}], [gpdfSecurityLevel=NONE], [gpdfSecurityKeyType=NONE], [autoCommissioning=false], [bidirectionalInfo=0], [gpdSecurityFrameCounter=4294967295], [gpdCommandId=224], [mic=4294967295], [proxyTableIndex=255], [gpdCommandPayload=0289f309d5dd181e4f36ebec0e1aecf5ad90c17ac66a4a3d0c000037647a626970613040475030303031b7], [packetInfo={"senderShortId":65535,"senderLongId":"0xFFFFFFFFFFFFFFFF","bindingIndex":255,"addressIndex":255,"lastHopLqi":213,"lastHopRssi":0,"lastHopTimestamp":0}]
zigbee2mqtt  | 2025-03-18T22:09:22.422302632Z [2025-03-18 22:09:22] info:       zh:controller:greenpower: [COMMISSIONING] from=26297
zigbee2mqtt  | 2025-03-18T22:09:22.426451635Z [2025-03-18 22:09:22] debug:      zh:controller:greenpower: [PAIRING] options=58696 (appId=0 communicationMode=2) wasBroadcast=true gppNwkAddr=undefined
zigbee2mqtt  | 2025-03-18T22:09:22.428237808Z [2025-03-18 22:09:22] debug:      zh:ember: ~~~> [ZCL BROADCAST apsFrame={"profileId":41440,"clusterId":33,"sourceEndpoint":242,"destinationEndpoint":242,"options":4416,"groupId":65533,"sequence":0} header={"frameControl":{"reservedBits":0,"frameType":1,"direction":1,"disableDefaultResponse":true,"manufacturerSpecific":false},"transactionSequenceNumber":3,"commandIdentifier":1}]
zigbee2mqtt  | 2025-03-18T22:09:22.429090200Z [2025-03-18 22:09:22] debug:      zh:ember:ezsp: ===> [FRAME: ID=54:"SEND_BROADCAST" Seq=35 Len=58]
zigbee2mqtt  | 2025-03-18T22:09:22.429975167Z [2025-03-18 22:09:22] debug:      zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=3 frmRx=7](ackRx=3)
zigbee2mqtt  | 2025-03-18T22:09:22.452509965Z [2025-03-18 22:09:22] debug:      zh:ember:uart:ash: <--- [FRAME type=DATA]
zigbee2mqtt  | 2025-03-18T22:09:22.453340691Z [2025-03-18 22:09:22] debug:      zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4](ackRx=3 frmTx=4)
zigbee2mqtt  | 2025-03-18T22:09:22.454138622Z [2025-03-18 22:09:22] debug:      zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4 frmNum=7](frmRx=7) Added to rxQueue
zigbee2mqtt  | 2025-03-18T22:09:22.455287806Z [2025-03-18 22:09:22] debug:      zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=0](ackRx=4)
zigbee2mqtt  | 2025-03-18T22:09:22.456784132Z [2025-03-18 22:09:22] debug:      zh:ember:ezsp: <=== [FRAME: ID=54:"SEND_BROADCAST" Seq=35 Len=10]
zigbee2mqtt  | 2025-03-18T22:09:22.458081295Z [2025-03-18 22:09:22] debug:      zh:ember:ezsp: ~~~> [SENT type=BROADCAST apsSequence=171 messageTag=2 status=OK]
zigbee2mqtt  | 2025-03-18T22:09:22.503991655Z [2025-03-18 22:09:22] debug:      zh:ember:uart:ash: <--- [FRAME type=DATA]
zigbee2mqtt  | 2025-03-18T22:09:22.504838159Z [2025-03-18 22:09:22] debug:      zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4](ackRx=4 frmTx=4)
zigbee2mqtt  | 2025-03-18T22:09:22.505644886Z [2025-03-18 22:09:22] debug:      zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4 frmNum=0](frmRx=0) Added to rxQueue
zigbee2mqtt  | 2025-03-18T22:09:22.506417317Z [2025-03-18 22:09:22] debug:      zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=1](ackRx=4)
zigbee2mqtt  | 2025-03-18T22:09:22.507739998Z [2025-03-18 22:09:22] debug:      zh:ember:ezsp: <=== [CBFRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=35 Len=69]
zigbee2mqtt  | 2025-03-18T22:09:22.508847812Z [2025-03-18 22:09:22] debug:      zh:ember:ezsp: ezspIncomingMessageHandler(): callback called with: [type=BROADCAST_LOOPBACK], [apsFrame={"profileId":41440,"clusterId":33,"sourceEndpoint":242,"destinationEndpoint":242,"options":256,"groupId":0,"sequence":171}], [packetInfo:{"senderShortId":0,"senderLongId":"0x0000000000000000","bindingIndex":255,"addressIndex":8,"lastHopLqi":255,"lastHopRssi":0,"lastHopTimestamp":173993955}], [messageContents=19030148e500b9668ec4840b023d0c0000c0c1c2c3c4c5c6c7c8c9cacbcccdcecf]
zigbee2mqtt  | 2025-03-18T22:09:22.960787661Z [2025-03-18 22:09:22] debug:      zh:controller: Green power device '{"sourceID":3297666745,"deviceID":2,"networkAddress":26297,"securityKey":{"type":"Buffer","data":[9,213,221,24,30,79,54,235,236,14,26,236,245,173,144,193]}}' joined
zigbee2mqtt  | 2025-03-18T22:09:22.961033601Z [2025-03-18 22:09:22] debug:      zh:controller: New green power device '0x00000000c48e66b9' joined
[... cut ...]
zigbee2mqtt  | 2025-03-18T22:11:49.495313775Z [2025-03-18 22:11:49] info:       z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x00000000c48e66b9', payload '{"last_seen":"2025-03-18T22:11:49.487Z","linkquality":219}'
zigbee2mqtt  | 2025-03-18T22:11:49.497803307Z [2025-03-18 22:11:49] info:       z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x00000000c48e66b9/last_seen', payload '2025-03-18T22:11:49.487Z'
zigbee2mqtt  | 2025-03-18T22:11:49.500562741Z [2025-03-18 22:11:49] info:       z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x00000000c48e66b9/linkquality', payload '219'

```

</details>

mathieuruellan, 2025-03-19:
That's a big step! Thank you for your work!

dduransseau, 2025-03-19:
Awesome @Nerivec you find a way to manage GP encryption logic, I didn't understood the documentation. @skuizy I prepared an [external_converter](https://gist.github.com/dduransseau/6958ec53f6e8665766903e69146285d3) that should work press 1 is from left to right. It seems that's not possible to combine multiple press as a specific action left+middle.

Nerivec, 2025-03-19:
@dduransseau I had to dig pretty deep for [ZigBee on Host](https://github.com/Koenkk/zigbee2mqtt/discussions/26746), so, I ended up porting some of that stuff over 😉 
Beware though, that external converter will override most green power devices on a same network. Since it's not possible to use the source ID to match against it and prevent this override, relying on the auto-generated converter from the [PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/9020) I mentioned should work, albeit with "limited support" (matching against the official GPDF command IDs directly, so here that would be left=> `off`, middle=> `on`, right => `recall_scene1`, _see [bottom file](https://github.com/Koenkk/zigbee-herdsman-converters/pull/9020/files#diff-47eee2b61258a9d48e4b9104d706a4aeacec2e1baba7d21f34db7d2963357d28R2144) for all the command IDs of the GP spec_).

dduransseau, 2025-03-19:
Interesting, I hadn't seen your ZigBee on Host initiative, I might help to provide some capture. Anyway that's a great improvement to have a fully functional GP stack. Your PR for auto-generated seems the most pragmatic, specific device mapping seems complicated, especially for this switch where ieee address change each time pairing occur.

huafu, 2025-03-20:
@Nerivec thanks for your work. Unfortunately I don't have much time to help more than the little logs I provided on this one. I barely spend a few minutes on my computer everyday.
Anyway, how can I test your converter (if it can be of any help)? Or should I just wait for it to be merged?

Nerivec, 2025-03-20:
The Green Power definition auto-generation PR should be available in a little while in latest-dev. Let me know how that works out for that Moes ZEU2S, if anyone can test it.

dduransseau, 2025-03-20:
Just tested, that work fine. Below generated definition:
```
import {genericGreenPower} from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    fingerprint: [{modelID: 'GreenPower_2', ieeeAddr: new RegExp('^0x00000000baa0b693$')}],
    model: 'GreenPower_2',
    vendor: '',
    description: 'Automatically generated definition for Green Power',
    extend: [genericGreenPower()],
};
```

huafu, 2025-03-21:
Thank you @Nerivec I'm using it in home assistant now, it works!

MaxwellJK, 2025-03-21:
Thanks @Nerivec for the pr, i am testing it as well - I own a 2 gang switch from Moes.
I am not so familiar with Z2M as i mostly use ZHA - I have installed Z2M just to make use of this switch.

I have an unconventional setup where I am using both ZHA and Z2M using the same dongle (consumed via `ser2net`).
The dongle is a Zigbee 3.0 USB Dongle Plus.

All is running smoothly and thanks to the generated definition, the switch is now supported in Z2M but when I press the buttons i don't see anything in the logs and also in HA i don't see the switch.
I generated a definition that looks similar to the one that [dduransseau](https://github.com/Koenkk/zigbee2mqtt/issues/19405#issuecomment-2741724952) reported and saved it in a folder called `external_converters`.

Mosquitto seems running fine and both Z2M and HA are able to connect to it; the topic is set up correctly. 

What am I missing?

Thanks in advance for any support.

Nerivec, 2025-03-21:
@MaxwellJK I assume you mean the zstack version of the Sonoff Dongle? If yes, can you provide some `debug` logs? I haven't seen any recent one, I'm not sure what zstack provides for the incoming GP data, might need a few tweaks (if at all possible).
You also don't want to be saving the generated definition as an external converter. You need to rely on the auto-generation that Z2M does internally (since this device changes source ID -off-spec- there is no way to properly identify it for the moment, so, Z2M will just update the definition internally whenever the source ID changes). The external converter will just get in the way and become obsolete on first source ID change anyway.

MaxwellJK, 2025-03-21:
@Nerivec thanks for the quick response.
Yes, I am using ZStack version of Sonoff Dongle.

This is the pairing log - i have quite some devices so I hope there is not much garbage.

Thanks a lot for your efforts 😃 

```
[2025-03-21 14:10:02] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254},"status":"ok","transaction":"8gx8x-10"}'
[2025-03-21 14:10:05] debug: 	zh:zstack:unpi:parser: <-- [254,81,68,129,0,0,33,0,103,102,242,242,1,174,0,185,40,204,0,0,61,17,108,4,0,8,103,102,116,199,234,0,0,0,224,43,2,137,243,1,229,40,120,251,139,83,105,191,206,44,2,83,90,186,46,184,181,184,254,234,61,0,0,55]
[2025-03-21 14:10:05] debug: 	zh:zstack:unpi:parser: --- parseNext [254,81,68,129,0,0,33,0,103,102,242,242,1,174,0,185,40,204,0,0,61,17,108,4,0,8,103,102,116,199,234,0,0,0,224,43,2,137,243,1,229,40,120,251,139,83,105,191,206,44,2,83,90,186,46,184,181,184,254,234,61,0,0,55]
[2025-03-21 14:10:05] debug: 	zh:zstack:unpi:parser: <-- [100,122,98,105,112,97,48,64,71,80,48,48,48,49,206,84,95,216,84,95,29,31]
[2025-03-21 14:10:05] debug: 	zh:zstack:unpi:parser: --- parseNext [254,81,68,129,0,0,33,0,103,102,242,242,1,174,0,185,40,204,0,0,61,17,108,4,0,8,103,102,116,199,234,0,0,0,224,43,2,137,243,1,229,40,120,251,139,83,105,191,206,44,2,83,90,186,46,184,181,184,254,234,61,0,0,55,100,122,98,105,112,97,48,64,71,80,48,48,48,49,206,84,95,216,84,95,29,31]
[2025-03-21 14:10:05] debug: 	zh:zstack:unpi:parser: --> parsed 81 - 2 - 4 - 129 - [0,0,33,0,103,102,242,242,1,174,0,185,40,204,0,0,61,17,108,4,0,8,103,102,116,199,234,0,0,0,224,43,2,137,243,1,229,40,120,251,139,83,105,191,206,44,2,83,90,186,46,184,181,184,254,234,61,0,0,55,100,122,98,105,112,97,48,64,71,80,48,48,48,49,206,84,95,216,84,95,29] - 31
[2025-03-21 14:10:05] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":33,"srcaddr":26215,"srcendpoint":242,"dstendpoint":242,"wasbroadcast":1,"linkquality":174,"securityuse":0,"timestamp":13379769,"transseqnumber":0,"len":61,"data":{"type":"Buffer","data":[17,108,4,0,8,103,102,116,199,234,0,0,0,224,43,2,137,243,1,229,40,120,251,139,83,105,191,206,44,2,83,90,186,46,184,181,184,254,234,61,0,0,55,100,122,98,105,112,97,48,64,71,80,48,48,48,49,206,84,95,216]}}
[2025-03-21 14:10:05] info: 	zh:controller:greenpower: [COMMISSIONING] from=26215
[2025-03-21 14:10:05] info: 	zh:controller:greenpower: [COMMISSIONING] from=26215 rssi=58 linkQuality=Moderate
[2025-03-21 14:10:05] debug: 	zh:controller:greenpower: [PAIRING] options=58696 (appId=0 communicationMode=2) wasBroadcast=true gppNwkAddr=25655
[2025-03-21 14:10:05] debug: 	zh:zstack:znp: --> SREQ: AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":60,"options":0,"radius":30,"len":33,"data":{"type":"Buffer","data":[25,19,1,72,229,0,103,102,116,199,132,11,2,234,61,0,0,192,193,194,195,196,197,198,199,200,201,202,203,204,205,206,207]}}
[2025-03-21 14:10:05] debug: 	zh:zstack:unpi:writer: --> frame [254,53,36,2,2,253,255,0,0,0,0,0,0,242,0,0,242,33,0,60,0,30,33,0,25,19,1,72,229,0,103,102,116,199,132,11,2,234,61,0,0,192,193,194,195,196,197,198,199,200,201,202,203,204,205,206,207,127]
[2025-03-21 14:10:05] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-03-21 14:10:05] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,2,0,103]
[2025-03-21 14:10:05] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,2,0,103]
[2025-03-21 14:10:05] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 2 - [0] - 103
[2025-03-21 14:10:05] debug: 	zh:zstack:znp: <-- SRSP: AF - dataRequestExt - {"status":0}
[2025-03-21 14:10:05] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-03-21 14:10:05] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,2,0,103,254,3,68,128,0,242,60,9]
[2025-03-21 14:10:05] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,2,0,103,254,3,68,128,0,242,60,9]
[2025-03-21 14:10:05] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 2 - [0] - 103
[2025-03-21 14:10:05] debug: 	zh:zstack:znp: <-- SRSP: AF - dataRequestExt - {"status":0}
[2025-03-21 14:10:05] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,242,60,9]
[2025-03-21 14:10:05] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,242,60] - 9
[2025-03-21 14:10:05] debug: 	zh:zstack:znp: <-- AREQ: AF - dataConfirm - {"status":0,"endpoint":242,"transid":60}
[2025-03-21 14:10:05] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-03-21 14:10:05] debug: 	zh:controller: Green power device '{"sourceID":3346294375,"deviceID":2,"networkAddress":26215,"securityKey":{"type":"Buffer","data":[1,229,40,120,251,139,83,105,191,206,44,2,83,90,186,46]}}' joined
[2025-03-21 14:10:05] debug: 	zh:controller: New green power device '0x00000000c7746667' joined
[2025-03-21 14:10:05] debug: 	zh:controller: Creating device '0x00000000c7746667'
[2025-03-21 14:10:05] debug: 	zh:controller:database: Writing database to '/app/data/database.db'
[2025-03-21 14:10:05] debug: 	zh:controller:database: Writing database to '/app/data/database.db'
[2025-03-21 14:10:05] debug: 	zhc: Getting definitions for: sunricher.js,48,sunricher.js,47,sunricher.js,46,philips.js,418,legrand.js,19,led_trading.js,0,enocean.js,1,enocean.js,0
[2025-03-21 14:10:05] debug: 	zhc: Getting definitions for: sunricher.js,48,sunricher.js,47,sunricher.js,46,philips.js,418,legrand.js,19,led_trading.js,0,enocean.js,1,enocean.js,0
[... cut ...]
[2025-03-21 14:10:06] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00000000c7746667","ieee_address":"0x00000000c7746667","status":"successful","supported":false},"type":"device_interview"}'
[2025-03-21 14:10:08] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":0,"transaction":"8gx8x-11"}'
[2025-03-21 14:10:08] info: 	z2m: Zigbee: disabling joining new devices.
[2025-03-21 14:10:08] debug: 	zh:controller: Disable joining

Nerivec, 2025-03-21:
Saying that it's `NOT supported` is normal, because it's using the auto-generated definition, which is not considered "supported" by Z2M (_auto-generated definitions can sometimes be missing some features_).
Do you have the logs from pushing a few buttons after this?

MiAutomations, 2025-03-21:
Hello to all and thank you for all your work developing and share this information  in order to get these devices supported by the Z2M integration !!

Regarding my tests, I used the external converted shared by the @dduransseau on this post [https://github.com/Koenkk/zigbee2mqtt/issues/19405#issuecomment-2736053380](url)

I successfully paired the 2 gans and 3 gans devices that I have, but I'm not able to use once there are no action from the keys pressed 

[image]

Can anyone please confirm if already have success controlling some device from this Switcth using the Z2M integration? 

Thank you

Nerivec, 2025-03-21:
@MiAutomations make sure you use "latest" latest-dev (edge), and [remove the external converter](https://github.com/Koenkk/zigbee2mqtt/issues/19405#issuecomment-2743447035), then pair the device again. If you have any troubles after that, share the `debug` log of the pairing/using.

MaxwellJK, 2025-03-21:
@Nerivec ok, thanks for clarifying it.

I managed to press the buttons and here is the debug log for them.

```
[2025-03-21 16:37:15] debug: 	zh:zstack:unpi:parser: <-- [254,42,68,129,0,0,33,0,61,7,242,242,1,174,0,223,153,225,0,0,22,17,35,4,48,11,61,7,203,143,114,64,0,0,194,0,84,95,222,211,154,18,41,84,95,29,216]
[2025-03-21 16:37:15] debug: 	zh:zstack:unpi:parser: --- parseNext [254,42,68,129,0,0,33,0,61,7,242,242,1,174,0,223,153,225,0,0,22,17,35,4,48,11,61,7,203,143,114,64,0,0,194,0,84,95,222,211,154,18,41,84,95,29,216]
[2025-03-21 16:37:15] debug: 	zh:zstack:unpi:parser: --> parsed 42 - 2 - 4 - 129 - [0,0,33,0,61,7,242,242,1,174,0,223,153,225,0,0,22,17,35,4,48,11,61,7,203,143,114,64,0,0,194,0,84,95,222,211,154,18,41,84,95,29] - 216
[2025-03-21 16:37:15] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":33,"srcaddr":1853,"srcendpoint":242,"dstendpoint":242,"wasbroadcast":1,"linkquality":174,"securityuse":0,"timestamp":14784991,"transseqnumber":0,"len":22,"data":{"type":"Buffer","data":[17,35,4,48,11,61,7,203,143,114,64,0,0,194,0,84,95,222,211,154,18,41]}}
[2025-03-21 16:37:15] debug: 	zh:controller:greenpower: [UNHANDLED_CMD/PASSTHROUGH] command=0xc2 from=1853
[2025-03-21 16:37:15] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-03-21 16:37:15] debug: 	zh:controller: Received payload: clusterID=33, address=1853, groupID=0, endpoint=242, destinationEndpoint=242, wasBroadcast=true, linkQuality=174, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":35,"commandIdentifier":4},"payload":{"options":2864,"srcID":2412447549,"frameCounter":16498,"commandID":194,"payloadSize":0,"commandFrame":{"raw":{"type":"Buffer","data":[84,95,222,211,154,18,41]}},"gppNwkAddr":24404,"gppGpdLink":222,"mic":689085139},"command":{"ID":4,"parameters":[{"name":"options","type":25},{"name":"srcID","type":35,"conditions":[{"type":"bitFieldEnum","param":"options","offset":0,"size":3,"value":0}]},{"name":"gpdIEEEAddr","type":240,"conditions":[{"type":"bitFieldEnum","param":"options","offset":0,"size":3,"value":2}]},{"name":"gpdEndpoint","type":32,"conditions":[{"type":"bitFieldEnum","param":"options","offset":0,"size":3,"value":2}]},{"name":"frameCounter","type":35},{"name":"commandID","type":32},{"name":"payloadSize","type":32},{"name":"commandFrame","type":1009},{"name":"gppNwkAddr","type":33,"conditions":[{"type":"bitMaskSet","param":"options","mask":2048}]},{"name":"gppGpdLink","type":24,"conditions":[{"type":"bitMaskSet","param":"options","mask":2048}]},{"name":"mic","type":35,"conditions":[{"type":"bitMaskSet","param":"options","mask":512}]}],"name":"commissioningNotification"}}
[2025-03-21 16:37:15] debug: 	zhc: Getting definitions for: sunricher.js,48,sunricher.js,47,sunricher.js,46,philips.js,418,legrand.js,19,led_trading.js,0,enocean.js,1,enocean.js,0
[2025-03-21 16:37:15] debug: 	zhc: Candidates for 0x000000008fcb073d/GreenPower_2: SR-ZGP2801K-5C/Sunricher,SR-ZGP2801K4-DIM/Sunricher,SR-ZGP2801K2-DIM/Sunricher,8718696743133/Philips,ZLGP17/ZLGP18/Legrand,9133/LED-Trading,PTM 215ZE/EnOcean,PTM 215Z/EnOcean
[2025-03-21 16:37:15] debug: 	z2m: Received Zigbee message from '0x000000008fcb073d', type 'commandCommissioningNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[84,95,222,211,154,18,41],"type":"Buffer"}},"commandID":194,"frameCounter":16498,"gppGpdLink":222,"gppNwkAddr":24404,"mic":689085139,"options":2864,"payloadSize":0,"srcID":2412447549}' from endpoint 242 with groupID 0
[2025-03-21 16:37:15] debug: 	z2m: Skipping message, still interviewing
[2025-03-21 16:37:15] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,2,0,103]
[2025-03-21 16:37:15] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,2,0,103]
[2025-03-21 16:37:15] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 2 - [0] - 103
[2025-03-21 16:37:15] debug: 	zh:zstack:znp: <-- SRSP: AF - dataRequestExt - {"status":0}
[2025-03-21 16:37:15] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-03-21 16:37:21] debug: 	zh:zstack:unpi:parser: <-- [254,42,68,129,0,0,33,0,61,7,242,242,1,171,0,200,32,231,0,0,22,17,36,4,48,11,61,7,203,143,123,64,0,0,134,0,84,95,220,179,169,171,227,84,95,29,29]
[2025-03-21 16:37:21] debug: 	zh:zstack:unpi:parser: --- parseNext [254,42,68,129,0,0,33,0,61,7,242,242,1,171,0,200,32,231,0,0,22,17,36,4,48,11,61,7,203,143,123,64,0,0,134,0,84,95,220,179,169,171,227,84,95,29,29]
[2025-03-21 16:37:21] debug: 	zh:zstack:unpi:parser: --> parsed 42 - 2 - 4 - 129 - [0,0,33,0,61,7,242,242,1,171,0,200,32,231,0,0,22,17,36,4,48,11,61,7,203,143,123,64,0,0,134,0,84,95,220,179,169,171,227,84,95,29] - 29
[2025-03-21 16:37:21] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":33,"srcaddr":1853,"srcendpoint":242,"dstendpoint":242,"wasbroadcast":1,"linkquality":171,"securityuse":0,"timestamp":15147208,"transseqnumber":0,"len":22,"data":{"type":"Buffer","data":[17,36,4,48,11,61,7,203,143,123,64,0,0,134,0,84,95,220,179,169,171,227]}}
[2025-03-21 16:37:21] debug: 	zh:controller:greenpower: [UNHANDLED_CMD/PASSTHROUGH] command=0x86 from=1853
[2025-03-21 16:37:21] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-03-21 16:37:21] debug: 	zh:controller: Received payload: clusterID=33, address=1853, groupID=0, endpoint=242, destinationEndpoint=242, wasBroadcast=true, linkQuality=171, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"d [... cut ...]
[2025-03-21 16:37:30] debug: 	zh:zstack:unpi:parser: --- parseNext [254,2,97,1,89,6,61]
[2025-03-21 16:37:30] debug: 	zh:zstack:unpi:parser: --> parsed 2 - 3 - 1 - 1 - [89,6] - 61
[2025-03-21 16:37:30] debug: 	zh:zstack:znp: <-- SRSP: SYS - ping - {"capabilities":1625}
[2025-03-21 16:37:30] debug: 	zh:zstack:unpi:parser: --- parseNext []
```
I noticed that soon after pairing and pressing the buttons, the link quality metric is updated on Z2M. After a few mins though, if i keep pressing the buttons, i don't see any log anymore.

[53 more replies]

closed 2025-06-20 by github-actions[bot]