Koenkk/zigbee2mqtt #19407: NOUS zigbee 16A socket plug randomly change its state

devironglove, 2023-10-24:
### What happened?

After an update of Z2MQTT, my zigbee network is experiencing random plug shutting down.
For some of them, when I try to change the state, the state change but after between 10sec and 15minutes, the state revert back and sometimes I see Z2MQTT returning a timeout.

The device I'm experiencing this behavior is the NOUS Plug  [A1Z](https://www.zigbee2mqtt.io/devices/A1Z.html#nous-a1z)

### What did you expect to happen?

If I set a plug to be ON, it should stay ON and not randomly become OFF for I don't know which reason.

### How to reproduce it (minimal and precise)

On the current branch from start Okt 2023.

### Zigbee2MQTT version

1.33.1

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Debug log

Info 2023-10-24 14:35:55Update of 'Prise NOUS bureau' at 70.79%, ≈ 6 minutes remaining
Info 2023-10-24 14:35:55MQTT publish: topic 'zigbee2mqtt/Prise NOUS bureau', payload '{"child_lock":"UNLOCK","current":0,"energy":81.59,"indicator_mode":"off/on","linkquality":80,"power":0,"power_outage_memory":"on","state":"OFF","update":{"installed_version":74,"latest_version":192,"progress":70.79,"remaining":337,"state":"updating"},"update_available":null,"voltage":245}'
Debug 2023-10-24 14:35:56Received MQTT message on 'zigbee2mqtt/Prise NOUS bureau/set' with data 'ON'
Debug 2023-10-24 14:35:56Publishing 'set' 'state' to 'Prise NOUS bureau'
Debug 2023-10-24 14:35:56Received Zigbee message from 'Prise NOUS bureau', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info 2023-10-24 14:35:56MQTT publish: topic 'zigbee2mqtt/Prise NOUS bureau', payload '{"child_lock":"UNLOCK","current":0,"energy":81.59,"indicator_mode":"off/on","linkquality":83,"power":0,"power_outage_memory":"on","state":"ON","update":{"installed_version":74,"latest_version":192,"progress":70.79,"remaining":337,"state":"updating"},"update_available":null,"voltage":245}'
Info 2023-10-24 14:35:56MQTT publish: topic 'zigbee2mqtt/Prise NOUS bureau', payload '{"child_lock":"UNLOCK","current":0,"energy":81.59,"indicator_mode":"off/on","linkquality":83,"power":0,"power_outage_memory":"on","state":"ON","update":{"installed_version":74,"latest_version":192,"progress":70.79,"remaining":337,"state":"updating"},"update_available":null,"voltage":245}'
Info 2023-10-24 14:36:26Update of 'Prise NOUS bureau' at 72.02%, ≈ 5 minutes remaining
Info 2023-10-24 14:36:26MQTT publish: topic 'zigbee2mqtt/Prise NOUS bureau', payload '{"child_lock":"UNLOCK","current":0,"energy":81.59,"indicator_mode":"off/on","linkquality":80,"power":0,"power_outage_memory":"on","state":"ON","update":{"installed_version":74,"latest_version":192,"progress":72.02,"remaining":329,"state":"updating"},"update_available":null,"voltage":245}'
Info 2023-10-24 14:36:56Update of 'Prise NOUS bureau' at 73.37%, ≈ 5 minutes remaining
Info 2023-10-24 14:36:56MQTT publish: topic 'zigbee2mqtt/Prise NOUS bureau', payload '{"child_lock":"UNLOCK","current":0,"energy":81.59,"indicator_mode":"off/on","linkquality":87,"power":0,"power_outage_memory":"on","state":"ON","update":{"installed_version":74,"latest_version":192,"progress":73.37,"remaining":319,"state":"updating"},"update_available":null,"voltage":245}'
Debug 2023-10-24 14:37:14Received Zigbee message from 'Prise NOUS bureau', type 'attributeReport', cluster 'genBasic', data '{"65506":53,"65508":0,"appVersion":74}' from endpoint 1 with groupID 0
Info 2023-10-24 14:37:26Update of 'Prise NOUS bureau' at 75.11%, ≈ 5 minutes remaining
Info 2023-10-24 14:37:26MQTT publish: topic 'zigbee2mqtt/Prise NOUS bureau', payload '{"child_lock":"UNLOCK","current":0,"energy":81.59,"indicator_mode":"off/on","linkquality":83,"power":0,"power_outage_memory":"on","state":"ON","update":{"installed_version":74,"latest_version":192,"progress":75.11,"remaining":301,"state":"updating"},"update_available":null,"voltage":245}'
Debug 2023-10-24 14:37:46Received Zigbee message from 'Prise NOUS bureau', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":250}' from endpoint 1 with groupID 0
Info 2023-10-24 14:37:46MQTT publish: topic 'zigbee2mqtt/Prise NOUS bureau', payload '{"child_lock":"UNLOCK","current":0,"energy":81.59,"indicator_mode":"off/on","linkquality":80,"power":0,"power_outage_memory":"on","state":"ON","update":{"installed_version":74,"latest_version":192,"progress":75.11,"remaining":301,"state":"updating"},"update_available":null,"voltage":250}'
Debug 2023-10-24 14:37:46Received Zigbee message from 'Prise NOUS bureau', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,8159]}' from endpoint 1 with groupID 0
Info 2023-10-24 14:37:46MQTT publish: topic 'zigbee2mqtt/Prise NOUS bureau', payload '{"child_lock":"UNLOCK","current":0,"energy":81.59,"indicator_mode":"off/on","linkquality":83,"power":0,"power_outage_memory":"on","state":"ON","update":{"installed_version":74,"latest_version":192,"progress":75.11,"remaining":301,"state":"updating"},"update_available":null,"voltage":250}'
Info 2023-10-24 14:37:56Update of 'Prise NOUS bureau' at 76.64%, ≈ 5 minutes remaining
Info 2023-10-24 14:37:56MQTT publish: topic 'zigbee2mqtt/Prise NOUS bureau', payload '{"child_lock":"UNLOCK","current":0,"energy":81.59,"indicator_mode":"off/on","linkquality":83,"power":0,"power_outage_memory":"on","state":"ON","update":{"installed_version":74,"latest_version":192,"progress":76.64,"remaining":286,"state":"updating"},"update_available":null,"voltage":250}'
Debug 2023-10-24 14:38:26Received Zigbee message from 'Prise NOUS bureau', type 'attributeReport', cluster 'manuSpecificTuya_3', data '{"53249":2}' from endpoint 1 with groupID 0
Debug 2023-10-24 14:38:26Received Zigbee message from 'Prise NOUS bureau', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info 2023-10-24 14:38:26MQTT publish: topic 'zigbee2mqtt/Prise NOUS bureau', payload '{"child_lock":"UNLOCK","current":0,"energy":81.59,"indicator_mode":"off/on","linkquality":83,"power":0,"power_outage_memory":"on","state":"OFF","update":{"installed_version":74,"latest_version":192,"progress":76.64,"remaining":286,"state":"updating"},"update_available":null,"voltage":250}'
Debug 2023-10-24 14:38:26Received Zigbee message from 'Prise NOUS bureau', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsVoltage":251}' from endpoint 1 with groupID 0
Info 2023-10-24 14:38:26MQTT publish: topic 'zigbee2mqtt/Prise NOUS bureau', payload '{"child_lock":"UNLOCK","current":0,"energy":81.59,"indicator_mode":"off/on","linkquality":83,"power":0,"power_outage_memory":"on","state":"OFF","update":{"installed_version":74,"latest_version":192,"progress":76.64,"remaining":286,"state":"updating"},"update_available":null,"voltage":251}'
Info 2023-10-24 14:38:27Update of 'Prise NOUS bureau' at 78.26%, ≈ 4 minutes remaining
Info 2023-10-24 14:38:27MQTT publish: topic 'zigbee2mqtt/Prise NOUS bureau', payload '{"child_lock":"UNLOCK","current":0,"energy":81.59,"indicator_mode":"off/on","linkquality":83,"power":0,"power_outage_memory":"on","state":"OFF","update":{"installed_version":74,"latest_version":192,"progress":78.26,"remaining":269,"state":"updating"},"update_available":null,"voltage":251}'
Debug 2023-10-24 14:38:31Received Zigbee message from 'Prise NOUS bureau', type 'attributeReport', cluster 'manuSpecificTuya_3', data '{"53249":0}' from endpoint 1 with groupID 0
Debug 2023-10-24 14:38:32Received Zigbee message from 'Prise NOUS bureau', type 'attributeReport', cluster 'manuSpecificTuya_3', data '{"53249":2}' from endpoint 1 with groupID 0
Debug 2023-10-24 14:38:32Received Zigbee message from 'Prise NOUS bureau', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info 2023-10-24 14:38:32MQTT publish: topic 'zigbee2mqtt/Prise NOUS bureau', payload '{"child_lock":"UNLOCK","current":0,"energy":81.59,"indicator_mode":"off/on","linkquality":83,"power":0,"power_outage_memory":"on","state":"OFF","update":{"installed_version":74,"latest_version":192,"progress":78.26,"remaining":269,"state":"updating"},"update_available":null,"voltage":251}'

devironglove, 2023-10-24:
As of writing this ticket, my update for some NOUS plugs finally completed... And the plugs went back to state as BEFORE the action ! (it was off while I was trying to set it to ON -> after the update, automagically went to ON)
I suspect that the firmware for this one was reaaally old (I don't often update my devices -_- )
I'll keep posted if the behavior still happens

meganiekje, 2023-10-24:
I see indeed 2 possible problems with software version 74: 
1) Random switch-off due to hardware communication problem inside the plug.
2) 'safety' switch-off above 250 volts.
Both problems should be solved with firmware 192.

devironglove, 2023-11-01:
Problem doesn't happens anymore once all devices are updated to latest firmware :) 
Sorry for the inconvenience

closed 2023-11-01 by devironglove