Koenkk/zigbee2mqtt #19412: BHT-002-GCLZB issue with edge version

Minims, 2023-10-24:
### What happened?

I am not able to define a setpoint below 10°C.
I was using 5°C in  1.33.1-1 , but it does not work anymore with the last edge version 1.33.1-dev commit: 4d8c6b2
If my thermostat is set to 5°C on HA, it is set to 50°C in Z2M.

### What did you expect to happen?

Be able to use a setpoint of 5°C  (below 10°c)

### How to reproduce it (minimal and precise)

Try to define a "Current heating setpoint" of 5°C.
As you can see in log, setpoint try to be set @50° instead of 5°C

### Zigbee2MQTT version

1.33.1-dev commit: 4d8c6b2

### Adapter firmware version

20230507

### Adapter

SONOFF ZigBee 3.0 USB Dongle Plus

### Debug log

Info 2023-10-24 21:38:13MQTT publish: topic 'zigbee2mqtt/Thermostat', payload '{"child_lock":"LOCK","current_heating_setpoint":50,"deadzone_temperature":1,"heat":"OFF","last_seen":"2023-10-24T21:38:13+02:00","linkquality":149,"local_temperature":21,"local_temperature_calibration":-2,"max_temperature":35,"max_temperature_limit":35,"min_temperature_limit":1,"preset":"hold","preset_mode":"hold","running_state":"idle","sensor":"IN","system_mode":"off"}'

Koenkk, 2023-10-25:
can you provide the data/database.db entry of your device?

Minims, 2023-10-25:
Here it is:
```
{"id":5,"type":"Router","ieeeAddr":"0x5c0272fffeecbe13","nwkAddr":5964,"manufId":4098,"manufName":"_TZE200_aoclfnxz","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_aoclfnxz","powerSource":1,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1698257324069,"defaultSendRequestWhen":"immediate"}
```

Koenkk, 2023-10-27:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-10-27 by Koenkk

Minims, 2023-10-28:
I will check that tomorrow, Thanks !