Koenkk/zigbee2mqtt #19416: Connectivity Issues (router dropouts/offline) - ZBDongle-P

charlesomer, 2023-10-24:
### What happened?

Hi, I've been running Zigbee2mqtt for a while now and have been trying to get it stable. Unfortunately I've been unsuccessful with everything I've tried and now I've pretty much run out of ideas so any help will be much appreciated!

I have 18 devices currently joined (all routers):
 - 11 x Philips Hue LCT001
 - 5 x IKEA control outlets
 - 2 x ZBDongle-P with router firmware (20221102)

The IKEA outlets and the 2 routers are generally okay, are available most of the time and respond as expected. They show up in the map etc. Sometimes one or two can become unavailable after a restart or similar, but this usually resolves itself (there's 1 IKEA outlet unavailable at the moment for example).

The hue bulbs are much more inconsistent, most will remain offline (at the moment 7 are offline). Due to the way they are setup, they are only powered at night (this is something I am not able to change). I understand this isn't ideal but I have a similar situation with the same bulbs in another setup (although not as many) and this works fine. They reconnect once powered on etc.

Here are some of the things I've tried so far:
 - Using ZBDongle-E as the coordinator. This is what I started with originally as I am using the same in a different setup elsewhere and it works fine. I know it's not one of the recommended ones though which is why I switched to ZBDongle-P.
 - Using a different Zigbee channel, with the ZBDongle-E I was using channel 25 but I switched this to the default 11 for the  ZBDongle-P.
 - Adjusting the power of the routers and coordinator, I have tried 5dBm to 20dBm. At the moment they are all set to 15.
 - Adding additional router devices close to the bulbs. Extra IKEA outlets and the ZBDongle-P routers - some of these are on the other side of the same wall as a hue bulb that will be marked as offline.
 - Updating the firmware of the coordinator.
 - Updating to latest-dev.
 - Powering off the PC for 15 minutes (or more).
 - Repairing all the devices (this was done when switching coordinators).
 - Using USB extension cables to get the coordinator as far away from the PC as possible.

I have read the map isn't the best tool but I noticed on there that some bulbs will have links drawn despite the LQI being 0. Some of the devices marked as offline will have single numbers associated with their routes but never a "/". I'm not 100% sure what this means but my guess is that something can "see" the device but the device can't talk back maybe?

Also, when the bulbs are powered on in the evening, they do show up as "online". Which leads me to believe that zigbee2mqtt can communicate with them in some way, but then upon the next availability check they go offline again.

The distance between some of these bulbs is only 2 metres or so and the distance to a known working router is sometimes less than a metre (albeit through a wall). In fact one of the ones furthest away from the next closest router appears to work all the time.

Is there anything else I can try to get this more reliable?

I'll add more if I remember or discover extra info.

Thank you
Charles

### What did you expect to happen?

Devices connect when powered on and are stable after a short amount of time.

### How to reproduce it (minimal and precise)

Run zigbee2mqtt (docker) with ZBDongle-P. Ubuntu Ubuntu 22.04.3 LTS.
Pair multiple LCT001.
Observe offline devices.

### Zigbee2MQTT version

1.33.1-dev commit: 0e55b38

### Adapter firmware version

20230507

### Adapter

ZBDongle-P

charlesomer, 2023-10-29:
I've also noticed that the Hue bulbs do not mesh with one another (at least according to the map). There's not a single link between two LCT001 devices. I was expecting the hue bulbs to behave correctly but perhaps this is an incorrect assumption.

charlesomer, 2023-12-12:
Some weak routes do appear when selecting "none of the above", but it's not consistent. I am able to force devices to an "online" state again by refreshing one of the devices properties a few times (4) very quickly - after 30 seconds or so the device is marked as online again. However, this is not permanent and devices will go back offline again shortly after. This behaviour includes devices which are powered all the time too.

So far I've tried the following firmware:
- 20220219
- 20221226
- 20230507
- 20231112

The next things I'm debating trying is switching coordinators to either zzh or slaesh and changing the channel back to 25.

xelemorf, 2023-12-13:
If you still have your Dongle-E you could use that one as a router and pair those problematic devices to it. Dongle-E is quite good at router functionality.

charlesomer, 2023-12-13:
Thanks for your reply. I don't have any end devices, they are all routers so should dynamically adjust their routes to the best path from what I understand. I have the Dongle-E setup and running as a router (firmware 20220515) along with two other Dongle-P.

Some permanently powered devices do end up in an "offline" state occasionally once the hue bulbs are powered on (one is offline at the moment). This confuses me the most really, if the Zigbee network knows a good route to a router, why is switching on another router elsewhere causing drop outs?

I believe it's a USB2 extension cable and a USB2 port.

charlesomer, 2023-12-13:
I've just checked again and every single router (including the hue bulbs) shows up as expected in Home Assistant history as "available" when they get powered on. I'm struggling to understand why the availability check fails if they are all shown online at this point.

xelemorf, 2023-12-13:
The default availability state values did not work for me so ended up changing to advanced mode and specifying custom values and then fine tuning them to the ones shown on the screenshot.
[image]

On the other hand you may want to experiment with the Legacy availability payload under Advanced settings. Attaching another screenshot for that.
[image]

Most of the Zigbee devices does not necessarily comply with Zigbee standards (I believe this is mainly for vendor-lock attempts to keep you in a closed eco-system), and even the standards has multiple revisions (currently I believe we are on Zigbee 3.0). I could observe major issues for example with all Xiaomi and Aqara devices (both are manufactured by LUMI) where they just straight up refuse pairing with any other vendor's product even refusing connecting to the coordinator directly (_In some case I was able to force it but they ended up disconnecting and having zero LQI, sometimes they are able to reconnect but in case of a switch thr first button press might be lost and only thr second gets registered but eventually gone from the network after a few hours)_

You need to find out which devices from which vendors are playing nice with each other. I have found that the cheapest devices sold on Aliexpress (using "Choice" sales) which are TuYa compatible works the best and the most reliable, bot to mention you can buy 8 devices for the price of one Aqara - Just a thought to consider.

charlesomer, 2023-12-13:
I have tried increasing the availably to 30 minutes in the past which didn't fix things, it just delayed how long it took for devices to be marked as offline when they were powered on. The communication to devices was still problematic.

Many people have recommended Hue, Sonoff ZB-Dongle E/P and Ikea on the bases they work well as Zigbee devices so I'd have expected these ones to comply with the specification enough to work reliably.

On using the legacy availability payload, I believe this only relates to how the availability is reported by Zigbee2mqtt rather than how the availability is retrieved for the devices: https://www.zigbee2mqtt.io/guide/configuration/device-availability.html#state-retrieval. I did try it though, it made no difference :/

xelemorf, 2023-12-14:
I'm using this fw below for Dongle-P which works quite well, can you check your version?
https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_coordinator_20230507.zip

Can you share a network map with visible LQI values?

Can you also share your /homeassistant/zigbee2mqtt/configuration.yaml file contents while making sure **not to share** the network_key, ext_pan_id, and pan_id segments?

Can you confirm you are in Edge version of Zigbee2Mqtt?

charlesomer, 2023-12-14:
My coordinator version is currently 20230507 and I'm using the dongle-P as the coordinator.

I haven't enabled hardware flow control. 
[z2m.zip](https://github.com/Koenkk/zigbee2mqtt/files/13678838/z2m.zip)

Current version of z2m is: 1.34.0-dev commit 25de4fd

Those settings are related to MQTT, I didn't think these made any difference to how z2m communicates with devices? I see the issues in the Z2M UI which I don't think is dependent on MQTT.

xelemorf, 2023-12-14:
I'd suggest:
- Set transmit power to 20 for Dongle-P
- Disassemble your Coordinator Dongle-P and enable thr DIP switch.
- Configure baudrate to 115200
- Check the Dongle-E router firmware, it appears to have a very poor signal while it has by default a transmit power of 20 on fw level (that device would be the stongest on your network with way above 200 LQI in relation of a 61nm flat and concrete walls in my example), this is fw I use: https://github.com/itead/Sonoff_Zigbee_Dongle_Firmware/tree/master/Dongle-E/Router
- How much distance do you have between these devices, what type of walls, any huge metal casing objects near the affected routers like a fridge, washmachine, dishwasher, etc?
- What Wifi channels are you using for 2.4GHz Wifi, are you living in a flat with close proximity neighbors, did you analyze 2.4GHz wireless congestion (neighborhood) in relation to Zigbee coexistence? You might need to disable auto channel selection for your 2.4Ghz aswell (I ended up using highest Ch13 for 2.4GHz wifi and lowest Ch11 for Zigbee)
https://www.metageek.com/training/resources/zigbee-wifi-coexistence/

This overall seems to me a Zigbee signal issue, specifically if you live in a very big house with maybe multiple floors at the same time time.
Would there be a way to get these devices to close proximity with each other for a day and see if the signal is adequate for them, that would easily rule out some of the things.

charlesomer, 2024-01-13:
So over Christmas I tried to stabilise the setup, I have now got it to a point where only roughly 3 of the hue bulbs are problematic (slow to respond sometimes). The rest so far have been better - I'm going to keep an eye on it and see how it goes.

I ended up moving the coordinator to a pi3 so I could place it somewhere else. I believe previously, Zigbee was struggling to determine which route from the coordinator was best. It appeared to switch between high LQIs to low ones randomly, my best guess so far is the lower LQI were fewer "hops" than the higher LQI route.

This also meant I could move some of my repeaters around too. I still need to see if I can get the remaining bulbs working reliably but at least it's an improvement. Hue bulbs still don't show as being linked to one another (just to other manufacturers) in the Zigbee map unless "none of the above" is checked.

closed 2024-08-11 by github-actions[bot]

antst, 2024-11-17:
actually, I think there is something. As it happened, I have couple of LCT001.
They used to work as a charm, but over last year I have seen clear degradation with updating Z2M to newer and newer version.
At the end, with 1.40.1 they almost stopped to work, but at least they were still formally part of the system (although they were paired long time ago). It was with ZDongle-P (TI).
Over last few days I finally made transition to ZDongle-E and  Z2M 1.41.0 in new setup. And those two LCT001  proved to be impossible to pair anymore. And I am not a virgin in pairing, with mu 120 devices of all possible brands in zigbee network.
But I spent hours trying to get LCT001 into new system. And while rest of devices showed kind of more smooth experience with ember that it was with stack.
It always ends up with 
"Interview failed because can not get node descriptor".

RezzZ, 2025-05-09:
i also have issues with my two LCT001 hue bulbs. Even though they appear offline in zigbee2mqtt (for severals hours now) I can still control them over mqtt (via homeassistant)...

drasch, 2025-05-14:
I was having connectivity issues with my LCT001 bulbs with my Hue Bridge v2. I switched to my zigbee dongle as an attempt to resolve this issue.  The issue has continued and/or gotten worse with zwave2mqtt. The bulbs seem to go offline and not come back ever or at lest for several hours. With the Hue Bridge, they'd be offline for a few minutes and then come back. Sometimes, sending a command like changing the brightness turning on/off will cause them to reconnect. 

I'm beginning to wonder if the bulbs/firmware on the bulbs are causing this issue and/or there are quirks that make this worse w/ Zwave2mqtt compared with the hue bridge.

WebfinityDesign, 2025-05-18:
I do have the same problem with 2 LCT001 bulbs after switching to Z2M. Before this exact hardware has been stable for years on ZHA. I have Lexman bulbs and some end devices that are rock-solid on z2m. The coordinator is a Skyconnect.
I'm 95% sure is a z2m issue as it was working in ZHA.