Koenkk/zigbee2mqtt #19419: [New device support]: lumi.remote.acn009 wireless switch zigbee

MiAutomations, 2023-10-25:
### Link

https://www.aqara.cn/smart-wall-switch-h1m_overview

### Database entry

{"id":41,"type":"EndDevice","ieeeAddr":"0x54ef441000559595","nwkAddr":61182,"manufId":4447,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.remote.acn009","epList":[1,2,3],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[0,1,3,18,64704],"outClusterList":[3,6,25],"clusters":{"genBasic":{"attributes":{"hwVersion":0,"appVersion":7,"modelId":"lumi.remote.acn009"}},"aqaraOpple":{"attributes":{"247":{"type":"Buffer","data":[1,33,188,11,3,40,25,4,33,0,0,5,33,1,0,8,33,7,0,10,33,0,0,12,32,30,100,32,1]}}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":0,"inClusterList":[18],"outClusterList":[6],"clusters":{"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":0,"inClusterList":[18],"outClusterList":[6],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":7,"stackVersion":2,"hwVersion":0,"zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1698235873810,"defaultSendRequestWhen":"immediate"}

### Comments

Hello 

I try to add this device to the last available release of the Z2M but appears in the list as not supported device 

Can you please include this in a new future release? 

Than k you for your support and best regards

jfcabral, 2023-11-01:
Hi

This has been reported here https://github.com/Koenkk/zigbee2mqtt/issues/17033 but unfortunately, the issue went stale 😔 

I also have one of these devices, and commented [here ](https://github.com/Koenkk/zigbee2mqtt/issues/17033#issuecomment-1771726787) that it is still not supported on Zigbee2MQTT version
[1.33.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.33.1) 

In the meanwhile, I have just tested [1.33.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.33.2) but the issue still stands.

I have also tested the external converter from: https://github.com/Koenkk/zigbee2mqtt/issues/17033#issue-1622305981 but, although it might be a good starting point, it can't seem to differ between left and right clicks 😕

jfcabral, 2023-12-02:
Hello
The recently released z2m version https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.34.0 included this PR that adds basic support for the H1M https://github.com/Koenkk/zigbee-herdsman-converters/pull/6481

What works:
* single action, now properly distinguishing between left and right clicks 😄 

What still doesn't work:
* **Click mode** - can't change from `fast` to `multi`
* **Operation mode** - can't change from `command` to `event`
* **Binding** also doesn't seem to be working
* **Device Temperature** isn't being mapped

closed 2024-06-30 by github-actions[bot]

Bluemeus, 2024-10-30:
> Hello The recently released z2m version https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.34.0 included this PR that adds basic support for the H1M [Koenkk/zigbee-herdsman-converters#6481](https://github.com/Koenkk/zigbee-herdsman-converters/pull/6481)
> 
> What works:
> 
>     * single action, now properly distinguishing between left and right clicks 😄
> 
> 
> What still doesn't work:
> 
>     * **Click mode** - can't change from `fast` to `multi`
> 
>     * **Operation mode** - can't change from `command` to `event`
> 
>     * **Binding** also doesn't seem to be working
> 
>     * **Device Temperature** isn't being mapped

Are there any news? i found solutions with ZHA, but not with Zigbee2MQTT.

Is anyone out there getting the WRS-02 work with multi-button support?

Best regards