Koenkk/zigbee2mqtt #19420: [New device support]: Eurotronic Comet Zigbee

Xenon69, 2023-10-25:
### Link

www.eurotronic.org (product page no longer exists at the manufacturer website)

### Database entry

{"id":8,"type":"EndDevice","ieeeAddr":"0xf4ce36994767b8a3","nwkAddr":42573,"manufId":4151,"manufName":"Eurotronic","powerSource":"Battery","modelId":"SPZB0001","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":769,"inClusterList":[0,1,3,513,10],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"SPZB0001","manufacturerName":"Eurotronic","powerSource":3,"zclVersion":8,"appVersion":1,"stackVersion":4,"hwVersion":34,"dateCode":"20221110","swBuildId":"001f0140\u0000\u0004\u0000�\u0002\u0000�\u0000\u0000�\u0003\u0002\u0000\f\u0000�\u0000\u0000\u0000\u0000\u0000�\u0006\u0001\u0000��gG�6���\u0002\u0000\u0005"}},"hvacThermostat":{"attributes":{"16386":0,"16387":1900,"16392":1,"localTemp":1800,"pIHeatingDemand":100,"occupiedHeatingSetpoint":1900,"unoccupiedHeatingSetpoint":1900}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":150,"batteryVoltage":229}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b002c3fb432","endpointID":1},{"cluster":513,"type":"endpoint","deviceIeeeAddress":"0x00124b002c3fb432","endpointID":1}],"configuredReportings":[{"cluster":513,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":10,"manufacturerCode":null},{"cluster":513,"attrId":8,"minRepIntval":0,"maxRepIntval":3600,"repChange":10,"manufacturerCode":null},{"cluster":513,"attrId":18,"minRepIntval":0,"maxRepIntval":3600,"repChange":10,"manufacturerCode":null},{"cluster":513,"attrId":20,"minRepIntval":0,"maxRepIntval":3600,"repChange":10,"manufacturerCode":null},{"cluster":513,"attrId":16387,"minRepIntval":0,"maxRepIntval":3600,"repChange":25,"manufacturerCode":4151},{"cluster":513,"attrId":16392,"minRepIntval":0,"maxRepIntval":3600,"repChange":1,"manufacturerCode":4151}],"meta":{}}},"appVersion":1,"stackVersion":4,"hwVersion":34,"dateCode":"20221110","swBuildId":"001f0140\u0000\u0004\u0000�\u0002\u0000�\u0000\u0000�\u0003\u0002\u0000\f\u0000�\u0000\u0000\u0000\u0000\u0000�\u0006\u0001\u0000��gG�6���\u0002\u0000\u0005","zclVersion":8,"i nterviewCompleted":true,"meta":{"configured":1138500639},"lastSeen":1698243001047,"defaultSendRequestWhen":"immediate"}

### Comments

Hello,
Eurotronic Comet Zigbee (radiator valve) is recognized as Eurotronic Spirit Zigbee, and then partially supported.

Thank you for your help to support Comet devices in addition to Spirit devivces.

If needed I can run tests with both models.

Thanks!!

mabl, 2023-11-09:
[53e9e8eae21bc84d32234cbafa0317b6a767860148562427fd3a2b9a28ab90dc_optim.pdf](https://github.com/Koenkk/zigbee2mqtt/files/13311836/53e9e8eae21bc84d32234cbafa0317b6a767860148562427fd3a2b9a28ab90dc_optim.pdf)

Interestingly, a manual including the register description can be found on the internet.

mabl, 2023-11-19:
See also discussion in https://github.com/Koenkk/zigbee2mqtt/discussions/19363.

cynster, 2024-02-16:
I'm having trouble setting up the actual radiator valve (Eurotronic Spirit Zigbee). It shows up as Unsupported:
[image]
[image]

I have tried re-pairing about 4 times but without any luck. I expected it to just show up as normal since it seemed supported: [https://www.zigbee2mqtt.io/devices/SPZB0001.html](https://www.zigbee2mqtt.io/devices/SPZB0001.html)

closed 2024-09-15 by github-actions[bot]

sstidl, 2025-02-22:
Has anyone seen a solution for this? or tried to contact support there?